rockchip-linux/mpp #756: RK3588多路编码消耗时间过长问题

excelement, 2025-01-02:
目前使用 camera_source 获取mppbuffer之后
将MppBuffer放入到编码器中  编码参数为4k  3840*2160 码率8m  h264
同时开启多个编码器对同一个MppBuffer编码 并在都编码完成后退出

2路 编码时间为14毫秒左右
4路 编码时间为28毫秒左右
6路 编码时间为41毫秒左右

rk3588 mpp是否支持同时4路4k编码？感觉像是只有两个vpu进行编码其他的编码任务等待当前任务完成后才会开始编码
怎么配置参数能够控制4路编码在16毫秒以内？

excelement, 2025-01-02:
env打印都是14毫秒左右

[ 4472.750340] rk_vcodec: fdbd0000.rkvenc-core:0 session 1954:55 time: 13567 us
[ 4472.750415] rk_vcodec: fdbe0000.rkvenc-core:1 session 1954:54 time: 13573 us
[ 4472.766863] rk_vcodec: fdbd0000.rkvenc-core:0 session 1954:55 time: 13616 us
[ 4472.766938] rk_vcodec: fdbe0000.rkvenc-core:1 session 1954:54 time: 13571 us

2路==================

very 1.0s: cat /proc/mpp_service/sessions-summary                                                                          rk3588-buildroot: Thu Jan  2 07:40:49 2025

session iova range dump:
   0: 0x00000000ff020000..0x00000000ffff1fff (     16200 KiB)
   1: 0x00000000fe400000..0x00000000ff013fff (     12368 KiB)
   2: 0x00000000f70b0000..0x00000000f7cc3fff (     12368 KiB)
   3: 0x00000000fe380000..0x00000000fe3fffff (       512 KiB)
   4: 0x00000000f7030000..0x00000000f70affff (       512 KiB)
   5: 0x00000000ef9a0000..0x00000000ef9bffff (       128 KiB)
   6: 0x00000000f06f0000..0x00000000f16dffff (     16320 KiB)
session: pid=1954 index=54
 device: fdbd0000.rkvenc-core
 memory: 57 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|      54|  RKVENC|    3840|    2160|    h264|      60|      60|     vbr| 4000000|      30|   60.00|    high|
session iova range dump:
   0: 0x00000000ff020000..0x00000000ffff1fff (     16200 KiB)
   1: 0x00000000f7df0000..0x00000000f8a03fff (     12368 KiB)
   2: 0x00000000fc720000..0x00000000fd333fff (     12368 KiB)
   3: 0x00000000f7d70000..0x00000000f7deffff (       512 KiB)
   4: 0x00000000fc6a0000..0x00000000fc71ffff (       512 KiB)
   5: 0x00000000f4080000..0x00000000f409ffff (       128 KiB)
   6: 0x00000000eb120000..0x00000000ec10ffff (     16320 KiB)
session: pid=1954 index=55
 device: fdbd0000.rkvenc-core
 memory: 57 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|      55|  RKVENC|    3840|    2160|    h264|      60|      60|     vbr| 4000000|      30|   60.00|    high|

4路============
Every 1.0s: cat /proc/mpp_service/sessions-summary                                                                          rk3588-buildroot: Thu Jan  2 07:45:04 2025

session iova range dump:
   0: 0x00000000f9600000..0x00000000fa5d1fff (     16200 KiB)
   1: 0x00000000e55e0000..0x00000000e61f3fff (     12368 KiB)
   2: 0x00000000fe400000..0x00000000ff013fff (     12368 KiB)
   3: 0x00000000e5560000..0x00000000e55dffff (       512 KiB)
   4: 0x00000000fe380000..0x00000000fe3fffff (       512 KiB)
   5: 0x00000000f7cd0000..0x00000000f7ceffff (       128 KiB)
   6: 0x00000000dd550000..0x00000000de53ffff (     16320 KiB)
session: pid=1954 index=60
 device: fdbd0000.rkvenc-core
 memory: 57 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|      60|  RKVENC|    3840|    2160|    h264|      60|      60|     vbr| 4000000|      30|   35.19|    high|
session iova range dump:
   0: 0x00000000f9600000..0x00000000fa5d1fff (     16200 KiB)
   1: 0x00000000fb730000..0x00000000fc343fff (     12368 KiB)
   2: 0x00000000d5f00000..0x00000000d6b13fff (     12368 KiB)
   3: 0x00000000fb630000..0x00000000fb6affff (       512 KiB)
   4: 0x00000000d5e80000..0x00000000d5efffff (       512 KiB)
   5: 0x00000000f4080000..0x00000000f409ffff (       128 KiB)
   6: 0x00000000d2eb0000..0x00000000d3e9ffff (     16320 KiB)
session: pid=1954 index=61
 device: fdbd0000.rkvenc-core
 memory: 57 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
[... 8 lines cut ...]
   6: 0x00000000d0190000..0x00000000d0213fff (       528 KiB)
   7: 0x00000000ef9a0000..0x00000000ef9bffff (       128 KiB)
   8: 0x00000000c7a30000..0x00000000c8a1ffff (     16320 KiB)
session: pid=1954 index=62
 device: fdbd0000.rkvenc-core
 memory: 58 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|      62|  RKVENC|    3840|    2160|    h265|      60|      60|     vbr| 8000000|      30|   35.26|    main|
session iova range dump:
   0: 0x00000000ff020000..0x00000000ffff1fff (     16200 KiB)
   1: 0x00000000c9ae0000..0x00000000ca74ffff (     12736 KiB)
   2: 0x00000000cd8c0000..0x00000000ce52ffff (     12736 KiB)
   3: 0x00000000c9aa0000..0x00000000c9ad1fff (       200 KiB)
   4: 0x00000000cd880000..0x00000000cd8b1fff (       200 KiB)
   5: 0x00000000c9a10000..0x00000000c9a93fff (       528 KiB)
   6: 0x00000000cd7f0000..0x00000000cd873fff (       528 KiB)
   7: 0x00000000f7cf0000..0x00000000f7d0ffff (       128 KiB)
   8: 0x00000000c8a20000..0x00000000c9a0ffff (     16320 KiB)
session: pid=1954 index=63
 device: fdbd0000.rkvenc-core
 memory: 58 MiB
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|      63|  RKVENC|    3840|    2160|    h265|      60|      60|     vbr| 8000000|      30|   35.40|    main|

HermanChen, 2025-01-03:
3588 是两个核并行编码，总能力是 8K@30fps，可以支持4路的 4K@30fps 编码

HermanChen, 2025-01-03:
这个统计应该是有包含一些额外的操作，多路时间读同一块 buffer 对性能也有些影响

closed 2025-01-14 by HermanChen