rockchip-linux/mpp #758: BGR图像进行JPEG压缩失败

cz000, 2025-01-07:
在3588的板子上，基于ros2系统，使用最新的mpp库，进行bgr图像的单帧压缩。运行时没有报错，但运行结果不能通过其他软件解码。
有一些疑问：
1. bgr图像是1280*720，数据是1280*720*3个字节。但看mpi_enc_test.c的代码， 计算的hor_stride和ver_stride按16位对齐后是1280和720，计算frame的时候是按64对齐，得到的frame_size是要比 1280*720*3要小。那该如何输入bgr图像数据
2. 哪些编码器的config是必须设置的，哪些是不必要的？
3. 是用同一个buffer生成frame和packet，还是分别用不同的buffer

测试的代码大概如下：
`
#include <memory>
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/image.hpp>
#include <sensor_msgs/msg/compressed_image.hpp>
#include "rockchip/rk_mpi.h"
#include "rockchip/mpp_buffer.h"

// 对齐宏
#define MPP_ALIGN(x, a)    (((x)+(a)-1)&~((a)-1))
#define MPP_ALIGN_HOR 16   // 水平对齐要求
#define MPP_ALIGN_VER 16   // 垂直对齐要求

class ImageCompressionNode : public rclcpp::Node {
public:
    ImageCompressionNode() : Node("image_compression_node") {
        // 创建订阅者和发布者
        subscription_ = this->create_subscription<sensor_msgs::msg::Image>(
            "input_image", 10,
            std::bind(&ImageCompressionNode::image_callback, this, std::placeholders::_1));
        
        publisher_ = this->create_publisher<sensor_msgs::msg::CompressedImage>(
            "compressed_image", 10);

        // 初始化MPP
        ret = mpp_create(&ctx_, &mpi_);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "mpp_create failed");
            return;
        }

        // 配置编码器参数
        MppCodingType codec_type = MPP_VIDEO_CodingMJPEG;
        ret = mpp_init(ctx_, MPP_CTX_ENC, codec_type);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "mpp_init failed");
            return;
        }

        // 计算对齐的宽度和高度
        width_ = 1280;
        height_ = 720;
        hor_stride_ = MPP_ALIGN(width_, MPP_ALIGN_HOR);
        ver_stride_ = MPP_ALIGN(height_, MPP_ALIGN_VER);

        // 设置JPEG编码参数
        MppEncConfig config;
        memset(&config, 0, sizeof(config));
        config.width = width_;
        config.height = height_;
        config.hor_stride = hor_stride_;
        config.ver_stride = ver_stride_;
        config.format = MPP_FMT_BGR888;
        config.rc_mode = MPP_ENC_RC_MODE_FIXQP;
        config.quality = 80;

        ret = mpi_->control(ctx_, MPP_ENC_SET_CFG, &config);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "Failed to set encoder config");
            return;
        }

        // 设置输入格式
        MppFrameFormat fmt = MPP_FMT_BGR888;
        ret = mpi_->control(ctx_, MPP_ENC_SET_FORMAT, &fmt);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "Failed to set input format");
            return;
        }
    }

    ~ImageCompressionNode() {
        if (ctx_) {
            mpp_destroy(ctx_);
            ctx_ = nullptr;
            mpi_ = nullptr;
        }
    }

private:
    void image_callback(const sensor_msgs::msg::Image::SharedPtr msg) {
        if (msg->encoding != "bgr8") {
            RCLCPP_ERROR(this->get_logger(), "Unsupported image encoding: %s", msg->encoding.c_str());
            return;
        }

        // 准备MPP输入buffer
        MppFrame frame = nullptr;
        mpp_frame_init(&frame);
        mpp_frame_set_width(frame, width_);
        mpp_frame_set_height(frame, height_);
        mpp_frame_set_hor_stride(frame, hor_stride_);
        mpp_frame_set_ver_stride(frame, ver_stride_);
        mpp_frame_set_fmt(frame, MPP_FMT_BGR888);

        // 设置输入数据
        MppBuffer input_buf;
        size_t input_size = hor_stride_ * ver_stride_ * 3;  // BGR格式，每像素3字节
        ret = mpp_buffer_get(nullptr, &input_buf, input_size);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "Failed to get input buffer");
            mpp_frame_deinit(&frame);
            return;
        }
        
        // 复制图像数据到MPP buffer，需要考虑stride
        void* buf_ptr = mpp_buffer_get_ptr(input_buf);
        uint8_t* src_ptr = msg->data.data();
        uint8_t* dst_ptr = (uint8_t*)buf_ptr;
        
        // 逐行复制，处理stride
        for (uint32_t i = 0; i < height_; i++) {
            memcpy(dst_ptr + i * hor_stride_ * 3, 
                   src_ptr + i * msg->step, 
                   width_ * 3);
        }

        mpp_frame_set_buffer(frame, input_buf);

        // 编码
        MppPacket packet = nullptr;
        ret = mpi_->encode_put_frame(ctx_, frame);
        if (ret) {
            RCLCPP_ERROR(this->get_logger(), "Failed to put frame");
            mpp_buffer_put(input_buf);
            mpp_frame_deinit(&frame);
            return;
        }

        ret = mpi_->encode_get_packet(ctx_, &packet);
        if (ret || !packet) {
            RCLCPP_ERROR(this->get_logger(), "Failed to get packet");
            mpp_buffer_put(input_buf);
            mpp_frame_deinit(&frame);
            return;
        }

        // 获取压缩后的数据
        void* compressed_data = mpp_packet_get_pos(packet);
        size_t compressed_size = mpp_packet_get_length(packet);

        // 创建压缩图像消息
        auto compressed_msg = std::make_shared<sensor_msgs::msg::CompressedImage>();
        compressed_msg->header = msg->header;
        compressed_msg->format = "jpeg";
        compressed_msg->data.resize(compressed_size);
        memcpy(compressed_msg->data.data(), compressed_data, compressed_size);

        // 发布压缩后的图像
        publisher_->publish(*compressed_msg);

        // 清理资源
        mpp_packet_deinit(&packet);
        mpp_buffer_put(input_buf);
        mpp_frame_deinit(&frame);
    }

    rclcpp::Subscription<sensor_msgs::msg::Image>::SharedPtr subscription_;
    rclcpp::Publisher<sensor_msgs::msg::CompressedImage>::SharedPtr publisher_;
    MppCtx ctx_ = nullptr;
    MppApi* mpi_ = nullptr;
    RK_S32 ret = 0;
    
    // 图像参数
    uint32_t width_;
    uint32_t height_;
    uint32_t hor_stride_;    // 水平步幅（对齐后的宽度）
    uint32_t ver_stride_;    // 垂直步幅（对齐后的高度）
};

int main(int argc, char** argv) {
    rclcpp::init(argc, argv);
    auto node = std::make_shared<ImageCompressionNode>();
    rclcpp::spin(node);
    rclcpp::shutdown();
    return 0;
}
`

cz000, 2025-01-07:
@HermanChen

HermanChen, 2025-01-07:
参考下 mpi_enc_test 这里的水平 stride 要乘 3，stride 是两行之间的 byte 距离

cz000, 2025-01-07:
> 参考下 mpi_enc_test 这里的水平 stride 要乘 3，stride 是两行之间的 byte 距离

上面的代码里有处理stride，hor_stride乘3了

HermanChen, 2025-01-14:
用原始的 mpi_enc_test 对比下呢？
等等，这里用的是结构体的 config？mpp 已经很久不使用结构体来配置了啊
感觉这里用结构体来配置都会失败的吧？

cz000, 2025-02-11:
[image]
@HermanChen 给您再提供下背景，我是在另一个进程里，使用
_gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=BGR_获取的数据源，通过进程间通信，希望在业务进程中用mpp将每帧进行jpeg压缩。

使用_gst-launch-1.0 filesink将数据源保存为文件bgr的文件，然后使用mpi_enc_test进行测试，测试结果如图，avi文件无法打开。同样的bgr文件，编码换成h264，输出的文件就可以打开，是我的使用方式不对吗？

`vepu_common: get_vepu_pixel_stride invalid fmt 65543`
`vepu_common: get_vepu_fmt unsupport frame format 10007`

HermanChen, 2025-02-11:
等下，mpi_enc_test 的输出祼码流，不支持 avi 这种打包，输出要是 test.h264 或者 test.jpg 这种

cz000, 2025-02-11:
> 等下，mpi_enc_test 的输出祼码流，不支持 avi 这种打包，输出要是 test.h264 或者 test.jpg 这种

默认不设置 -n 参数，就会编码1帧，我试过编码1帧，图像打不开

`./mpi_enc_test -i output_bgr.bgr -w 1280 -h 720 -f 65543 -t 8 -n 1 -o ret.jpg`

HermanChen, 2025-02-11:
log 有没？
直接 mpi_enc_test -w 1280 -h 720 -n -1 -o test.jpg 看下是不是正常

cz000, 2025-02-11:
> log 有没？ 直接 mpi_enc_test -w 1280 -h 720 -n -1 -o test.jpg 看下是不是正常

[image]

cz000, 2025-02-11:
> log 看，直接编码是正常的，要确认下 bgr buffer 大小是多大，是不是读文件编码的时候给崩溃了，加点 log 跟进去看看

我没明白-_-!
编码出的ret.jpg是不能打开的，直接编码还是正常吗？

"是不是读文件编码的时候给崩溃了"
您的意思是读文件有异常，执行 mpi->encode_put_frame() 编码本身没问题?

HermanChen, 2025-02-11:
意思是编码器本身没有问题，不加 -i 的情况下，跑的 test.jpg 是正常的话，说明编码器本身是好的
问题应该出在输入和配置

cz000, 2025-02-11:
> 意思是编码器本身没有问题，不加 -i 的情况下，跑的 test.jpg 是正常的话，说明编码器本身是好的 问题应该出在输入和配置

额，我以为你敲错了，没有-i 竟然也可以运行，得到的图片可打开，彩色条形图。
`./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg` 
```
root@test# ./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg
mpi_enc_utils: cmd parse result:
mpi_enc_utils: input  file name: (null)
mpi_enc_utils: output file name: test.jpg
mpi_enc_utils: width      : 1280
mpi_enc_utils: height     : 720
mpi_enc_utils: format     : 0
mpi_enc_utils: type       : 8
mpi_enc_test: ./mpi_enc_test start
mpp_info: mpp version: unknown mpp version for missing VCS info
mpi_enc_test: 0x7f7c001540 encoder test start w 1280 h 720 type 8
mpp_platform: client 12 driver is not ready!
mpp_platform: client 18 driver is not ready!
mpp_enc: MPP_ENC_SET_RC_CFG bps 3456000 [0 : 0] fps [30:30] gop 60
mpi_enc_test: chn 0 encoded frame 0    size 65913   qp 80
mpi_enc_test: chn 0 encode 1 frames time 13 ms delay   8 ms fps 75.11 bps 15819120
mpi_enc_test: ./mpi_enc_test average frame rate 75.11
```
对比
./mpi_enc_test -i output_bgr.bgr -w 1280 -h 720 -f 65543 -t 8 -n 1 -o ret.jpg
没有出现这两行错误
```
vepu_common: get_vepu_pixel_stride invalid fmt 65543
vepu_common: get_vepu_fmt unsupport frame format 10007
```

@HermanChen  那我再加log跟进行去看吧，感谢👍

cz000, 2025-02-12:
> log 看，直接编码是正常的，要确认下 bgr buffer 大小是多大，是不是读文件编码的时候给崩溃了，加点 log 跟进去看看

数据的读取应该是没有问题的。
对比
```
./mpi_enc_test -i output_bgr.bgr -w 1280 -h 720 -f 65543 -t 8 -n 1 -o ret.jpg
```
和
```
./mpi_enc_test -i output_bgr.bgr -w 1280 -h 720 -f 65543 -t 7 -n 1 -o ret.h264
```
使用同样的数据源，编码为h264时是不会报错的，数据源读取都是一样的，都是使用
```
read_image(buf, p->fp_input, p->width, p->height,p->hor_stride, p->ver_stride, p->fmt)
```

如果问题是出在配置上， 我不知道该如何配置。毕竟mpi_enc_test本身就是demo

cz000, 2025-02-12:
另外，使用yuv420p的数据源是可以正常编码为jpeg的
```
./mpi_enc_test -i output_video.yuv -w 1280 -h 720 -f 4 -t 8 -n 1 -o ret.jpg
```
YUV数据的获取和前面的BGR一样，都是使用gst-aunch-1.0，保存为文件。
```
gst-launch-1.0 v4l2src device=/dev/video1 ! video/x-raw,format=I420
```

@HermanChen

cz000, 2025-02-14:
其他同事也复现了这个问题。
这个应该是个BUG，**使用mpp对BGR的数据进行Jpeg压缩**，版本1.0.8即可复现。
@HermanChen

HermanChen, 2025-02-14:
等下，3588 上的 jpeg 编码器只支持 BGRX 32bit 的 RGB，不支持 BGR 这种 24bit 的 RGB
他的 jpeg 编码器还是 vepu2 这种比较旧的……

cz000, 2025-02-14:
> 等下，3588 上的 jpeg 编码器只支持 BGRX 32bit 的 RGB，不支持 BGR 这种 24bit 的 RGB 他的 jpeg 编码器还是 vepu2 这种比较旧的……

好的，我再测试下

cz000, 2025-02-14:
> 等下，3588 上的 jpeg 编码器只支持 BGRX 32bit 的 RGB，不支持 BGR 这种 24bit 的 RGB 他的 jpeg 编码器还是 vepu2 这种比较旧的……

测试可行，使用Bgra的图像数据是可以进行jpeg压缩的。可以关闭了，感谢 @HermanChen

cz000, 2025-05-27:
我用mpi_enc_test先生成一个jpg文件
` ./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg `

然后再用mpi_dec_test解码
`./mpi_dec_test -i test.jpg -w 1280 -h 720 -t 8 -f 4 -o test.yuv`
但是得到的test.yuv打开颜色不对
@HermanChen

HermanChen, 2025-05-28:
3588 的 jpeg 解码器是新的，不带转 RGB 格式的功能了