rockchip-linux/mpp #761: AV1d FBC decoding crashes the system on RK3588

nyanmisaka, 2025-01-09:
AV1d **FBC** decoding crashes the system on RK3588, but it works on RK3576. No logs were recorded.

Regression occurred in the following commits:

- [feat[mpp_sys_cfg]: Add mpp_sys_cfg function](https://github.com/rockchip-linux/mpp/commit/78ad332e7e62ba6e62d68c327d5a6ea4f6b19f04)
- [feat[mpp_buf_slots]: Add coding attribute to buf slots](https://github.com/rockchip-linux/mpp/commit/87f23da2e6ff047e2ce9c5e24f45fe048dc6abb4)
- [feat[mpp_sys_cfg]: Support sys_cfg buffer alignment](https://github.com/rockchip-linux/mpp/commit/e52ec5e1e82209fff12737d6b61b4b7a2e9c7cdf)
- [feat[mpp_sys_cfg]: Add raster/tile/fbc buffer alignment](https://github.com/rockchip-linux/mpp/commit/5931c416d80cce9ade083168dfca4bb1dbb846e7)

wenyue7, 2025-01-09:
Can the issue be reproduced with specific media sources?
I tested it on the RK3588 platform without any anomalies.

wenyue7, 2025-01-09:
Hello, the RK3588 platform AV1 doesn't seem to support FBC.

nyanmisaka, 2025-01-09:
> Hello, the RK3588 platform AV1 doesn't seem to support FBC.

It does support that, I've been using it for a few years now. So this is a regression.

wenyue7, 2025-01-09:
Please help confirm the MPP and kernel version information:

mpp:
strings libmpp.so | grep author

kernel:
cat /proc/mpp_service/version
uname -a

nyanmisaka, 2025-01-09:
@wenyue7 Here's the required info

I'm on the latest mpp commit 9bcf3672c93b36817edc35284aaa227e9727b074
```
strings librockchip_mpp.so.0 | grep author

9bcf3672 author: Yandong Lin   2025-01-07 feat[rk_mpi_cmd]: Merge cmds from mpp_interface  (HEAD -> upstream/develop)
e447e076 author: Yandong Lin   2025-01-07 feat[mpp_enc_cfg]: Merge enc cfgs from mpp_interface
e1621412 author: Yandong Lin   2025-01-07 fix[mpp_venc_kcfg]: Get objdef at runtime
7132f62e author: Rimon Xu      2025-01-07 fix[jpegd]: Avoid buffer overrun
330dc3d4 author: Hongjin Li    2025-01-07 fix[sys_cfg/buf_slot]: fix fbc yuv444sp buf calculation issue
07bef128 author: Hongjin Li    2025-01-03 feat[mpp_sys_cfg_st]: Provide packaging for use on products
5931c416 author: Hongjin Li    2025-01-03 feat[mpp_sys_cfg]: Add raster/tile/fbc buffer alignment
e52ec5e1 author: Hongjin Li    2025-01-03 feat[mpp_sys_cfg]: Support sys_cfg buffer alignment
fa095373 author: Yandong Lin   2025-01-03 feat[kmpp_obj]: Add kmpp_obj_get_hnd func
47fbbe74 author: Yandong Lin   2025-01-03 feat[mpp_venc_kcfg]: Add mpp_venc_kcfg module
9bcf3672 author: Yandong Lin   2025-01-07 feat[rk_mpi_cmd]: Merge cmds from mpp_interface
```

And Linux 6.1 kernel (rkr 4.1)
```
0308f07933f7 author: Yandong Lin 2024-10-14 video: rockchip: mpp: iep2: fix err log false print

Linux 6.1.84-vendor-rk35xx
```

FYI you just need to revert the 4 commits I mentioned and everything will be back to normal.

wenyue7, 2025-01-10:
Hi,
可以测试一下附件中的两个补丁，看能否解决问题
[0001-fix-fbc-Fix-RK3588-av1-FBC-usage-issue.patch](https://github.com/user-attachments/files/18369684/0001-fix-fbc-Fix-RK3588-av1-FBC-usage-issue.patch)
[0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch](https://github.com/user-attachments/files/18369685/0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch)

nyanmisaka, 2025-01-10:
@wenyue7 Thanks for the patch, it solves the problem on rk3588.

BTW there is a fix about `MPP_FMT_YUV422SP_10BIT / DRM_FORMAT_NV20` + FBC use case, it only occurs in h264 high 422 video. But my test cases catch it when using RGA to convert it to other formats.

It fixes this error:
```
[62388.940750] rga_mm: Only get buffer 4030464 byte from dma_buf_ptr = 0xffff0001045acc00, but current image required 4177920 byte
[62388.940786] rga_common: memory param: w = 1920, h = 1088, f = YCbCr422SP10B(0x22), size = 0
[62388.940796] rga_mm: rga_mm_map_buffer map dma_buf error!
[62388.940801] rga_mm: job buffer map failed!
[62388.940805] rga_mm: src channel map job buffer failed!
[62388.940812] rga_mm: failed to map buffer
[62388.940817] rga_job: rga_job_commit: failed to map job info
[62388.940954] rga_job: request[197944] task[0] job_commit failed.
[62388.940968] rga_job: rga request[197944] commit failed!
[62388.940983] rga: request[197944] submit failed!
[62402.190014] rga_mm: Only get buffer 4030464 byte from dma_buf_ptr = 0xffff0001ad5a3400, but current image required 4177920 byte
```

[0007-fix-mpp_sys_cfg-Support-FBC-buffer-size-calc-for-NV20.patch](https://github.com/user-attachments/files/18371199/0007-fix-mpp_sys_cfg-Support-FBC-buffer-size-calc-for-NV20.patch)

```diff
diff --git a/mpp/base/mpp_buf_slot.cpp b/mpp/base/mpp_buf_slot.cpp
index cdb936f0..a0dae282 100644
--- a/mpp/base/mpp_buf_slot.cpp
+++ b/mpp/base/mpp_buf_slot.cpp
@@ -318,7 +318,8 @@ static void prepare_info_set_legacy(MppBufSlotsImpl *impl, MppFrame frame,
     }
 
     switch (fmt & MPP_FRAME_FMT_MASK) {
-    case MPP_FMT_YUV420SP_10BIT: {
+    case MPP_FMT_YUV420SP_10BIT:
+    case MPP_FMT_YUV422SP_10BIT: {
         hor_stride_pixel = hal_hor_stride * 8 / 10;
     } break;
     case MPP_FMT_YUV422_YVYU:
@@ -344,6 +345,9 @@ static void prepare_info_set_legacy(MppBufSlotsImpl *impl, MppFrame frame,
         case MPP_FMT_YUV420SP_10BIT : {
             size = get_afbc_min_size(hor_stride_pixel, hal_ver_stride, 15);
         } break;
+        case MPP_FMT_YUV422SP_10BIT : {
+            size = get_afbc_min_size(hor_stride_pixel, hal_ver_stride, 20);
+        } break;
         case MPP_FMT_YUV420SP : {
             size = get_afbc_min_size(hor_stride_pixel, hal_ver_stride, 12);
         } break;
diff --git a/mpp/base/mpp_sys_cfg.cpp b/mpp/base/mpp_sys_cfg.cpp
index e0df3054..27d1665d 100644
--- a/mpp/base/mpp_sys_cfg.cpp
+++ b/mpp/base/mpp_sys_cfg.cpp
@@ -352,6 +352,9 @@ MPP_RET mpp_sys_dec_buf_chk_proc(MppSysDecBufChkCfg *cfg)
         case MPP_FMT_YUV420SP_10BIT : {
             size_total = get_afbc_min_size(aligned_pixel, aligned_height, 15);
         } break;
+        case MPP_FMT_YUV422SP_10BIT : {
+            size_total = get_afbc_min_size(aligned_pixel, aligned_height, 20);
+        } break;
         case MPP_FMT_YUV420SP : {
             size_total = get_afbc_min_size(aligned_pixel, aligned_height, 12);
         } break;
```

wenyue7, 2025-01-10:
Alright, thank you for your patch.

closed 2025-01-10 by nyanmisaka

wenyue7, 2025-01-10:
We will submit it internally directly.

nyanmisaka, 2025-02-12:
@wenyue7 Just want to let you know that this one commit https://github.com/rockchip-linux/mpp/commit/80d7377a0bab4d963f34f8d4a82813de6b551a6e alone is not enough to fix the av1d fbc crash.

The second patch is still needed, which is not merged yet.
[0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch](https://github.com/user-attachments/files/18369685/0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch)

HermanChen, 2025-02-13:
> [@wenyue7](https://github.com/wenyue7) Just want to let you know that this one commit [80d7377](https://github.com/rockchip-linux/mpp/commit/80d7377a0bab4d963f34f8d4a82813de6b551a6e) alone is not enough to fix the av1d fbc crash.
> 
> The second patch is still needed, which is not merged yet. [0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch](https://github.com/user-attachments/files/18369685/0002-fix-mpp_sys_cfg-Fix-abnormal-stride-calculation.patch)

Done