rockchip-linux/mpp #763: CBR encoding not working perfectly

smzahraee, 2025-01-13:
If I set the rc_mode to CBR, the minimum bitrate is not maintained during scene changes, and it drops suddenly.
for example for the following settings 
```
mpp_enc_cfg_set_s32(cfg, "rc:bps_max",8000000);
mpp_enc_cfg_set_s32(cfg, "rc:bps_target", 7000000);
mpp_enc_cfg_set_s32(cfg, "rc:bps_min", 6000000);
```
I got 
[image]

HermanChen, 2025-01-14:
How about the min / max qp config？

smzahraee, 2025-01-14:
I need to define the minimum and maximum QP for each bitrate manually, which requires significant effort since there are no established rules for doing so. Additionally, the specified minimum and maximum QP values are not consistently maintained in certain scenarios.

HermanChen, 2025-01-14:
emm, the default value of min / max qp may not good enough for all the cases.
Try set all min to 8 and max to 48.
Can you provide some sample stream for us to check the issue?

smzahraee, 2025-01-15:
If there are no changes, the CBR stream stays within the defined range, but during sudden changes in video, it drops as mentioned.

here is my detailed configs:

```
rc_mode:1 bps:7000000 bps_max:7600000 bps_min:6800000
qp_init:26 qp_max:48 qp_min:4 qp_max_i:48 qp_min_i:4 qp_ip:1
```
[image]

smzahraee, 2025-01-15:
for the following settings, the peaks are weird
```
mode:1 bps:1312500 bps_max:1425000 bps_min:1275000
qp_init:26 qp_max:48 qp_min:4 qp_max_i:48 qp_min_i:4 qp_ip:1

```
[image]

smzahraee, 2025-01-15:
I think the main problem may be is the calculation of the following rates in the [rc_model_v2.c](https://github.com/rockchip-linux/mpp/blob/9bcf3672c93b36817edc35284aaa227e9727b074/mpp/codec/rc/rc_model_v2.c#L623)
```
r_bits|    t_bits|   ins_bps| p_ins_bps|target_bps|    watl
7304      23411     386824     386695    1312500        0
```
as you can see there is a gap between ```p_ins_bps``` and ```target_bps``` but the algorithm can not adapt it self to the ```target_bps```

HermanChen, 2025-01-22:
Can you show me the input yuv source file? Before scene change and after scene change are both needed.