rockchip-linux/mpp #764: mpp是否可以将bmp格式编码为jpeg

zhaoyuRobotics, 2025-01-14:
# ./mpi_enc_test -w 2048 -h 1536 -t 8 -i ./test.bmp -o ./test.jpg
mpp[19612]: mpi_enc_utils: cmd parse result:
mpp[19612]: mpi_enc_utils: input  file name: ./test.bmp
mpp[19612]: mpi_enc_utils: output file name: ./test.jpg
mpp[19612]: mpi_enc_utils: width      : 2048
mpp[19612]: mpi_enc_utils: height     : 1536
mpp[19612]: mpi_enc_utils: format     : 0
mpp[19612]: mpi_enc_utils: type       : 8
mpp[19612]: mpi_enc_test: ./mpi_enc_test start
mpp[19612]: mpi_enc_test: jpege default encode only one frame. Use -n [num] for rc case
mpp[19612]: mpp_info: mpp version: 331ee68f author: nyanmisaka    2024-12-11 fix[mpp]: Fix input_task_count for async enc
mpp[19612]: mpi_enc_test: 0xffff900019e0 encoder test start w 2048 h 1536 type 8
mpp[19612]: mpp_platform: client 1 driver is not ready!
mpp[19612]: mpp_platform: client 3 driver is not ready!
mpp[19612]: mpp_platform: client 12 driver is not ready!
mpp[19612]: mpp_platform: client 18 driver is not ready!
mpp[19612]: mpp_enc: MPP_ENC_SET_RC_CFG bps 11796480 [0 : 0] fps [30:30] gop 60
mpp[19612]: mpi_enc_test: chn 0 found last frame. feof 1
mpp[19612]: mpi_enc_test: chn 0 encoded frame 0    size 0      
mpp[19612]: mpi_enc_test: chn 0 found last packet
mpp[19612]: mpi_enc_test: chn 0 encode 1 frames time 5 ms delay   0 ms fps 170.82 bps 0
mpp[19612]: mpi_enc_test: ./mpi_enc_test average frame rate 170.82

zhaoyuRobotics, 2025-01-14:
[test.zip](https://github.com/user-attachments/files/18408766/test.zip)

HermanChen, 2025-01-14:
目前只支持图像像素数据，不支持这种带格式头的

zhaoyuRobotics, 2025-01-15:
图像像素数据指的是？可否细说下。如果最好有相关转换示例的话会更好的帮助我理解，谢谢。

123lj-z, 2025-01-16:
你好，请问我运行mpi_enc_test时，输入为一段yuv视频时，能编码出jpg的图片（但是图片色彩不对），但是只输入一帧yuv时，输出的jpg图片错误。
[image]

HermanChen, 2025-01-16:
看起来是高度上的对齐问题

HermanChen, 2025-01-16:
jpeg 高度 16 对齐看下

closed 2025-02-20 by HermanChen