rockchip-linux/mpp #768: mpp支持bgr888的4k 60fps编码吗

huochen1, 2025-01-17:
我测试了一下好像有点丢帧

https://github.com/user-attachments/assets/48ba1807-3bb1-4fc8-a091-2f70befd1654

huochen1, 2025-01-18:
同时想请问一下可以这样子使用这个demo的吗？，我发现这样子出来的颜色以及数据都是错误的。./mpi_enc_test -i /dev/video20 -o /data/test_bgr3.h264 -w 3840 -h 2160  -t 7  ，如果要实现4k 60fps的话是不是需要零拷贝以及4k对齐才行的。

HermanChen, 2025-01-20:
要零拷贝比较好，用 cpu 拷贝图像数据是来不及的

huochen1, 2025-01-21:
> 要零拷贝比较好，用 cpu 拷贝图像数据是来不及的

谢谢您的建议，现在不会卡顿，但是视频明显加速了，我设置固定的输入输出都为60fps了，4k的图像。但是看到fps_calc好像一直为30左右。

[image]
设置了下面的这些参数也不会有改善，这是什么情况呢？

[image]

huochen1, 2025-01-21:
虽然保存的视频文件是60fps的，但是感觉是两倍速的播放速度在播放

HermanChen, 2025-01-21:
编码帧率要和实际采集帧率匹配

huochen1, 2025-01-21:
> 要零拷贝比较好，用 cpu 拷贝图像数据是来不及的

再请问您一下，就是使用mpp的例子./mpi_enc_test -i /dev/video20 -o /data/test_bgr3.h264 -w 3840 -h 2160 -t 7  直接这样子运行是不可以没办法零拷贝的是吧，我看到文档里有说零拷贝的事情

[image]
因此，我从HDMI IN 获取到的数据需要转化要变成这种类型的数据才比较好编码

HermanChen, 2025-01-21:
./mpi_enc_test -i /dev/video20 -o /data/test_bgr3.h264 -w 3840 -h 2160 -t 7
这个就是输入图像零拷贝了

huochen1, 2025-01-21:
> ./mpi_enc_test -i /dev/video20 -o /data/test_bgr3.h264 -w 3840 -h 2160 -t 7 这个就是输入图像零拷贝了

您好，还有一个小问题，就是现在我想看看打印的信息，之前我记得是执行这个命令的tail -f /var/log/syslog。但是现在我最新下载了这个github上的然后重新编译，cd mpp-develop/build/linux/aarch64/   ./make-Makefiles.bash    sudo make   sudo make install 

[image]
发现找不到这个文件了，现在的是变了打印的方式了吗？我想查看编码的一些输入指令的命令

HermanChen, 2025-01-22:
这个文件是内核的，不是 mpp 的，mpp 只是往 syslog 里写 log
用 export mpp_syslog_perror=1 打开打印的环境变量，再跑测试用例

huochen1, 2025-01-22:
根据您的提示打开这个log打印之后使用该命令./mpi_enc_test -i /dev/video0 -o test_bgr3.h264 -w 3840 -h 2160 -t 7 -f 65543 -fps 60 -n  2400
发现这个平均帧率没有60fps

[image]

[image]

但是使用ffmpeg查看保存的文件又是有60fps的

[image]
这个好像达不到满帧率的保存，我需要怎么去优化呢？

> 这个文件是内核的，不是 mpp 的，mpp 只是往 syslog 里写 log 用 export mpp_syslog_perror=1 打开打印的环境变量，再跑测试用例

huochen1, 2025-01-22:
如果采用相同的参数，但是使用的是mpi_enc_mt_test，程序反而卡主不会动了，不会往下编码了，执行的命令如下：./mpi_enc_mt_test -i /dev/video0 -o test_bgr3.h264 -w 3840 -h 2160 -t 7 -f 65543 -fps 60 -n  2400

[image]

HermanChen, 2025-01-22:
mt_test 还没有适配过 camera_source，分配和释放的流程可能会有问题
这读写文件的时间就很长了，会很大程度上影响性能
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug
看下串口打印的内核驱动运行时间

HermanChen, 2025-01-22:
抖动一下是 I 帧？

huochen1, 2025-01-22:
> 抖动一下是 I 帧？

额额，这个编码好的视频放在专门支持4k播放的PC电脑上得到很大的改善，可能一分钟就出现一小会的那种卡顿现象，可能是电脑的播放问题

HermanChen, 2025-01-22:
好的，问题解决请关闭问题

closed 2025-01-22 by huochen1

huochen1, 2025-01-23:
> mt_test 还没有适配过 camera_source，分配和释放的流程可能会有问题 这读写文件的时间就很长了，会很大程度上影响性能 echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug 看下串口打印的内核驱动运行时间

再询问您一下，现在就是大概需要两分钟的时间才能到59.6左右，有什么方法可以加快这个编码效率吗，就是我想在一分钟左右就到59.6左右这种有办法吗？

HermanChen, 2025-01-23:
其实在刚开始就已经满效率了，只是统计代码加算了不少的其他时间

huochen1, 2025-01-23:
好的谢谢，我可以这么理解吗，就是我们在统计帧数的时候其实花了不少的时间但是实际上的话已经差不多在一分半的时间内达到59.6fps左右了

HermanChen, 2025-01-23:
前几帧应该就已经满足要求了

HermanChen, 2025-01-23:
统计的代码因为统计了比较长的读写文件时间，需要一分钟多才能消化这些额外的开销

huochen1, 2025-01-23:
好的谢谢

huochen1, 2025-03-08:
额，再请问一下，就是咋们的这个不是支持8k 30fps的编码嘛，这个需要怎么验证呢，从HDMI IN进来的数据好像只支持4k 60fps，8k的视频进不来，然后我又打算使用ffmpeg生成8k 30fps的裸数据来进行编码也不行，很慢ffmpeg生成的大概一秒一帧那样而且20秒就57g了，编码的时候也是平均只有1秒2帧，需要怎么测试比较好呢，就是测试8k30fps的

huochen1, 2025-03-10:
那应该使用什么样的测试源来进行测试呢？额，使用本地的裸数据文件8k 30的来进行测试吗

   
用 mpi_enc_mt_test 来验证

 
用 mpi_enc_mt_test 来验证

HermanChen, 2025-03-10:
不加 -i，让 cpu 自己花图来测试
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug 看硬件时间打印

huochen1, 2025-03-10:
[image]
好像还是不行，我使用的是./mpi_enc_mt_test  -o test_bgr3.h264 -w 7680 -h 4320 -t 7 -f 65543 -fps 30 -n  100
这个命令

HermanChen, 2025-03-11:
去掉 -o ，不要写码流，看硬件时间比较好，这个时间是上层软件看到的整体时间，并不是编码器本身的性能
mpi_enc_mt_test -w 7680 -h 4320 -t 7 -v qf

huochen1, 2025-03-11:
mpi_enc_mt_test -w 7680 -h 4320 -t 7  -o test_bgr3.h264 -fps 15 -n 3600
我还发现一个问题，使用这个命令保存的视频显示的帧数是30fps
如果使用mpi_enc_mt_test -w 7680 -h 4320 -t 7  -o test_bgr3.h264 -fps 30 -n 1200
这个命令保存的视频是60fps

平均帧数都是这个

HermanChen, 2025-03-11:
是不是 vui 里的信息写错了？

huochen1, 2025-03-11:
额，不太懂这个vui是什么了，上网也没看到有相关介绍，然后我使用过ffmpeg+mpp以及gstreamer+mpp的时候也都是会出现这种情况的，除非就是输入的帧率刚刚好满足编码的设置，不然的话就是编码出来的视频都是会快的。不过使用命令查看cat /proc/mpp_service/rkvenc-core0/sessions-info

[image]

帧数缺少in是15out也是15，测试命令就是这个mpi_enc_mt_test -w 7680 -h 4320 -t 7  -o test_bgr3.h264 -fps 15 -n 3600

[image]

查看的视频帧率确实是30，使用ffmpeg来查看视频的属性的ffprobe -v error -select_streams v:0 -show_entries stream=pix_fmt,width,height,codec_name,r_frame_rate -of default=noprint_wrappers=1:nokey=1 test_bgr3.h264
使用这个命令的话就是会变成60fps了
mpi_enc_mt_test -w 7680 -h 4320 -t 7  -o test_bgr3.h264 -fps 30 -n 2400

[image]

[image]