rockchip-linux/mpp #770: mpi->reset(ctx) 卡死

denghengli, 2025-01-21:
1、软件版本：
mpp：1.0.7
kernel：5.10
处理器：rk3568

2、应用场景
（1）一个nvrd程序 从摄像头拉取视频流进行解码-->编码-->rtsp推流，同时处理2-3路摄像头
（2）一个界面程序（QT）播放某一通道摄像头的实时视频，通道可以切换，在切换时会调用reset复位解码器然后重新再打开解码器 以解码下一通道的视频

3、问题描述
在调用reset复位解码器时会出现卡死在这个接口中，卡死前后内核会报如下信息：
[ 1291.865631] mpp_rkvdec2 fdf80200.rkvdec: reset done
[ 1291.865705] mpp_rkvdec2 fdf80200.rkvdec: resend task 62867
[ 1291.865753] mpp_rkvdec2 fdf80200.rkvdec: resend task 62868
[ 1291.865773] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865788] mpp_rkvdec2 fdf80200.rkvdec: resend task 62869
[ 1291.865803] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865815] mpp_rkvdec2 fdf80200.rkvdec: resend task 62870
[ 1291.865827] mpp_rkvdec2 fdf80200.rkvdec: resend task 62871
[ 1291.865840] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865852] mpp_rkvdec2 fdf80200.rkvdec: resend task 62872
[ 1291.865865] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865877] mpp_rkvdec2 fdf80200.rkvdec: resend task 62873
[ 1291.865889] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865901] mpp_rkvdec2 fdf80200.rkvdec: resend task 62874
[ 1291.865916] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1291.865931] mpp_rkvdec2 fdf80200.rkvdec: resend task 62875
[ 1292.369591] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 62875 state 0x14f timeout, cnt 9
[ 1292.369650] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 0 state 0x4c timeout, cnt 10
[ 1292.369662] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 62874 state 0x14f timeout, cnt 11
[ 1292.369676] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 0 state 0x4c timeout, cnt 12
[ 1292.369686] mpp_rkvdec2 fdf80200.rkvdec: session 14 task 62873 state 0x14f timeout, cnt 13
[ 1292.369697] mpp_rkvdec2 fdf80200.rkvdec: session 14 task 0 state 0x4c timeout, cnt 14
[ 1292.369708] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 62872 state 0x14f timeout, cnt 15
[ 1292.369718] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 0 state 0x4c timeout, cnt 16
[ 1292.369729] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62871 state 0x14f timeout, cnt 17
[ 1292.369745] mpp_rkvdec2 fdf80200.rkvdec: session 16 task 62870 state 0x14f timeout, cnt 18
[ 1292.369760] mpp_rkvdec2 fdf80200.rkvdec: session 16 task 0 state 0x4c timeout, cnt 19
[ 1292.369774] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 62869 state 0x14f timeout, cnt 20
[ 1292.369788] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 0 state 0x4c timeout, cnt 21
[ 1292.369802] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62868 state 0x14f timeout, cnt 22
[ 1292.369815] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62867 state 0x14f timeout, cnt 23
[ 1292.369824] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62867 timeout 1 abort 0 force_dequeue 0
[ 1292.369830] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[ 1292.371052] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
[ 1292.371244] mpp_rkvdec2 fdf80200.rkvdec: reset done
[ 1292.371336] mpp_rkvdec2 fdf80200.rkvdec: resend task 62868
[ 1292.371415] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371447] mpp_rkvdec2 fdf80200.rkvdec: resend task 62869
[ 1292.371476] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371503] mpp_rkvdec2 fdf80200.rkvdec: resend task 62870
[ 1292.371530] mpp_rkvdec2 fdf80200.rkvdec: resend task 62871
[ 1292.371562] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371591] mpp_rkvdec2 fdf80200.rkvdec: resend task 62872
[ 1292.371618] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371647] mpp_rkvdec2 fdf80200.rkvdec: resend task 62873
[ 1292.371675] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371702] mpp_rkvdec2 fdf80200.rkvdec: resend task 62874
[ 1292.371726] mpp_rkvdec2 fdf80200.rkvdec: resend task 0
[ 1292.371752] mpp_rkvdec2 fdf80200.rkvdec: resend task 62875
[ 1292.481210] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[ 1292.876198] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 62876 state 0x14f timeout, cnt 24
[ 1292.876251] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 0 state 0x4c timeout, cnt 25
[ 1292.876262] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 62875 state 0x14f timeout, cnt 26
[ 1292.876272] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 0 state 0x4c timeout, cnt 27
[ 1292.876281] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 62874 state 0x14f timeout, cnt 28
[ 1292.876289] mpp_rkvdec2 fdf80200.rkvdec: session 15 task 0 state 0x4c timeout, cnt 29
[ 1292.876298] mpp_rkvdec2 fdf80200.rkvdec: session 14 task 62873 state 0x14f timeout, cnt 30
[ 1292.876306] mpp_rkvdec2 fdf80200.rkvdec: session 14 task 0 state 0x4c timeout, cnt 31
[ 1292.876314] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 62872 state 0x14f timeout, cnt 32
[ 1292.876322] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 0 state 0x4c timeout, cnt 33
[ 1292.876330] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62871 state 0x14f timeout, cnt 34
[ 1292.876338] mpp_rkvdec2 fdf80200.rkvdec: session 16 task 62870 state 0x14f timeout, cnt 35
[ 1292.876346] mpp_rkvdec2 fdf80200.rkvdec: session 16 task 0 state 0x4c timeout, cnt 36
[ 1292.876354] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 62869 state 0x14f timeout, cnt 37
[ 1292.876362] mpp_rkvdec2 fdf80200.rkvdec: session 60 task 0 state 0x4c timeout, cnt 38
[ 1292.876369] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62868 state 0x14f timeout, cnt 39
[ 1292.876392] mpp_rkvdec2 fdf80200.rkvdec: session 13 task 62868 timeout 1 abort 0 force_dequeue 0
[ 1292.876401] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[ 1292.877536] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
[ 1292.877552] mpp_rkvdec2 fdf80200.rkvdec: reset done

这问题卡了一段时间了，实在是没有思路了，来此请教下：
（1）这种 timeout 是啥原因导致呢，mpp解码器解码不过来了？
（2）多个实例同时解码时，复位某一路解码器时会影响其他路解码器吗
（3）有啥解决建议呢

HermanChen, 2025-01-22:
报错是因为有错误码流，硬件出错之后会复位，但是有些错误会让硬件出不了中断，就会超时
可以把内核驱动更新下看看，加了 sip 复位的新内核会解决这个问题，可以的话，把 sdk 升级下，光更新 mpp 是没法解决的，问题在内核

denghengli, 2025-01-22:
> 报错是因为有错误码流，硬件出错之后会复位，但是有些错误会让硬件出不了中断，就会超时 可以把内核驱动更新下看看，加了 sip 复位的新内核会解决这个问题，可以的话，把 sdk 升级下，光更新 mpp 是没法解决的，问题在内核

我们现在用的是5.10.198版本的内核，是在2024.5月份更新的，当时也是为了解决跟这个问题类似的一个问题（码流不对导致解码器异常后直接死掉了，必须重新断电上电才能恢复），现在这个是重新执行下程序重新初始化解码器还能恢复。
后面mpp驱动这块又有更新吗，还是说需要更新成6.x版本的呢？

HermanChen, 2025-01-22:
2024.5月份 应该算是比较新的了，正常不应该出现这样的问题，看下公司服务器上是不是还有更新一些的，用 6.1 高版本的也会好些

closed 2025-02-20 by HermanChen