rockchip-linux/mpp #771: GST硬解码插件报错

ThinkWD, 2025-01-23:
香橙派5plus，RK3588，手动编译安装的最新版 mpp 和 gst 插件。

执行命令：( 别的机器上用软解码可以正常跑 )

`
gst-launch-1.0 uridecodebin uri=http://devimages.apple.com/iphone/samples/bipbop/bipbopall.m3u8 ! videoconvert ! autovideosink
`

打印信息显示 pipeline 成功切换到 PLAYING 状态，但 uridecodebin 似乎没有数据出来或者出来的数据不正确

具体报错：
```
lxx@orangepi5plus:~$ gst-launch-1.0 uridecodebin uri=http://devimages.apple.com/iphone/samples/bipbop/bipbopall.m3u8 ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'source': gst.soup.session=context, session=(SoupSession)NULL, force=(boolean)false;
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
Prerolled, waiting for buffering to finish...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
WARNING: from element /GstPipeline:pipeline0/GstVideoConvert:videoconvert0: Internal GStreamer error: code not implemented.  Please file a bug at http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer.
Additional debug info:
gstvideofilter.c(292): gst_video_filter_transform (): /GstPipeline:pipeline0/GstVideoConvert:videoconvert0:
invalid video buffer received
Bus error (core dumped)
```

HermanChen, 2025-01-23:
gst 插件的问题到官方的 redmine 上提问吧，这里没法处理……

ThinkWD, 2025-01-23:
> gst 插件的问题到官方的 redmine 上提问吧，这里没法处理……

这里不是官方吗，用的是 mpp 的 mppvideodec，这是 pipeline 的 dump 文件：

[0.00.02.235854607-gst-launch.READY_PAUSED.pdf](https://github.com/user-attachments/files/18515818/0.00.02.235854607-gst-launch.READY_PAUSED.pdf)
[0.00.02.246537945-gst-launch.PAUSED_PLAYING.pdf](https://github.com/user-attachments/files/18515819/0.00.02.246537945-gst-launch.PAUSED_PLAYING.pdf)
[0.00.02.685683807-gst-launch.warning.pdf](https://github.com/user-attachments/files/18515817/0.00.02.685683807-gst-launch.warning.pdf)

HermanChen, 2025-01-23:
这里是 mpp 库啊，不是 gst 的插件库……
分析问题先从上层的 gst 插件开始，一下子到底层的 mpp 库里没法回答，我们这边没有 gst 相关的环境
rockchip 有个 redmine，到那里提非 mpp 的问题
https://redmine.rock-chips.com/

HermanChen, 2025-01-23:
/dev/mpp_service 这个设备有么？

HermanChen, 2025-01-23:
mpp_platform_test 跑一下

ThinkWD, 2025-01-23:
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_platform_test 
mpp[21978]: mpp_platform: client 12 driver is not ready!
mpp[21978]: mpp_plat_test: kernel version: 5.10
mpp[21978]: mpp_plat_test: ioctl  version: mpp_service
mpp[21978]: mpp_plat_test: 
mpp[21978]: mpp_plat_test: chip name: rockchip,rk3588-orangepi-5-plus rockchip,rk3588
mpp[21978]: mpp_plat_test: 
mpp[21978]: mpp_plat_test: chip vcodec type 100d221a
mpp[21978]: mpp_plat_test: found vpu2 codec
mpp[21978]: mpp_plat_test: found rkvdec decoder
mpp[21978]: mpp_plat_test: found rk jpeg decoder
mpp[21978]: mpp_plat_test: found rkvenc encoder
mpp[21978]: mpp_plat_test: found vpu2 encoder
mpp[21978]: mpp_plat_test: 
mpp[21978]: mpp_plat_test: start probing decoder device name:
mpp[21978]: mpp_plat_test: H.264 decoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: H.265 decoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: MJPEG decoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: VP9   decoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: avs+  decoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: 
mpp[21978]: mpp_plat_test: start probing encoder device name:
mpp[21978]: mpp_plat_test: H.264 encoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: H.265 encoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: MJPEG encoder: /dev/mpp_service
mpp[21978]: mpp_plat_test: mpp platform test done

lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ cd /dev
lxx@orangepi5plus:/dev$ ll -h | grep mpp
crw-rw-rw-   1 root     video   241,   0 Jan 22 13:01 mpp_service
```

ThinkWD, 2025-01-23:
编译 mpp 用的命令：
```
git clone https://github.com/rockchip-linux/mpp.git --depth=1
cd mpp && mkdir mppbuild && cd mppbuild 
cmake ..
make -j8
sudo make install
```
编译过程没有报错，看起来一切都很正常 ......

HermanChen, 2025-01-23:
运行时候加 sudo

HermanChen, 2025-01-23:
mpp_syslog_pinfo 打错了，是 mpp_syslog_perror=1

ThinkWD, 2025-01-23:
> mpp_syslog_pinfo 打错了，是 mpp_syslog_perror=1

这个是我想试下换成info会不会有更多信息来着，mpp_syslog_perror 是之前就运行过了

> 运行时候加 sudo

mpi_dec_test 加 sudo 运行，还是什么都不打印，报段错误

mpp_platform_test 加 sudo 运行没有任何打印了，反而不加 sudo 能正常打印

ThinkWD, 2025-01-23:
文件夹下所有测试：（mpp_dmabuf_test、mpp_runtime_test 似乎没有通过）

mpp_dmabuf_test 
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_dmabuf_test 
mpp[26535]: mpp_dmabuf_test: mpp dmabuf test start
mpp[26535]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
mpp[26535]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20004
mpp[26535]: mpp_dmabuf_test: get dmaheap buffer group failed ret -1
mpp[26535]: mpp_dmabuf_test: mpp dmabuf test done failed
```
mpp_runtime_test
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_runtime_test
mpp[26594]: mpp_rt_test: mpp found ion buffer is invalid
mpp[26594]: mpp_rt_test: mpp found drm buffer is invalid
```
mpp_env_test 
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_env_test 
mpp[26536]: mpp_env_test: set env: test_env_debug to 256
mpp[26536]: mpp_env_test: set env: test_env_string to just for debug
mpp[26536]: mpp_env_test: start reading env:
mpp[26536]: mpp_env_test: get env: test_env_debug is 256
mpp[26536]: mpp_env_test: get env: test_env_string is just for debug
```
mpp_eventfd_test 
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_eventfd_test 
mpp[26538]: mpp_eventfd_test: eventfd test start
mpp[26538]: mpp_eventfd_test: eventfd get data_fd 4 sync_fd 5
mpp[26538]: mpp_eventfd_test: eventfd test on writing 1 -> 10
mpp[26538]: mpp_eventfd_test: eventfd write 4 val 1
mpp[26538]: mpp_eventfd_test: eventfd read 4 timeout -1 ret 0 val 1
mpp[26538]: mpp_eventfd_test: eventfd block mode     test wr: 1 ret 0 rd: 1 ret 0 - success
mpp[26538]: mpp_eventfd_test: eventfd test on writing 1 -> 11
mpp[26538]: mpp_eventfd_test: eventfd read 4 timeout 0 ret 0 val 11
mpp[26538]: mpp_eventfd_test: eventfd write 4 val 1
mpp[26538]: mpp_eventfd_test: eventfd non-block mode test wr: 1 ret 0 rd: 11 ret 0 - success
mpp[26538]: mpp_eventfd_test: eventfd test on writing 1 -> 12
mpp[26538]: mpp_eventfd_test: eventfd write 4 val 1
mpp[26538]: mpp_eventfd_test: eventfd read 4 timeout 100 ret 0 val 1
mpp[26538]: mpp_eventfd_test: eventfd timeout   mode test wr: 1 ret 0 rd: 1 ret 0 - success
mpp[26538]: mpp_eventfd_test: eventfd test on writing 1 -> 13
mpp[26538]: mpp_eventfd_test: eventfd read 4 timeout 1 ret 0 val 13
mpp[26538]: mpp_eventfd_test: eventfd write 4 val 1
mpp[26538]: mpp_eventfd_test: eventfd timeout   mode test wr: 1 ret 0 rd: 13 ret 0 - success
mpp[26538]: mpp_eventfd_test: eventfd test done
```
mpp_log_test 
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_log_test 
mpp[26543]: mpp_log_test: mpp log test start
mpp[26543]: mpp_log_test: 
set log level to MPP_LOG_VERBOSE
mpp[26543]: mpp_log_test: test mpp_logf
mpp[26543]: mpp_log_test: test mpp_loge
mpp[26543]: mpp_log_test: test mpp_logw
mpp[26543]: mpp_log_test: test mpp_logi
mpp[26543]: mpp_log_test: test mpp_logd
mpp[26543]: mpp_log_test: test mpp_logv
mpp[26543]: mpp_log_test: 
set log level to MPP_LOG_DEBUG
mpp[26543]: mpp_log_test: test mpp_logf
mpp[26543]: mpp_log_test: test mpp_loge
mpp[26543]: mpp_log_test: test mpp_logw
mpp[26543]: mpp_log_test: test mpp_logi
mpp[26543]: mpp_log_test: test mpp_logd
mpp[26543]: mpp_log_test: 
set log level to MPP_LOG_INFO
[... 109 lines cut ...]
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_time_test
mpp[26613]: mpp_time_test: mpp time test start
mpp[26613]: mpp_time_test: time 0 93825235395 us
mpp[26613]: mpp_time_test: time 1 93825245568 us
mpp[26613]: mpp_time_test: diff expected 10 ms real 10.17 ms
mpp[26613]: mpp_time_test: mpp time test done
mpp[26613]: mpp_time_test: mpp timer test start
mpp[26613]: mpp_time_test: mpp timer trigger 0
mpp[26613]: mpp_time_test: mpp timer trigger 1
mpp[26613]: mpp_time_test: mpp timer test done
mpp[26613]: mpp_time_test: mpp clock test start
mpp[26613]: mpp_time_test: loop 1000 times clock 2ms start/pause operation:
mpp[26613]: mpp_time_test: total   time  2124.638 ms
mpp[26613]: mpp_time_test: average time     2.124 ms
mpp[26613]: mpp_time_test: mpp_time cost 0.000 ms
mpp[26613]: mpp_time_test: mpp_time pause 0 at 20.136 ms pause 1 at 40.235 ms
mpp[26613]: mpp_time_test: mpp time test done
```
mpp_trace_test
```
lxx@orangepi5plus:~/VStreamer/workspace/mpp/mppbuild/osal/test$ ./mpp_trace_test
mpp[26615]: mpp_trace_test: mpp trace test start
B|26615|mpp_trace_testE|26615|mpp_trace_testS|26615|mpp_trace_test async|10F|26615|mpp_trace_test async|10C|26615|mpp_trace_test int32|256C|26615|mpp_trace_test int64|100000000mpp[26615]: mpp_trace_test: mpp trace test done
```

HermanChen, 2025-01-23:
内存分配器没有，ion / drm / dmaheap 这三个必须要有一个

ThinkWD, 2025-01-23:
> 内存分配器没有，ion / drm / dmaheap 这三个必须要有一个

可算有方向了，这个内存分配器要怎么安装，有相关文档吗

HermanChen, 2025-01-23:
估计是显示模块没打开，内核编译的时候，要打开 DRM 或者 DMABUF

ThinkWD, 2025-01-23:
系统内核吗？我看 mpp 编译好像没有 HAVE_DRM 这个选项，能具体说一下怎么做或者有相关文档吗，多谢多谢

HermanChen, 2025-01-23:
这个取决于内核编码时候的选项，这块不在 mpp 里

ThinkWD, 2025-01-23:
应该找买板子的商家吗？还是说自己编译系统内核然后重做系统 ......

HermanChen, 2025-01-23:
都可以试下，先找下板子的商家

ThinkWD, 2025-01-23:
> 都可以试下，先找下板子的商家

谢谢谢谢，这个i ssue 先关了

closed 2025-01-23 by ThinkWD