rockchip-linux/mpp #775: MPP编码出来的h264文件过大

cht851, 2025-02-06:
用mpp编码出来的h264文件过大，几百mb；用x264重新编译才几mb左右，这是怎么回事，以下是参数配置：
`VideoEnCoder::VideoEnCoder(BUFFERS_INFO_S* BufInfo)
{
    int ret = -1;

    if(!BufInfo)
    {
        printf("bufferInfo is NULL\n");
        return;
    }
    
    ret = mpp_create(&ctx, &mpi);
    if(ret != 0)
    {
        printf("mpp_create error, ret=%d\n", ret);
        return;
    }

    ret = mpp_init(ctx, MPP_CTX_ENC, MPP_VIDEO_CodingAVC);
    if(ret != 0)
    {
        printf("mpp_init error, ret=%d\n", ret);
        return;
    }

    // 配置编码器参数
    ret = mpp_enc_cfg_init(&cfg);
    if(ret != 0)
    {
        printf("mpp_enc_cfg_init, ret=%d\n", ret);
        return;
    }
    
    mpp_enc_cfg_set_s32(cfg, "prep:width", VENC_WIDTH);
    mpp_enc_cfg_set_s32(cfg, "prep:height", VENC_HEIGHT);
    mpp_enc_cfg_set_s32(cfg, "prep:hor_stride", VENC_ALIGN(VENC_WIDTH));
    mpp_enc_cfg_set_s32(cfg, "prep:ver_stride", VENC_HEIGHT);
    mpp_enc_cfg_set_s32(cfg, "prep:format", VENC_FMT);

    mpp_enc_cfg_set_s32(cfg, "rc:mode", VENC_RCMODE);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_flex", 0);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_num", VENC_FPS);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_denorm", 1);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_flex", 0);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_num", VENC_FPS);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_denorm", 1);
    mpp_enc_cfg_set_s32(cfg, "rc:bps_target", 1000000);
    switch(VENC_RCMODE)
    {
        case MPP_ENC_RC_MODE_CBR:
        mpp_enc_cfg_set_s32(cfg, "rc:bps_max", 1000000 * 17 / 16);
        mpp_enc_cfg_set_s32(cfg, "rc:bps_min", 1000000 * 15 / 16);
        break;
        case MPP_ENC_RC_MODE_VBR:
        break;
    }

    // h264 config
    mpp_enc_cfg_set_s32(cfg, "codec:type", MPP_VIDEO_CodingAVC);
    mpp_enc_cfg_set_s32(cfg, "h264:profile", 77);
    mpp_enc_cfg_set_s32(cfg, "h264:level", 40);
    mpp_enc_cfg_set_s32(cfg, "h264:cabac_en", 1);
    mpp_enc_cfg_set_s32(cfg, "h264:cabac_idc", 0);
    mpp_enc_cfg_set_s32(cfg, "h264:trans8x8", 0);
    mpp_enc_cfg_set_s32(cfg, "h264:qp_init", 26);
    mpp_enc_cfg_set_s32(cfg, "h264:qp_max", 28);
    mpp_enc_cfg_set_s32(cfg, "h264:qp_min", 25);
    mpp_enc_cfg_set_s32(cfg, "split:mode", MPP_ENC_SPLIT_NONE);
    mpp_enc_cfg_set_s32(cfg, "split:arg", 0);
    mpp_enc_cfg_set_s32(cfg, "rc:gop", 2 * VENC_FPS);

    ret = mpi->control(ctx, MPP_ENC_SET_CFG, cfg);
    if(ret != 0)
    {
        printf("MPP_ENC_SET_CFG error, ret=%d\n", ret);
        return;
    }

    sei_mode = MPP_ENC_SEI_MODE_DISABLE;
    ret = mpi->control(ctx, MPP_ENC_SET_SEI_CFG, &sei_mode);
    if(ret != 0)
    {
        printf("MPP_ENC_SET_SEI_CFG error, ret=%d\n", ret);
        return;
    }

    header_mode = MPP_ENC_HEADER_MODE_DEFAULT;
    ret = mpi->control(ctx, MPP_ENC_SET_HEADER_MODE, &header_mode);
    if(ret != 0)
    {
        printf("MPP_ENC_HEADER_MODE_EACH_IDR error, ret=%d\n", ret);
        return;
    }

    // 导入外部buffers
    for(int i = 0; i < BUFF_NUM; i++)
    {
        MppBufferInfo Info;
        // 必须全部置0，否则报错
        memset(&Info, 0, sizeof(MppBufferInfo));
        Info.fd = BufInfo[i].fd;
        Info.index = BufInfo[i].index;
        Info.type = MPP_BUFFER_TYPE_EXT_DMA;
        Info.size = BufInfo[i].length;

        ret = mpp_buffer_import(&group[i], &Info);
        if(ret != MPP_OK)
        {
            printf("mpp_buffer_import, ret=%d\n", ret);
            return;
        }
    }
}`

cht851, 2025-02-06:
是1080p 25fps的，cbr模式

HermanChen, 2025-02-07:
qp_max 太小，配置到 48

HermanChen, 2025-02-07:
上传下码流

cht851, 2025-02-10:
https://github.com/user-attachments/assets/4f0933db-10b4-42dc-bd1b-63d71e6cc7a2

HermanChen, 2025-02-20:
看起来这个图像的噪声比较大，内容不太稳定，最好把摄像头的效果调正常一些再来测试