rockchip-linux/mpp #776: H.264 hardware encoder bug

cppcoffee, 2025-02-07:
Use NanoPi R5C Rockchip RK3568B2 CPU to capturing the Windows 10 installation interface and then performing H.264 hardware encoding using RK3568 triggers a bug in the H.264 hardware encoder.  

The command used is as follows:  

```bash
gst-launch-1.0 rtspclientsink name=s location=rtsp://localhost:8554/mystream -e v4l2src device=/dev/kvmd-video ! image/jpeg,width=1920,height=1080 ! mppjpegdec ! videoconvert ! mpph264enc rc-mode=vbr ! h264parse ! queue ! s.sink_0
```

[image]

cppcoffee, 2025-02-07:
List devices:

```shell
root@NanoPi-R5C:/home/pi# v4l2-ctl --list-devices
USB3. 0 capture: USB3. 0 captur (usb-xhci-hcd.4.auto-1):
        /dev/video0
        /dev/video1
        /dev/media0
```

The video0 information:

```shell
root@NanoPi-R5C:/home/pi# v4l2-ctl -d /dev/video0 --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : USB3. 0 capture: USB3. 0 captur
        Bus info         : usb-xhci-hcd.4.auto-1
        Driver version   : 6.1.25
        Capabilities     : 0x84a00001
root@NanoPi-R5C:/home/pi# v4l2-ctl -d /dev/video0 --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : USB3. 0 capture: USB3. 0 captur
        Bus info         : usb-xhci-hcd.4.auto-1
        Driver version   : 6.1.25
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : USB3. 0 capture: USB3. 0 captur
        Serial           :
        Bus info         : usb-xhci-hcd.4.auto-1
        Media version    : 6.1.25
        Hardware revision: 0x00002100 (8448)
        Driver version   : 6.1.25
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : USB3. 0 capture: USB3. 0 captur
        Function         : V4L2 I/O
        Flags            : default
        Pad 0x01000007   : 0: Sink
          Link 0x0200000d: from remote pad 0x100000a of entity 'Processing 2' (Video Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Input 1: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1920, Height 1080
        Default     : Left 0, Top 0, Width 1920, Height 1080
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 60.000 (60/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=-128 max=127 step=1 default=-11 value=-11
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=148 value=148
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=180 value=180
                            hue 0x00980903 (int)    : min=-128 max=127 step=1 default=0 value=0
```

HermanChen, 2025-02-07:
获取到的码流有问题么？

cppcoffee, 2025-02-07:
windows10 安装界面，测试到只有这两个界面出现问题，其他界面正常。1920x1080 分辨率，当碰到这两个界面就从 30帧 掉到 10+帧

cppcoffee, 2025-02-20:
观察到日志输出

```
Feb 20 14:19:12 lubancat mpp[1434]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:12 lubancat mpp[1434]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:12 lubancat mpp[1434]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:12 lubancat mpp[1434]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:23 lubancat mpp[1439]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:23 lubancat mpp[1439]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:23 lubancat mpp[1439]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Feb 20 14:19:23 lubancat mpp[1439]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
```

cppcoffee, 2025-02-20:
使用 LubanCat1 RK3566 查看系统日志观察到

```
...
Feb 20 17:40:00 lubancat mpp[789]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
Feb 20 17:40:09 lubancat mpp[789]: jpegd_parser: jpegd_decode_sos len 12 is too large
Feb 20 17:40:09 lubancat mpp[789]: jpegd_parser: jpegd_decode_frame sos decode error
Feb 20 17:40:09 lubancat mpp[789]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
Feb 20 17:40:11 lubancat kernel: [  322.226172] mpp_jpgdec fded0000.jpegd: resetting...
Feb 20 17:40:11 lubancat kernel: [  322.226300] mpp_jpgdec fded0000.jpegd: reset done
Feb 20 17:40:09 lubancat mpp[789]: message repeated 2 times: [ mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!]
Feb 20 17:40:11 lubancat mpp[789]: hal_jpegd_rkv: decode result: failed, irq 0x0000214d
Feb 20 17:40:11 lubancat mpp[789]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
...
```

cppcoffee, 2025-02-21:
NanoPi R5C 出现花屏的系统日志输出

```
...
2025-02-21T01:15:23.816781+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.817780+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.823762+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.897513+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.898654+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.904248+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.983029+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:23.988632+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.064525+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.070465+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.154963+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.239976+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.320268+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.321378+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.327370+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.407921+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.413605+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.489489+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.495146+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.578928+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.664088+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.743425+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
2025-02-21T01:15:24.748939+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
...
```

cppcoffee, 2025-02-21:
将视频编码后输出到文件中，命令：

```
gst-launch-1.0 -e v4l2src device=/dev/kvmd-video ! image/jpeg,width=1920,height=1080 ! mppjpegdec ! videoconvert ! mpph264enc rc-mode=vbr ! h264parse ! qtmux ! filesink location=output.mp4
``

输出的文件:

https://github.com/user-attachments/assets/00db5a3d-cbf2-4690-8e6b-1f1f828f157c

HermanChen, 2025-02-21:
> NanoPi R5C RK3588 在 windows 装机的时候，出现花屏的系统日志输出：
> 
> ```
> ...
> 2025-02-21T01:15:23.816781+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.817780+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.823762+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.897513+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.898654+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.904248+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.983029+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:23.988632+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.064525+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.070465+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.154963+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.239976+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.320268+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.321378+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.327370+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.407921+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.413605+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.489489+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.495146+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.578928+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.664088+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.743425+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> 2025-02-21T01:15:24.748939+00:00 NanoPi-R5C mpp[2840]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
> ...
> ```

这个说明 jpeg 码流解起来有问题，可以提供下 jpeg 码流

cppcoffee, 2025-02-21:
确定是 jpeg 硬解码出错导致的问题，使用 `video/x-raw,format=YUY2` 格式就正常

查看了一下采集卡支持的格式如下:

```shell
root@NanoPi-R5C:/home/pi# v4l2-ctl --device=/dev/kvmd-video --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1600x1200
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1360x768
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.020s (50.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.020s (50.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.020s (50.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 720x576
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.020s (50.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 720x480
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.020s (50.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
[... 35 lines cut ...]
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```

输出 jpeg 视频文件，命令:

```shell
$ gst-launch-1.0 -e v4l2src device=/dev/kvmd-video ! image/jpeg,width=1920,height=1080 ! filesink location=output.mjpg
```

压缩的视频文件:

[output.mjpg.tar.gz](https://github.com/user-attachments/files/18898099/output.mjpg.tar.gz)

解压命令与校验和:

```shell
$ tar -xzvf ./output.mjpg.tar.gz
x ./output.mjpg
$ openssl dgst -md5 ./output.mjpg
MD5(./output.mjpg)= 2658bbc95fc4b2710a1af777b05bbe4a
```

cppcoffee, 2025-02-21:
可能是视频采集卡的问题，换了一张视频采集卡，使用 jpeg 编解码就正常了

HermanChen, 2025-02-21:
jpeg 解码只支持 baseline 的，不支持 progressive，对比下 jpeg 码流

cppcoffee, 2025-02-21:
尝试将 output.mjpeg 文件的数据流, 找到 jpeg 头部，切出来观察到它是 baseline:

```shell
# file ./2.jpeg
./2.jpeg: JPEG image data, baseline, precision 8, 1920x1080, components 3
```

切出来的文件:

[2.jpeg.tar.gz](https://github.com/user-attachments/files/18902436/2.jpeg.tar.gz)

附件的解压与校验码:

```shell
$ tar -xzvf ./2.jpeg.tar.gz
x ./2.jpeg
$ openssl dgst -md5 ./2.jpeg
MD5(./2.jpeg)= 660af29aedd5390eb98d194280ed391f
$ file 2.jpeg
2.jpeg: JPEG image data, baseline, precision 8, 1920x1080, components 3
```

cppcoffee, 2025-02-21:
查看 `2.jpeg` 发现采集卡的数据流有一串序列，可能是两个JPEG文件连接在一起了。第一个JPEG非常短（只有开始和结束标记），然后立即开始了第二个JPEG:

```
00014120  80 0a 28 00 a2 80 0a 28  00 a2 80 3f ff d9 ff d8  |..(....(...?....|
                                               -EOI---SOI--
00014130  ff d9 ff d8 ff d9 ff d8  ff db 00 43 00 08 06 06  |...........C....|
          -EOI---SOI--EOI----SOI--
00014140  07 06 05 08 07 07 07 09  09 08 0a 0c 14 0d 0c 0b  |................|
```

qvoid, 2025-02-21:
将你提供的jpeg 送给 PC上ffplay解码，解码也报错了：
```
[mjpeg @ 0x79ab24004480] No JPEG data found in image
[mjpeg @ 0x79ab24004480] error count: 67
[mjpeg @ 0x79ab24004480] error y=49 x=75
[mjpeg @ 0x79ab24004480] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x79ab24004480] No JPEG data found in image
[mjpeg @ 0x79ab24004480] error count: 66
[mjpeg @ 0x79ab24004480] error y=49 x=75
[mjpeg @ 0x79ab24004480] Found EOI before any SOF, ignoring f=0/0   
[mjpeg @ 0x79ab24004480] No JPEG data found in image
[mjpeg @ 0x79ab24004480] error count: 68
[mjpeg @ 0x79ab24004480] error y=49 x=66
```
画面上看也是有错乱的。所以应该是 JPEG 码流出错。可能是采集卡的问题。

cppcoffee, 2025-02-21:
3q

cppcoffee, 2025-02-21:
> 将你提供的jpeg 送给 PC上ffplay解码，解码也报错了：
>
> ```
> [mjpeg @ 0x79ab24004480] No JPEG data found in image
> [mjpeg @ 0x79ab24004480] error count: 67
> [mjpeg @ 0x79ab24004480] error y=49 x=75
> [mjpeg @ 0x79ab24004480] [...]

这里是使用什么命令去播放 JPEG 码流？执行 `ffplay output.mjpeg` 吗？