rockchip-linux/mpp #777: 编码之后的图像有几率不正常

2398tr, 2025-02-08:
编码之后的图像有几率不正常，这个是随机出现的，一分钟的视频中有几分钟如下所示，这个分辨率是3840x3840的。我还有一路480x480的是通过3840x3840缩放而来的，480x480编码出的图像在3840x3840的图像不正常的时间段里的 图像正常的，显然这不是原图像的问题，很大可能是编码问题
[image]

2398tr, 2025-02-08:
使用ffmpeg查看 会出现以下问题 

[image]

HermanChen, 2025-02-10:
把异常的码流存一段放上来

2398tr, 2025-02-10:
貌似是镜头怼着桌面拍的原因，我有两个摄像头，一个拍的天花板，一个怼着桌面，拍天花板的摄像头编码还没有发现有问题。

2398tr, 2025-02-10:
这怎么上传视频，我点击上传视频一直显示加载失败

2398tr, 2025-02-10:
异常的码流我放到了这里 https://1drv.ms/v/c/023d8c3d2b731359/EVrTYwFOJ-hIjzMI48e7J9gB4vuOLdoWeEG2SUujTeYF2g?e=kNq4jb
下载源码后用vlc查看，不用用网站的播放器查看

2398tr, 2025-02-20:
RK3588S,我找到原因了，是因为镜头距离桌面太近，导致拍摄的图像比较模糊，编码后就有几率出现这种花屏现象，拍远处的物体现在还没有出现这种现象