rockchip-linux/mpp #778: 关于usb摄像头mjpeg解码输入问题

yyazzz, 2025-02-11:
3588S对usb摄像头mjpeg包解码
参考https://github.com/rockchip-linux/mpp/issues/592代码能正常解码。
但是里面代码
memcpy((RK_U8 *)mpp_buffer_get_ptr(input_buf), pkt_data, pkt_size);
mpp_packet_init_with_buffer(&packet, input_buf);
使用MppBuffer将输入包数据构建成MppPacket。
其中在构建MppBuffer使用了copy来复制mjpeg数据耗时最高十几毫秒。
为解决这个问题，我尝试
mpp_packet_init(&packet, pkt_data, pkt_size);
    mpp_packet_set_data(packet, pkt_data);
    mpp_packet_set_size(packet, pkt_size);
    mpp_packet_set_pos(packet, pkt_data);
    mpp_packet_set_length(packet, pkt_size);
来使用mjpeg指针来构建MppPacket，但显示绿屏。
还有零拷贝的方式来构建MppPacket吗

HermanChen, 2025-02-11:
用 MppBuffer 来直接保存收到的 mjpeg 码流

yyazzz, 2025-02-11:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

yyazzz, 2025-02-11:
> 用 MppBuffer 来直接保存收到的 mjpeg 码流
参照mpp里的camera_source.c文件，我在usb驱动那边添加了
camera_source_ioctl(ctx->fd, VIDIOC_EXPBUF, &expbuf)相关代码
这些代码好像是将v4l2摄像头数据顺便也导出到dma外部内存，即导出到MppBuffer 内存里，但是出现花屏现象。
在mpp_packet_init_with_buffer(&packet, input_buf);
同样的代码，如果input_buf是mmap映射的虚拟内存copy的MppBuffer 就没事，如果MppBuffer 是通过v4l2自己导出的MppBuffer 就花屏。

[image]

yyazzz, 2025-02-11:
[image]
参照mpp里的camera_source.c文件在V4L2初始化时添加的代码

HermanChen, 2025-02-11:
直接用 mpi_enc_test -i /dev/videox 这样的方式来编码呢？

yyazzz, 2025-02-11:
> 图上的线是 cache 问题，图像有用 cpu 拷贝？
图像在解码输出后是有copy，但在送进解码器前没copy操作

HermanChen, 2025-02-11:
问题是出在解码之后图像拷贝，拷贝之后要 flush 一下 MppBuffer 把数据写进 ddr 再让编码器硬件去访问

yyazzz, 2025-02-11:
> 问题是出在解码之后图像拷贝，拷贝之后要 flush 一下 MppBuffer 把数据写进 ddr 再让编码器硬件去访问

我是通过rga让编码器访问的

[image]
通过解码器回调的FD构建YUV帧RGA，然后通过rga的imcopy把copy后的RGA传给编码器
我不太懂flush 操作，具体是通过什么接口来执行，我找到了mpp_dec_flush接口

HermanChen, 2025-02-11:
用 rga 拷贝应该是没问题的，存一下 rga 拷贝后的图像，看有没有问题

yyazzz, 2025-02-11:
> 用 rga 拷贝应该是没问题的，存一下 rga 拷贝后的图像，看有没有问题

我尝试在回调里保存解码出来的yuv图片，图片看起来没问题，考虑到保存yuv会有很大的耗时，我在imcopy前加个20ms延迟替换yuv保存操作，流开始时会有白条，一会就没了。如果是10ms堵塞，白条还在。

yyazzz, 2025-02-12:
[https://github.com/rockchip-linux/mpp/issues/655](url)
我测试下来也跟这个描述的一样

yyazzz, 2025-02-12:
根据前面加延迟测试，我更改了v4l2读数据的逻辑。
由之前V4L2通过VIDIOC_DQBUF获取到一帧马上将当前帧MppBuffer推到解码器，改为读取上一帧的MppBuffer推解码
即通过缓存了2帧图像来替换延迟操作，图像变为偶尔出现花屏而不是一直花屏了。但这样并不能作为解决问题的操作。
判断MppBuffer或者v4l2的VIDIOC_EXPBUF是不是少了一些什么同步操作？导致解码的MppBuffer异常。
我尝试mpp_buffer_sync_end，mpp_buffer_sync_begin好像没用

HermanChen, 2025-02-12:
读取上一帧肯定是不对的，读取当前帧之后，做解码，解码完成之后再 EQBUF 释放回 V4L2，这样才是正常的流程吧
sync 操作不生效的话，跟下 dmabuf 的驱动，有不少内核这块处理有 bug

closed 2025-02-14 by yyazzz

DianKeWei, 2025-04-19:
> _

如何操作，可以给出示例么

chakloong883, 2025-09-28:
> 这是来自QQ邮箱的假期自动回复邮件。
>  
> 您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

大佬能给出代码不