rockchip-linux/mpp #780: rk3588上，Mpp支持解码GDR 编码格式的H264嘛

huanying3200, 2025-02-14:
你好，请问rk3588上，Mpp支持解码GDR 编码格式的H264嘛？mpi_dec_test.c中pkt_size最大能设置多大？

HermanChen, 2025-02-14:
支持 GDR，pkt_size 可以很大，没啥限制

huanying3200, 2025-02-14:
好的，这边需使用Mpp解码从大疆飞机获取的(GDR 编码格式的 H20/H20T)H.264 码流，单次接收码流长度接近6万字节，pkt_size没有限制长度就可以一次都塞进去。另外可以设置输出的yuv格式嘛？需要使用I420格式的，设置-f 4后输出的还是NV12格式的？

/* mpp color format index definition */
typedef enum {
    MPP_FMT_YUV420SP        = (MPP_FRAME_FMT_YUV + 0),  /* YYYY... UV... (NV12)     */
    /*
     * A rockchip specific pixel format, without gap between pixel aganist
     * the P010_10LE/P010_10BE
     */
    MPP_FMT_YUV420SP_10BIT  = (MPP_FRAME_FMT_YUV + 1),
    MPP_FMT_YUV422SP        = (MPP_FRAME_FMT_YUV + 2),  /* YYYY... UVUV... (NV16)   */
    MPP_FMT_YUV422SP_10BIT  = (MPP_FRAME_FMT_YUV + 3),  ///< Not part of ABI
    MPP_FMT_YUV420P         = (MPP_FRAME_FMT_YUV + 4),  /* YYYY... U...V...  (I420) */
    MPP_FMT_YUV420SP_VU     = (MPP_FRAME_FMT_YUV + 5),  /* YYYY... VUVUVU... (NV21) */
    MPP_FMT_YUV422P         = (MPP_FRAME_FMT_YUV + 6),  /* YYYY... UU...VV...(422P) */
    MPP_FMT_YUV422SP_VU     = (MPP_FRAME_FMT_YUV + 7),  /* YYYY... VUVUVU... (NV61) */
    MPP_FMT_YUV422_YUYV     = (MPP_FRAME_FMT_YUV + 8),  /* YUYVYUYV... (YUY2)       */
    MPP_FMT_YUV422_YVYU     = (MPP_FRAME_FMT_YUV + 9),  /* YVYUYVYU... (YVY2)       */
    MPP_FMT_YUV422_UYVY     = (MPP_FRAME_FMT_YUV + 10), /* UYVYUYVY... (UYVY)       */
    MPP_FMT_YUV422_VYUY     = (MPP_FRAME_FMT_YUV + 11), /* VYUYVYUY... (VYUY)       */
    MPP_FMT_YUV400          = (MPP_FRAME_FMT_YUV + 12), /* YYYY...                  */
    MPP_FMT_YUV440SP        = (MPP_FRAME_FMT_YUV + 13), /* YYYY... UVUV...          */
    MPP_FMT_YUV411SP        = (MPP_FRAME_FMT_YUV + 14), /* YYYY... UV...            */
    MPP_FMT_YUV444SP        = (MPP_FRAME_FMT_YUV + 15), /* YYYY... UVUVUVUV...      */
    MPP_FMT_YUV444P         = (MPP_FRAME_FMT_YUV + 16), /* YYYY... UUUU... VVVV...  */
    MPP_FMT_YUV444SP_10BIT  = (MPP_FRAME_FMT_YUV + 17),
    MPP_FMT_YUV_BUTT,

    MPP_FMT_RGB565          = (MPP_FRAME_FMT_RGB + 0),  /* 16-bit RGB               */
    MPP_FMT_BGR565          = (MPP_FRAME_FMT_RGB + 1),  /* 16-bit RGB               */
    MPP_FMT_RGB555          = (MPP_FRAME_FMT_RGB + 2),  /* 15-bit RGB               */
    MPP_FMT_BGR555          = (MPP_FRAME_FMT_RGB + 3),  /* 15-bit RGB               */
    MPP_FMT_RGB444          = (MPP_FRAME_FMT_RGB + 4),  /* 12-bit RGB               */
    MPP_FMT_BGR444          = (MPP_FRAME_FMT_RGB + 5),  /* 12-bit RGB               */
    MPP_FMT_RGB888          = (MPP_FRAME_FMT_RGB + 6),  /* 24-bit RGB               */
    MPP_FMT_BGR888          = (MPP_FRAME_FMT_RGB + 7),  /* 24-bit RGB               */
    MPP_FMT_RGB101010       = (MPP_FRAME_FMT_RGB + 8),  /* 30-bit RGB               */
    MPP_FMT_BGR101010       = (MPP_FRAME_FMT_RGB + 9),  /* 30-bit RGB               */
    MPP_FMT_ARGB8888        = (MPP_FRAME_FMT_RGB + 10), /* 32-bit RGB               */
    MPP_FMT_ABGR8888        = (MPP_FRAME_FMT_RGB + 11), /* 32-bit RGB               */
    MPP_FMT_BGRA8888        = (MPP_FRAME_FMT_RGB + 12), /* 32-bit RGB               */
    MPP_FMT_RGBA8888        = (MPP_FRAME_FMT_RGB + 13), /* 32-bit RGB               */
    MPP_FMT_RGB_BUTT,

    MPP_FMT_BUTT,
} MppFrameFormat;

HermanChen, 2025-02-14:
解码器硬件是固定 YUV420SP 输出，能变的只有 jpeg 解码器在一些芯片上能变化

huanying3200, 2025-02-14:
既然输出格式是固定的YUV420SP ，那可以把YUV420SP中Y、U、V分量分别取出重排一下变成YUV420P，下面是源码中YUV420SP写入文件部分，base_y所在循环是取Y分量写入文件的，如何分别取出U分量和V分量呢？

dump_mpp_frame_to_file：
   case MPP_FMT_YUV420SP : {
        RK_U32 i;
        RK_U8 *base_y = base;
        RK_U8 *base_c = base + h_stride * v_stride;

        for (i = 0; i < height; i++, base_y += h_stride) {
            fwrite(base_y, 1, width, fp);
        }
        for (i = 0; i < height / 2; i++, base_c += h_stride) {
            fwrite(base_c, 1, width, fp);
        }
    } break;

HermanChen, 2025-02-14:
默认写文件是 YUV420SP 格式的，要写成 420P 的话，看 420P 的方式，反过来即可

huanying3200, 2025-02-14:
有点不明白，怎么反过来？RK_U8 *base_c = base + h_stride * v_stride，利用base_c怎么分别取出U分量和V分量？

HermanChen, 2025-02-17:
搜一下 YUV420SP 和 YUV420P

closed 2025-02-20 by HermanChen