rockchip-linux/mpp #781: regression in mpp_buf_slot ?

aarnoud, 2025-02-15:
works with previous dev : `mpp version: 9bcf367 author: Yandong Lin    2025-01-07 feat[rk_mpi_cmd]: Merge cmds from mpp_interface`

works but constantly shouting errors in logs with : `mpp_info: mpp version: 70b572a author: Hongjin Li     2025-02-13 fix[mpp_sys_cfg]: Fix abnormal stride calculation.` https://github.com/rockchip-linux/mpp/commit/997cfe881dad2a256538d881eae2c046143d4923 ?
```
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 15 15:25:45 rock-5b mpp[3624]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
```

HermanChen, 2025-02-17:
Try the latest patch and which chip are you using?

aarnoud, 2025-02-17:
using. Rock5B (RK3588) and latest commit 

```
Feb 17 05:12:03 rock-5b mpp[1728]: mpp_info: mpp version: 834ea97 author: Herman Chen    2025-02-14 fix[mpp_sys_cfg]: Fix align pixel stride on rk3576

Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_pixel == sys_cfg_info_set.h_stride_by_pixel failed at generate_info_set:506
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.h_stride_by_byte == sys_cfg_info_set.h_stride_by_byte failed at generate_info_set:507
Feb 17 05:12:04 rock-5b mpp[1728]: mpp_buf_slot: Assertion legacy_info_set.size_total == sys_cfg_info_set.size_total failed at generate_info_set:509
```

HermanChen, 2025-02-17:
What is the format of the decoded video ?

HermanChen, 2025-02-17:
[0001-chore-mpp_buf_slot-Modify-sys_cfg-mismatch-print.patch](https://github.com/user-attachments/files/18821226/0001-chore-mpp_buf_slot-Modify-sys_cfg-mismatch-print.patch)
Add this patch and show the mismatch error log

aarnoud, 2025-02-17:
Debian Trixie / X11 / Mesa 25.1.0-devel (git-6652eb0ec3)
```
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:29 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:30 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:30 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:30 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:30 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:31 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:31 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
Feb 17 10:36:31 rock-5b mpp[2090]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Feb 17 10:36:31 rock-5b mpp[2090]: mpp_buf_slot: mismatch v_stride 1088 - 1088
```

HermanChen, 2025-02-17:
2304 is 1920 align to 256 odd times.
How do you use the libmpp.so? Is it mpi_dec_test?

nyanmisaka, 2025-02-20:
Reproduced on rk3588.
```
mpp[12829]: mpp_buf_slot: mismatch size_total 4478976 - 3732480
mpp[12829]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
mpp[12829]: mpp_buf_slot: mismatch h_stride_by_byte 2304 - 1920
```

```
mpp[12253]: mpp_buf_slot: mismatch h_stride_by_byte 2400 - 2432
```

```
// 2304 != 1920 (mismatch h_stride_by_byte)
legacy hevc 8bit  linear: info_set->h_stride_by_byte = hal_hor_stride = hal_hor_align(1920 * 8  / 8) = align(1920, 256)|256 = 2304
syscfg hevc 8bit  linear: cfg->h_stride_by_byte = aligned_byte = align((align(1920, 64)  * 8 / 8), 64) = 1920

// 2816 == 2816 (ok)
legacy hevc 10bit linear: info_set->h_stride_by_byte = hal_hor_stride = hal_hor_align(1920 * 10 / 8) = align(2400, 256)|256 = 2816
syscfg hevc 10bit linear: cfg->h_stride_by_byte = aligned_byte = align(align((align(1920, 64) * 10 / 8), 64), 256)|256 = 2816

// 1920 == 1920 (ok)
legacy hevc 8bit  afbc: info_set->h_stride_by_byte = hal_hor_stride = align(1920, 64) * 8 / 8 = 1920
syscfg hevc 8bit  afbc: cfg->h_stride_by_byte = aligned_byte = align(align(align(1920, 64), 64) * 8 / 8, 64) = 1920

// 2400 != 2432 (mismatch h_stride_by_byte)
legacy hevc 10bit afbc: info_set->h_stride_by_byte = hal_hor_stride = align(1920, 64) * 10 / 8 = 2400
syscfg hevc 10bit afbc: cfg->h_stride_by_byte = aligned_byte = align(align(align(1920, 64), 64) * 10 / 8, 64) = 2432
```

aarnoud, 2025-02-20:
note: still digging this, but h264 8bit seems unaffected, only hevc 8bit on my end

nyanmisaka, 2025-02-20:
The very recent commit https://github.com/rockchip-linux/mpp/commit/834ea9758690c0a2d7f8480a27b63a664b077dbd also introduced a vertical alignment regression on vp9 decoding on rk3588.

```diff
         case MPP_VIDEO_CodingHEVC :
         case MPP_VIDEO_CodingVP9 : {
             aligned_pixel = MPP_ALIGN(aligned_pixel ? aligned_pixel : cfg->width, 64);
-            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 64);
+            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 8);
         } break;
         case MPP_VIDEO_CodingAV1 : {
             aligned_pixel = MPP_ALIGN(aligned_pixel ? aligned_pixel : cfg->width, 128);
-            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 128);
+            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 8);
         } break;
         default : {
             aligned_pixel = MPP_ALIGN(aligned_pixel ? aligned_pixel : cfg->width, 16);
-            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 16);
+            aligned_height = MPP_ALIGN(aligned_height ? aligned_height : cfg->height, 8);
         } break;
```

It seems there's a mismatching in vp9 HAL. On the other hand, h264 & avs2 seem to expect `ver_align == 16`.
https://github.com/rockchip-linux/mpp/blob/413e74bfe661c06078f48aae4e52e64d6e40773d/mpp/hal/rkdec/vp9d/hal_vp9d_com.c#L992-L995
https://github.com/rockchip-linux/mpp/blob/413e74bfe661c06078f48aae4e52e64d6e40773d/mpp/codec/dec/vp9/vp9d_parser.c#L414

See the green stripe and chroma shift shown in the pic:
[image]

hbiyik, 2025-03-23:
i can confirm same issue in latest 2f2e5f5c3efb85430ed2d02974cb057beb56df26, however 75e9008e does not have it

wenyue7, 2025-03-31:
You can update the following patches and test them. They will be released after the subsequent tests are stable.

[0001-fix-sys_cfg-Fix-HAL-layer-buffer-alignment-issue.patch](https://github.com/user-attachments/files/19529102/0001-fix-sys_cfg-Fix-HAL-layer-buffer-alignment-issue.patch)
[0002-fix-sys_cfg-Fix-RK3399-hor-ver-stride-calculation-is.patch](https://github.com/user-attachments/files/19529101/0002-fix-sys_cfg-Fix-RK3399-hor-ver-stride-calculation-is.patch)
[0003-fix-sys_cfg-AVC-is-aligned-to-ctu-to-avoid-info-chan.patch](https://github.com/user-attachments/files/19529098/0003-fix-sys_cfg-AVC-is-aligned-to-ctu-to-avoid-info-chan.patch)
[0004-fix-sys_cfg-Support-alignment-for-mpeg2-mpeg4-h263-v.patch](https://github.com/user-attachments/files/19529100/0004-fix-sys_cfg-Support-alignment-for-mpeg2-mpeg4-h263-v.patch)
[0005-fix-sys_cfg-Fix-external-configuration-stride-issue.patch](https://github.com/user-attachments/files/19529099/0005-fix-sys_cfg-Fix-external-configuration-stride-issue.patch)
[0006-fix-sys_cfg-fix-fbc-ver-stride-calc-issue.patch](https://github.com/user-attachments/files/19529097/0006-fix-sys_cfg-fix-fbc-ver-stride-calc-issue.patch)

aarnoud, 2025-03-31:
I'm getting the green line in HEVC now with those patches (wasn't the case on develop branch)
```
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch h_stride_by_byte 2304 - 1920
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch v_stride 1088 - 1080
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch size_total 4512153 - 3732480
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_dec: mpp_dec use internal frame buffer group
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch h_stride_by_pixel 2304 - 1920
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch h_stride_by_byte 2304 - 1920
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch v_stride 1088 - 1080
Mar 31 10:03:02 rock-5b mpp[90662]: mpp_buf_slot: mismatch size_total 4512153 - 3732480
```

[image]

wenyue7, 2025-04-01:
What is the image resolution? Maybe you could upload the sequence?

aarnoud, 2025-04-04:
> What is the image resolution? Maybe you could upload the sequence?

1920x1080 HEVC bitstream not using AFBC, now the issue is on 1.0.9/Develop (using https://github.com/armbian/linux-rockchip/tree/rk-6.1-rkr5 / Gnome 48 / Debian Trixie / Mesa dev)

hbiyik, 2025-04-06:
can confirm that issues i reported are gone in 1.0.9, did not try out your patchset though @wenyue7

hbiyik, 2025-04-06:
@wenyue7 issue still persists in 1080p hevc videos both on 8bit or 10bit. Rest seems to be fixed.

You can reproduce with those test files

https://repo.jellyfin.org/test-videos/SDR/HEVC%208bit/Test%20Jellyfin%201080p%20HEVC%208bit%203M.mp4
https://repo.jellyfin.org/test-videos/SDR/HEVC%2010bit/Test%20Jellyfin%201080p%20HEVC%2010bit%203M.mp4

aarnoud, 2025-05-13:
any progress or details needed for this ?

hbiyik, 2025-05-13:
i am convinced that this is a mesa alignment issue though.

wenyue7, 2025-05-15:
Hello, you can use the following patch test

`
diff --git a/mpp/base/mpp_sys_cfg.cpp b/mpp/base/mpp_sys_cfg.cpp
index dfcdd329..fe8725ce 100644
--- a/mpp/base/mpp_sys_cfg.cpp
+++ b/mpp/base/mpp_sys_cfg.cpp
@@ -443,6 +443,7 @@ MPP_RET mpp_sys_dec_buf_chk_proc(MppSysDecBufChkCfg *cfg)

         aligned_pixel_byte = cfg->h_stride_by_byte ? cfg->h_stride_by_byte :
                              aligned_pixel * depth / 8;
+        aligned_height = cfg->v_stride ? cfg->v_stride : aligned_height;

         switch (type) {
         case MPP_VIDEO_CodingHEVC : {
`

aarnoud, 2025-05-15:
> Hello, you can use the following patch test
>
> diff --git a/mpp/base/mpp_sys_cfg.cpp b/mpp/base/mpp_sys_cfg.cpp
> index dfcdd329..fe8725ce 100644
> --- a/mpp/base/mpp_sys_cfg.cpp
> +++ [...]

sadly, no changes on my end

wenyue7, 2025-05-15:
You can execute the command in the command line:
android: setprop mpp_sys_cfg_debug 0x10 && setprop vendor.mpp_sys_cfg_debug 0x10
linux:  export mpp_sys_cfg_debug=0x10

and then upload your log

aarnoud, 2025-05-15:
> I suspect that you haven't applied the patch properly. Can you add a log to verify that the mpp library is updated normally?

you were right, I may have tested this a bit too quickly. Good news is, that it look fixed now, thanks !

wenyue7, 2025-05-15:
Great!  Any other questions? Maybe could close the issue?

wenyue7, 2025-05-23:
https://github.com/rockchip-linux/mpp/commit/39e40c3aa8a58ca5ea6865527edd85bf8e724c75

closed 2025-05-23 by aarnoud