rockchip-linux/mpp #784: 3588s生成mp4花屏

yyazzz, 2025-02-18:
3588s读取两个usb摄像头
640*512红外usb摄像头yuyv①->5Menc->MP4；        ①->5Menc->rtsp
3840*2160 mjpeg usb摄像头->dec->yuv②->40Menc->mp4；    ②->rga rezise1920*1080->10Menc->rtsp
mp4文件出现花屏现象，这个从哪里排查。

yyazzz, 2025-02-18:
https://github.com/user-attachments/assets/85c7c94e-0ca0-4e96-bd97-2c9eeaaf8be6

yyazzz, 2025-02-18:
https://github.com/user-attachments/assets/5bc650fb-9bc6-4c01-987b-ac4d412e2d84

HermanChen, 2025-02-18:
这个是有编码线程因为信号退出了，上层有 ctrl+c 这种发信号的操作么？

yyazzz, 2025-02-18:
> 这个是有编码线程因为信号退出了，上层有 ctrl+c 这种发信号的操作么？

那上面的LOG应该跟花屏没关系，ctrl+c是我录完发送的

yyazzz, 2025-02-18:
但不确定是不是我发送的那个ctrl+c

HermanChen, 2025-02-18:
先把解码出来的 buffer 看一下有没有错，区分下问题是出在解码还是编码

yyazzz, 2025-02-18:
像热成像那边没解码的，USB直接出来的YUV也会花一下

yyazzz, 2025-02-19:
查看保存下来的h264，发现花屏的地方是P帧数据突然增大

[image]

HermanChen, 2025-02-19:
会是码流丢包导致的么？ 是不是 3840x2160 的 USB 解出来 jpeg 码流就有问题了，导致后面都有问题？

closed 2025-03-03 by HermanChen

yyazzz, 2025-03-04:
不是mpp问题，是代码问题