rockchip-linux/mpp #787: rk3588 运行mpp_runtime_test时提示无法打开kmpp_objs设备

hxhyyy, 2025-02-21:
2025-02-20 16:47:58.423  1893-1893  kmpp_obj                pid-1893                             E  kmpp_objs_init /dev/kmpp_objs open failed ret fd -1
2025-02-20 16:47:58.424  1893-1893  mpp_rt_test             pid-1893                             I  mpp found ion buffer is invalid
2025-02-20 16:47:58.424  1893-1893  mpp_rt_test             pid-1893                             I  mpp found drm buffer is valid

hxhyyy, 2025-02-21:
mpp_platform_test的log信息也会有这个错误，编码解码也无法正常运行。

2025-02-20 16:53:24.105  1910-1910  kmpp_obj                pid-1910                             E  kmpp_objs_init /dev/kmpp_objs open failed ret fd -1
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  kernel version: 5.10
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  ioctl  version: mpp_service
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  chip name: rockchip,rk3588-evb7-v11 rockchip,rk3588
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  
2025-02-20 16:53:24.105  1910-1910  mpp_plat_test           pid-1910                             I  chip vcodec type 100d321a
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found vpu2 codec
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found rkvdec decoder
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found avs+ decoder
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found rk jpeg decoder
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found rkvenc encoder
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  found vpu2 encoder
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  start probing decoder device name:
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  H.264 decoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  H.265 decoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  MJPEG decoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  VP9   decoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  avs+  decoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  start probing encoder device name:
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  H.264 encoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  H.265 encoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  MJPEG encoder: /dev/mpp_service
2025-02-20 16:53:24.106  1910-1910  mpp_plat_test           pid-1910                             I  mpp platform test done

HermanChen, 2025-02-21:
kmpp_objs_init 打不开这个没关系，这个是新的机制，还没实装，无视这个 log 就好

closed 2025-02-21 by HermanChen