rockchip-linux/mpp #788: mpi_enc_test  运行后会出现横纹

jiangchunyu, 2025-02-23:
./test/mpi_enc_test -i /dev/video1 -o test.h264 -w 640 -h 480 -f 8 -t 7
运行这个命令会出现横纹，是什么原因引起的？,以下这段代码只有mpp_buffer_sync_end没有mpp_buffer_sync_begin 是不是这个原因引起的？如果是的话想要加mpp_buffer_sync_begin，应该加在什么位置？

MppBuffer camera_frame_to_buf(CamSource *ctx, RK_S32 idx)
{
    MppBuffer buf = NULL;

    if (idx < 0)
        return buf;

    buf = ctx->fbuf[idx].buffer;
    if (buf)
        mpp_buffer_sync_end(buf);

    return buf;
}

HermanChen, 2025-03-03:
在硬件访问数据之前用 mpp_buffer_sync_end，把cpu访问后的数据写入到ddr

Li-Chang-Yi, 2025-03-18:
我的也是> ./test/mpi_enc_test -i /dev/video1 -o test.h264 -w 640 -h 480 -f 8 -t 7
> 运行这个命令会出现横纹，是什么原因引起的？,以下这段代码只有mpp_buffer_sync_end没有mpp_buffer_sync_begin 是不是这个原因引起的？如果是的话想要加mpp_buffer_sync_begin，应该加在什么位置？
>
> MppBuffer camera_frame_to_buf(CamSource *ctx, RK_S32 idx)
> {
>     MppBuffer [...]

我的也是，解决了吗，是 mpp 库版本问题吗？ 之前的 mpp 有这个问题吗？

Li-Chang-Yi, 2025-03-18:
怎么同步啊

HermanChen, 2025-03-18:
这样调用用应该是可以的
cpu 访问 ddr 数据之前加 start，之后加 end

Li-Chang-Yi, 2025-03-18:
上面就代码是 demo 的 camera_source 的函数源码，有问题该怎么改 没有横纹

cppcoffee, 2025-04-02:
有人解决了吗？我也碰到了运行 mpp 自带的 `mpi_enc_test` 程序，播放输出的文件，发现画面都是横纹的：

```shell
# ./test/mpi_enc_test -i /dev/video0 -o /tmp/a.h264 -t 7 -f 8 -w 1920 -h 1080
```

[image]

cppcoffee, 2025-04-02:
`/dev/video0` 设备的信息，使用 gstreamer 是可以采集的

```
# v4l2-ctl -d /dev/video0 --all
Driver Info:
        Driver name      : rk_hdmirx
        Card type        : rk_hdmirx
        Bus info         : fdee0000.hdmirx-controller
        Driver version   : 5.10.160
        Capabilities     : 0x84201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
Priority: 2
DV timings:
        Active width: 1920
        Active height: 1080
        Total width: 2200
        Total height: 1125
        Frame format: progressive
        Polarities: -vsync -hsync
        Pixelclock: 148348000 Hz (59.94 frames per second)
        Horizontal frontporch: 88
        Horizontal sync: 44
        Horizontal backporch: 148
        Vertical frontporch: 4
        Vertical sync: 5
        Vertical backporch: 36
        Standards:
        Flags:
DV timings capabilities:
        Minimum Width: 640
        Maximum Width: 4096
        Minimum Height: 480
        Maximum Height: 2160
        Minimum PClock: 20000000
        Maximum PClock: 600000000
        Standards: CTA-861
        Capabilities: Interlaced, Progressive
Format Video Capture Multiplanar:
        Width/Height      : 1920/1080
        Pixel Format      : 'NV24' (Y/CbCr 4:4:4)
        Field             : None
        Number of planes  : 1
        Flags             : premultiplied-alpha, 0x000000fe
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: xvYCC 601
        Quantization      : Default
        Plane 0           :
           Bytes per Line : 1920
           Size Image     : 6220800

Digital Video Controls

                  power_present 0x00a00964 (bitmask): max=0x00000001 default=0x00000000 value=1 flags=read-only
```