rockchip-linux/mpp #797: Stream and record video by ffmpeg

tranhien1612, 2025-03-06:
Hi, i am using the OrangePi5Plus and a camera to stream video via RTSP and record video by ffmpeg. The camera has /dev/device0.

With stream video via RTSP, i used mediantx and ffmpeg with mediantx_custom.yml file:
```
paths:
  stream:
    runOnInit: ffmpeg -f v4l2 -i /dev/video0 -c:v libx264 -preset ultrafast -b:v 600k -f rtsp rtsp://localhost:8554/stream
    runOnInitRestart: yes
```

And i record video by command:
```
ffmpeg -f v4l2 -input_format nv12 -video_size 640x512 -framerate 30 -i /dev/video0 -c:v libx264 -preset ultrafast -tune zerolatency -b:v 5M output.mp4
```

When i run one fuctions, the /dev/video is busy, so i don't run the next fuction.
I don't know how to run 2 functions at the same time. how can i do 2 tasks at the same time? Pls, help me solve this problem!

HermanChen, 2025-03-06:
Does the ffmpeg binary use mpp?

tranhien1612, 2025-03-06:
i am not using mpp at the moment. i am looking for a way to do 2 jobs at the same time. if mpp can do that i will use it