rockchip-linux/mpp #799: rock64 soc rk3288  开发板直接编译，测试解码失败

zp88899, 2025-03-12:
# uname -a
Linux rockpie 4.4.194-19-rockchip-g287ff75afc47 #1 SMP Wed Aug 4 15:04:27 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

mpp-1.0.5# ./osal/test/mpp_platform_test
mpp[14091]: mpp_plat_test: kernel version: 4.4
mpp[14091]: mpp_plat_test: ioctl  version: vcodec_service
mpp[14091]: mpp_plat_test:
mpp[14091]: mpp_plat_test: chip name: pine64,rock64 rockchip,rk3328
mpp[14091]: mpp_plat_test:
mpp[14091]: mpp_plat_test: chip vcodec type 0104020a
mpp[14091]: mpp_plat_test: found vpu2 codec
mpp[14091]: mpp_plat_test: found rkvdec decoder
mpp[14091]: mpp_plat_test: found vpu2 encoder
mpp[14091]: mpp_plat_test: found h265 stand-alone encoder
mpp[14091]: mpp_plat_test:
mpp[14091]: mpp_plat_test: start probing decoder device name:
mpp[14091]: mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp[14091]: mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp[14091]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[14091]: mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp[14091]: mpp_plat_test: avs+  decoder: /dev/vpu_service
mpp[14091]: mpp_plat_test:
mpp[14091]: mpp_plat_test: start probing encoder device name:
mpp[14091]: mpp_plat_test: H.264 encoder: /dev/vepu
mpp[14091]: mpp_plat_test: H.265 encoder: /dev/h265e
mpp[14091]: mpp_plat_test: MJPEG encoder: /dev/vepu
mpp[14091]: mpp_plat_test: mpp platform test done

mpp-1.0.5# ./osal/test/mpp_runtime_test
mpp[14092]: mpp_rt_test: mpp found ion buffer is invalid
mpp[14092]: mpp_rt_test: mpp found drm buffer is valid

#wget https://dl.radxa.com/media/video/1080p.264

测试h264视频
/mpp-1.0.5# ./test/mpi_dec_test -i /root/mpp/build/test/1080p.264  -t 7  -n 1
mpp[14095]: mpi_dec_utils: input file /root/mpp/build/test/1080p.264 size 10076324
mpp[14095]: mpi_dec_utils: cmd parse result:
mpp[14095]: mpi_dec_utils: input  file name: /root/mpp/build/test/1080p.264
mpp[14095]: mpi_dec_utils: output file name:
mpp[14095]: mpi_dec_utils: width      :    0
mpp[14095]: mpi_dec_utils: height     :    0
mpp[14095]: mpi_dec_utils: type       :    7
mpp[14095]: mpi_dec_utils: max frames :    1
mpp[14095]: mpi_dec_test: mpi_dec_test start
mpp[14095]: mpi: mpp_create enter ctx 0x7fc63190f8 mpi 0x7fc6319100
mpp[14095]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[14095]: mpi: mpp_create leave ret 0 ctx 0x559007b680 mpi 0x7f9b67b748
mpp[14095]: mpi_dec_test: 0x559007b680 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[14095]: mpi: mpp_init enter ctx 0x559007b680 type 0 coding 7
mpp[14095]: mpi: mpp_init leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_control enter ctx 0x559007b680 cmd 310202 parm 0x5590095440
mpp[14095]: mpi: mpi_control leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_control enter ctx 0x559007b680 cmd 310201 parm 0x5590095440
mpp[14095]: mpi: mpi_control leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
[... 112 lines cut ...]
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret -1012
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpi_decode_put_packet enter ctx 0x559007b680 packet 0x559007b4e8
mpp[14095]: mpi: mpi_decode_put_packet leave ctx 0x559007b680 ret -1012
mpp[14095]: mpi: mpi_decode_get_frame enter ctx 0x559007b680 frame 0x7f991577f0
mpp[14095]: mpi: mpi_decode_get_frame leave ctx 0x559007b680 ret 0
mpp[14095]: mpi_dec_test: 0x559007b680 decode get frame 0
mpp[14095]: mpi_dec_test: decode 1 frames time 38 ms delay  36 ms fps 25.96
mpp[14095]: mpi: mpi_reset enter ctx 0x559007b680
mpp[14095]: mpi: mpi_reset leave ctx 0x559007b680 ret 0
mpp[14095]: mpi: mpp_destroy enter ctx 0x559007b680
mpp[14095]: mpi: mpp_destroy leave ctx 0x559007b680 ret 0
mpp[14095]: mpi_dec_test: test success max memory 3.98 MB

同时内核有一条信息：dmesg|tail
 rk_vcodec: vpu_service_ioctl:2141: error: unknown vpu service ioctl cmd 40086c01

请帮忙看看是什么问题，感谢！

HermanChen, 2025-03-12:
> mpi_dec_test: 0x559007b680 decode get frame 0

解码是成功的

zp88899, 2025-03-12:
确实成功了，谢谢，我将264解码后到文件，然后再编码为VP8输出文件播放正常。
再想尝试AVS+解码，好像是不支持{mpp[6366]: mpi_dec_test: 0x55b90dff80 mpi_dec_test decoder test start w 0 h 0 type 16777221
mpp[6366]: mpp: unable to create dec avs+ for soc rk3328 unsupported}，但上面mpp_platform_test输出有支持AVS+，
改了源码mpp_soc.cpp
{
$ diff ./osal/mpp_soc.cpp ./osal/mpp_soc.cpp.bk
790,791c790,791
<         HAVE_VDPU2 | HAVE_VDPU2_PP | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_AVSDEC | HAVE_VEPU22,
<         {   &vdpu341, &vdpu2, &vdpu2_jpeg_pp, &avspd, NULL, NULL, },
---
>         HAVE_VDPU2 | HAVE_VDPU2_PP | HAVE_VEPU2 | HAVE_RKVDEC | HAVE_VEPU22,
>         {   &vdpu341, &vdpu2, &vdpu2_jpeg_pp, NULL, NULL, NULL, },
}想跳过支持SOC检测.

测试如下：
$ sudo test/mpi_dec_test -i ./cctv6-avs+.ts  -t 16777221  -n 30 -o out-cctv6+.yuv
mpp[7485]: mpi_dec_utils: input file ./cctv6-avs+.ts size 9091116
mpp[7485]: mpi_dec_utils: cmd parse result:
mpp[7485]: mpi_dec_utils: input  file name: ./cctv6-avs+.ts
mpp[7485]: mpi_dec_utils: output file name: out-cctv6+.yuv
mpp[7485]: mpi_dec_utils: width      :    0
mpp[7485]: mpi_dec_utils: height     :    0
mpp[7485]: mpi_dec_utils: type       : 16777221
mpp[7485]: mpi_dec_utils: max frames :   30
mpp[7485]: mpi_dec_test: mpi_dec_test start
mpp[7485]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[7485]: mpi_dec_test: 0x559e479f80 mpi_dec_test decoder test start w 0 h 0 type 16777221
mpp[7485]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[7485]: mpi_dec_test: 0x559e479f80 decode_get_frame get info changed found
mpp[7485]: mpi_dec_test: 0x559e479f80 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 0 err 3 discard 0
mpp[7485]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 1 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 2 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 3 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 4 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 5 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 6 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 7 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 8 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 9 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 10 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 11 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 12 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 13 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 14 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 15 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 16 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 17 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 18 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 19 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 20 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 21 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 22 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 23 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 24 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 25 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 26 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 27 err 0 discard 3
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 28 err 3 discard 0
mpp[7485]: mpi_dec_test: 0x559e479f80 decode get frame 29 err 0 discard 3
mpp[7485]: mpi_dec_test: decode 30 frames time 2086 ms delay 667 ms fps 14.38
mpp[7485]: mpi_dec_test: test success max memory 10.76 MB

这个是什么问题呢，谢谢，我用ffmpeg -i "" -map 0:v:0 -c:v copy test.ts这样导出的视频流来测试是会有问题吗

zp88899, 2025-03-12:
我再试了这样导出AVS+格式电视裸流：ffmpeg -i "http://xxxxxx" -c:v copy -an -f rawvideo cctv6+.raw，mpi_dec_test测试跟上面输出一样（-t 16777222也试了一样）：
/mpp$  sudo test/mpi_dec_test -i ./cctv6+.raw  -t 16777221  -n 10
mpp[7948]: mpi_dec_utils: input file ./cctv6+.raw size 8797501
mpp[7948]: mpi_dec_utils: cmd parse result:
mpp[7948]: mpi_dec_utils: input  file name: ./cctv6+.raw
mpp[7948]: mpi_dec_utils: output file name:
mpp[7948]: mpi_dec_utils: width      :    0
mpp[7948]: mpi_dec_utils: height     :    0
mpp[7948]: mpi_dec_utils: type       : 16777221
mpp[7948]: mpi_dec_utils: max frames :   10
mpp[7948]: mpi_dec_test: mpi_dec_test start
mpp[7948]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[7948]: mpi_dec_test: 0x5592bb2f80 mpi_dec_test decoder test start w 0 h 0 type 16777221
mpp[7948]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode_get_frame get info changed found
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[7948]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 0 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 1 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 2 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 3 err 0 discard 3
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 4 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 5 err 0 discard 3
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 6 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 7 err 0 discard 3
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 8 err 3 discard 0
mpp[7948]: mpi_dec_test: 0x5592bb2f80 decode get frame 9 err 0 discard 3
mpp[7948]: mpi_dec_test: decode 10 frames time 221 ms delay 149 ms fps 45.07
mpp[7948]: mpi_dec_test: test success max memory 10.76 MB

HermanChen, 2025-03-12:
AVSDEC 是 avs 解码器，avs+ 是 AVSPLUS_DEC

HermanChen, 2025-03-12:
3328 没有 avs+，是 3228h 有，看下 mpp_soc.cpp 里 avspd 设备在哪些芯片里

zp88899, 2025-03-12:
多谢！

closed 2025-03-12 by zp88899