rockchip-linux/mpp #801: mpp解码内存泄漏

circlefangzm, 2025-03-12:
大神好，
使用mpp解码usb摄像头，mjpeg格式，存在内存泄漏，随着时间推移，占用内存越来越多，挨个测试每个功能模块，确定是mpp库或相关应用层代码存在的问题，大概代码如下，
半内部分配模式，8个buffer的group，大小也固定。使用其中4个作为frame轮流解码显示（drmprimefdtohandle，drm部分也没问题），另外一个buffer固定做packet，捕获线程一旦捕获到mjpeg帧，就直接调用解码函数（没有新线程，不会重入），具体就是：拷贝到packet里面解码，使用poll/dequeue/task_set_packet/task_set_frame/enqueue/poll/dequeue，drmmodesetplane显示都很正常。frame轮流使用（没有互斥锁），packet和frame都不会deinit。
请问怎么定位这个内存泄漏的问题，应该都是dma_buf？能否提供些思路，谢谢

circlefangzm, 2025-03-12:
1080P，30fps，内存占用每3分钟大概长10m。

circlefangzm, 2025-03-12:
解码成nv12，输入输出task应该都是缺省的阻塞模式吧

HermanChen, 2025-03-12:
参考最新的 mpi_dec_test，用新的接口

circlefangzm, 2025-03-12:
谢谢，看到了，不过很遗憾只是develop分支，没有发布正式版本，不知道之前1.0.5这个正式版本能用这个方法不？
还有两个小问题：
1. 怎么看目前我系统上的mpp版本，试了MPP_VERSION变量等诸多方法，都没有。估计是1.0.5. 这个版本能支持最新的mpi_dec_test中所用的方法吗？如果不能，老版本是否存在这样的bug？如果没有我这个内存泄漏的原因可能是什么？

HermanChen, 2025-03-12:
目前没有 release 分支了，直接用 develop 吧

HermanChen, 2025-03-12:
泄漏的话，看下 fd 有没有在增长，有增长的话，估计是哪里的 MppBuffer 泄漏了

circlefangzm, 2025-03-12:
谢谢您。我的fd也是固定的4个轮流的，就是group里拿出来4个mppbuffer，根据fd得到的drm handle，不明白为啥会增长。怎么看fd有没有增长，我用valgrind ，还有别的好用的工具吗？

circlefangzm, 2025-03-13:
通过监看 /proc/进程id/fd，实时解码过程中无变化，固定为0~27

HermanChen, 2025-03-13:
fd 没泄漏的话，那就是一般性的 buffer 泄漏了，mpp_mem.cpp 里有个 mpp_mem_total_max，函数，一段时间调用一下看看有没有增长 mpp_mem_debug 要开 1

circlefangzm, 2025-03-13:
我这里需要重新编译mpp，系统里缺省可能没有编译或安装这个mpp_mem，但是我看/sys/kernel/debug/dma_buf/bufinfo，实时解码渲染显示过程中，是没有什么变化的，您看看下面粘贴的

Dma-buf Objects:
size            flags           mode            count           exp_name        ino
08355840        00000002        00080007        00000004        system-uncached 00000091        15421-httpsd
        Attached Devices:
        display-subsystem
Total 1 devices attached

08355840        00000002        00080007        00000004        system-uncached 00000090        15421-httpsd
        Attached Devices:
        display-subsystem
Total 1 devices attached

08355840        00000002        00080007        00000004        system-uncached 00000089        15421-httpsd
        Attached Devices:
        ff870000.jpegd
        display-subsystem
Total 2 devices attached

08355840        00000002        00080007        00000004        system-uncached 00000088        15421-httpsd
        Attached Devices:
        ff870000.jpegd
        display-subsystem
Total 2 devices attached

08355840        00000002        00080007        00000002        system-uncached 00000087        15421-httpsd
        Attached Devices:
        ff870000.jpegd
Total 1 devices attached

00004096        00000002        00080007        00000002        system-uncached 00000086        15421-httpsd
        Attached Devices:
        ff870000.jpegd
Total 1 devices attached

00524288        00000002        00080007        00000001        system-uncached 00000085        15421-httpsd
        Attached Devices:
Total 0 devices attached

Total 7 objects, 42307584 bytes

circlefangzm, 2025-03-13:
上面就是我程序占用的dma_buf，我虽然申请了8个mppbuffer，但是其用了其中5个，4个是frame轮流（目前只用了2个轮流，另外两个暂时空闲预留给rga用作翻转啥的），1个固定给packet。这里能得出mppbuffer没有泄漏吗？

circlefangzm, 2025-03-13:
我commit了8个mppbuffer（nv12，每个大小为1920*1088*4，可能实际比nv12要大一些）。

HermanChen, 2025-03-13:
看下最新方法中对 meta 的使用

circlefangzm, 2025-03-13:
就是最新的方法没太明白，看了doc、google、deepseek之类也不太明白，如果能详细讲下就好了。感谢!
另外如果/sys/kernel/debug/dma_buf/bufinfo，和fd，都没有泄漏的话，能说明不是buffer或fd的事情吗？我基本就是抄着dec的例子，加了drm显示，以及可能的rga旋转啥的（暂时没有设置）。如果把我程序里dec_advanced（copy图片内容到packet等，跟官方例子差不多，除了frame是循环使用的）屏蔽掉，就不泄漏了。

circlefangzm, 2025-03-13:
搞定了，packet也需要deinit

closed 2025-03-13 by circlefangzm