rockchip-linux/mpp #806: 请问一下mpi_enc_mt_test 的demo

huochen1, 2025-03-25:
./mpi_enc_mt_test -i /dev/video0 -o test_bgr3.h264 -w 3840 -h 2160 -t 7 -f 65543 -fps 60 -n  3600
运行这个没有反应但是
如果执行这个demo的话./mpi_enc_test -i /dev/video0 -o test_bgr3.h264 -w 3840 -h 2160 -t 7 -f 65543 -fps 60 -n  3600
他是可以正常编码的

HermanChen, 2025-03-25:
export mpp_syslog_perror=1 再跑

huochen1, 2025-03-25:
[image]是的已经开启的了，这个是./mpi_enc_mt_test -i /dev/video0 -o test_bgr3.h264 -w 3840 -h 2160 -t 7 -f 65543 -fps 60 -n 3600的他一到达这个就不玩下执行了，如果是mpi_enc_test的话就会执行下去

[image]

huochen1, 2025-03-25:
如果不加-i的话这个mpi_enc_mt_test就可以跑下去，是不支持video设备吗

HermanChen, 2025-03-26:
不加 -i 是 cpu 画图来编码
-i /dev/video0 看下错误 log

huochen1, 2025-03-26:
[image]额没有显示有错误的，就是执行不下去了不知是否在等待数据

huochen1, 2025-03-26:
[image]确实这个mpi_enc_test可以执行下去

HermanChen, 2025-03-26:
mt 的 buffer 轮转和释放可能不完善

huochen1, 2025-03-26:
这样