rockchip-linux/mpp #809: 3588 mpp编码 码率控制稳定

FanPeterTT, 2025-03-27:
平台 3588
mpp 1.0.7
程序 mpi_enc_mt_test 
我在示例程序基础上增加了实时打印码率的日志。发现在cbr模式(-rc 1)下码率很设置的不一致，反而在vbr(-rc 0)模式模式下码率和设置基本符合。好像和定义不太一致呢，cbr应该是固定码率啊，请指教一下

[image]

FanPeterTT, 2025-03-31:
> 平台 3588 mpp 1.0.7 程序 mpi_enc_mt_test 我在示例程序基础上增加了实时打印码率的日志。发现在cbr模式(-rc 1)下码率很设置的不一致，反而在vbr(-rc 0)模式模式下码率和设置基本符合。好像和定义不太一致呢，cbr应该是固定码率啊，请指教一下
> 
> [image]

HermanChen, 2025-04-01:
把 rc_debug 开起来看下 log

FanPeterTT, 2025-04-01:
> 把 rc_debug 开起来看下 log

这是rc_debug的日志

[image]

[image]

HermanChen, 2025-04-02:
rc_debug 有好几个选项，开 bps 的选项

FanPeterTT, 2025-04-02:
> rc_debug 有好几个选项，开 bps 的选项
这是cbr模式编码信息，麻烦您看一下

[image]

HermanChen, 2025-04-03:
提供下文本的 log，这个截图很不好看

FanPeterTT, 2025-04-03:
> 提供下文本的 log，这个截图很不好看

这是百度网盘链接，麻烦您查看一下！
https://pan.baidu.com/s/1jdnviP_HebOzsoQ0KG3XZw?pwd=1111

Liaoyyyy, 2025-04-03:
你好，可以把测试源也放网盘一起发下，我们本地复现看看

Liaoyyyy, 2025-04-03:
相机源的话就不用了，我们本地试试就好

FanPeterTT, 2025-04-15:
换了SDK里面的mpp库，就解决了

closed 2025-04-15 by FanPeterTT