rockchip-linux/mpp #810: rk3566 mpp解码jpg图片错误

xlwlp, 2025-03-27:
decode_put_packet返回值是1，这个错误我在头文件里没找到
typedef enum {
    MPP_SUCCESS                 = RK_SUCCESS,
    MPP_OK                      = RK_OK,

    MPP_NOK                     = -1,
    MPP_ERR_UNKNOW              = -2,
    MPP_ERR_NULL_PTR            = -3,
    MPP_ERR_MALLOC              = -4,
    MPP_ERR_OPEN_FILE           = -5,
    MPP_ERR_VALUE               = -6,
    MPP_ERR_READ_BIT            = -7,
    MPP_ERR_TIMEOUT             = -8,
    MPP_ERR_PERM                = -9,

    MPP_ERR_BASE                = -1000,

    /* The error in stream processing */
    MPP_ERR_LIST_STREAM         = MPP_ERR_BASE - 1,
    MPP_ERR_INIT                = MPP_ERR_BASE - 2,
    MPP_ERR_VPU_CODEC_INIT      = MPP_ERR_BASE - 3,
    MPP_ERR_STREAM              = MPP_ERR_BASE - 4,
    MPP_ERR_FATAL_THREAD        = MPP_ERR_BASE - 5,
    MPP_ERR_NOMEM               = MPP_ERR_BASE - 6,
    MPP_ERR_PROTOL              = MPP_ERR_BASE - 7,
    MPP_FAIL_SPLIT_FRAME        = MPP_ERR_BASE - 8,
    MPP_ERR_VPUHW               = MPP_ERR_BASE - 9,
    MPP_EOS_STREAM_REACHED      = MPP_ERR_BASE - 11,
    MPP_ERR_BUFFER_FULL         = MPP_ERR_BASE - 12,
    MPP_ERR_DISPLAY_FULL        = MPP_ERR_BASE - 13,
} MPP_RET;
请问这个是什么错误，通过errno查看到的信息是No such file or directory

HermanChen, 2025-03-27:
更新下版本看看

xlwlp, 2025-03-27:
更新mpp库的版本吗？现在解码单张jpg图片，是参考mpi_dec_test.c文件吧？

HermanChen, 2025-03-27:
是的，整体更新下

HermanChen, 2025-03-27:
新 IP 不支持了，要转格式需要用 rga

xlwlp, 2025-03-28:
mpp解码jpg图片默认输出格式应该是MPP_FMT_YUV420SP吧？我保存下来的图片用YUView看颜色不对么，图片分辨率是2240x1024的，
width    = mpp_frame_get_width(frame);
height   = mpp_frame_get_height(frame);
h_stride = mpp_frame_get_hor_stride(frame);
v_stride = mpp_frame_get_ver_stride(frame);
fmt      = mpp_frame_get_fmt(frame);
bufsize = mpp_frame_get_buf_size(frame);
输出的frame信息：
frame info : width=2240, height=1024, size=4587520, h_stride=2240, v_stride=1024, fmt=0
为什么bufsize会是4587520；
我把输出格式设为MPP_FMT_RGB888，保存的图片颜色显示就是正常的，输出frame信息：
frame info : width=2240, height=1024, size=4587520, h_stride=6720, v_stride=1024, fmt=65542
为什么bufsize还是4587520

HermanChen, 2025-03-28:
jpeg 默认分的 RGBX 大小的 buffer

xlwlp, 2025-03-28:
那2240x1024这个分辨率， size=4587520，这个buf小了吧？

HermanChen, 2025-03-29:
这个刚好是 2 倍，对 RGBX 的确是小了

xlwlp, 2025-03-29:
那现在mpp解码输出格式只能支持yuv，无法支持rgb或rgbx格式？可我设置输出rgb888保存下来文件却可以正常显示，反而yuv420sp显示颜色却不对，这是什么原因？

xlwlp, 2025-03-31:
可以正常显示了，格式配置错了

xlwlp, 2025-03-31:
奇怪的是输出格式设为MPP_FMT_RGB888后，为什么frame的buf大小还是4587520，这明显偏小了，感觉这个大小固定的不会变，可通过dump_mpp_frame_to_file保存的rgb文件却能正常显示