rockchip-linux/mpp #811: 3588 mpp 编码 800x600 (stride 设置为832) 的NV12图像 异常

nazjat-ar, 2025-03-27:
参考mpp_enc_test 写的demo 。 输入是一个 832x600 的 nv12 buffer，其中一行有效宽度为 800 .所以将enc cfg 中的 width 设置为800， hor_stride 设置为 832， height/ver_stride 设置为 600。得到的H264码流 异常。

HermanChen, 2025-03-28:
直接用 mpi_enc_test 编码对应的分辨率，打印看下配置的宽高 stirde

nazjat-ar, 2025-03-28:
直接用mpi_enc_test 设置了-hstride 832 ，效果如下花屏

[image]

nazjat-ar, 2025-03-28:
执行日志如下：
# ./mpi_enc_test -i 800x600.nv12  -w 800 -h 600 -hstride 832  -o enc_out.h264
mpi_enc_utils: cmd parse result:
mpi_enc_utils: input  file name: 800x600.nv12
mpi_enc_utils: output file name: enc_out.h264
mpi_enc_utils: width      : 800
mpi_enc_utils: height     : 600
mpi_enc_utils: format     : 0
mpi_enc_utils: type       : 7
mpi_enc_test: ./mpi_enc_test start
mpp_info: mpp version: aaa4c8e9 author: Chandler Chen  2024-10-29 fix[hal_vdpu383]: fix fbc hor_stride mismatch issue
mpi_enc_test: 0x7faf7b12a0 encoder test start w 800 h 600 type 7
mpp_platform: client 12 driver is not ready!
mpp_enc: MPP_ENC_SET_RC_CFG bps 1800000 [112500 : 1912500] fps [30:30] gop 60
h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [800:600] stride [832:600]
mpp_enc: mode vbr bps [112500:1800000:1912500] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
rc: using rc impl default
mpp_enc: mode vbr bps [112500:1800000:1912500] fps fix [30/1] -> fix [30/1] gop i [60] v [0]

HermanChen, 2025-03-28:
看下 mpi_enc_test 里读文件的方式，需要做修改，stride 和 width 要和内存里的实际情况匹配

closed 2025-03-28 by nazjat-ar