rockchip-linux/mpp #812: gstreamer和mpp,如何使用可以解码AVS+

zp88899, 2025-03-28:
gst-launch-1.0 filesrc  location=./cctv6+.raw  ! parsebin ! mppvideodec ! autovideosink
mpp[2976]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[2976]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[2976]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[2976]: mpp: unable to create enc vp8 for soc rk3528a unsupported
mpp[2976]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
Setting pipeline to PAUSED ...
mpp[2976]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
Pipeline is PREROLLING ...
mpp[2976]: mpg4d_parser: unsupported shape 1
mpp[2976]: mpg4d_parser: unsupported shape 1
mpp[2976]: mpg4d_parser: unsupported shape 1
mpp[2976]: mpg4d_parser: unsupported shape 2
mpp[2976]: mpg4d_parser: Warning: unsupport larger than 1920x1088
mpp[2976]: mpg4d_parser: unsupported shape 3

它把视频格式识别错为mpg4了

zp88899, 2025-03-28:
mpp本身解码AVS+是可以的
$mpi_dec_test -i ~/cctv6+.raw -t 16777221  -o cctv6+.out
mpp[28825]: mpi_dec_utils: input file /home/cat/cctv6+.raw size 8797501
mpp[28825]: mpi_dec_utils: cmd parse result:
mpp[28825]: mpi_dec_utils: input  file name: /home/cat/cctv6+.raw
mpp[28825]: mpi_dec_utils: output file name: cctv6+.out
mpp[28825]: mpi_dec_utils: width      :    0
mpp[28825]: mpi_dec_utils: height     :    0
mpp[28825]: mpi_dec_utils: type       : 16777221
mpp[28825]: mpi_dec_utils: max frames :    0
mpp[28825]: mpi_dec_test: mpi_dec_test start
mpp[28825]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[28825]: mpi_dec_test: 0x556e244120 mpi_dec_test decoder test start w 0 h 0 type 16777221
mpp[28825]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[28825]: mpi_dec_test: 0x556e244120 decode_get_frame get info changed found
mpp[28825]: mpi_dec_test: 0x556e244120 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[28825]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 0
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 1
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 2
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 3
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 4
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 5
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 6
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 7
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 8
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 9
mpp[28825]: mpi_dec_test: 0x556e244120 decode get frame 10

HermanChen, 2025-03-29:
gst 里的格式识别问题

zp88899, 2025-03-29:
谢谢！我在gst加了格式识别，
$ diff gstmppvideodec.c gstmppvideodec.c.old 
68d67
<         "video/x-avs, parsed = (boolean) true, alignment = (string) au;"
93,95d91
< 
<   if (gst_structure_has_name (s, "video/x-avs"))
<     return MPP_VIDEO_CodingAVSPLUS;

还是报错：
$ gst-launch-1.0 filesrc location=./cctv6+.raw ! video/x-avs,width=1920,height=1080,framerate=25/1 ! mppvideodec ! autovideosink
mpp[6867]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[6867]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[6867]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[6867]: mpp: unable to create enc vp8 for soc rk3528a unsupported
mpp[6867]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
Setting pipeline to PAUSED ...
mpp[6867]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
Pipeline is PREROLLING ...
Caught SIGSEGV
exec gdb failed: No such file or directory

zp88899, 2025-03-29:
gdb调试看了一下，请参考。
Starting program: /usr/bin/gst-launch-1.0 filesrc location=../cctv6+.raw \! video/x-avs,width=1920,height=1080,framerate=25/1 \! mppvideodec \! autovideosink
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
mpp[12594]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
[New Thread 0x7fb430a1b0 (LWP 12597)]
mpp[12594]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
[Thread 0x7fb430a1b0 (LWP 12597) exited]
[New Thread 0x7fb430a1b0 (LWP 12598)]
[Thread 0x7fb430a1b0 (LWP 12598) exited]
mpp[12594]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[12594]: mpp: unable to create enc vp8 for soc rk3528a unsupported
mpp[12594]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
[New Thread 0x7fb430a1b0 (LWP 12599)]
[Thread 0x7fb430a1b0 (LWP 12599) exited]
Setting pipeline to PAUSED ...
[New Thread 0x7fb430a1b0 (LWP 12600)]
[Thread 0x7fb430a1b0 (LWP 12600) exited]
[New Thread 0x7fb430a1b0 (LWP 12601)]
mpp[12594]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
[New Thread 0x7fb389d1b0 (LWP 12602)]
Pipeline is PREROLLING ...
[New Thread 0x7fb309c1b0 (LWP 12603)]
[New Thread 0x7fb269a1b0 (LWP 12604)]
[New Thread 0x7fb1e991b0 (LWP 12605)]

Thread 9 "filesrc0:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fb269a1b0 (LWP 12604)]
0x0000007fb46687c0 in avsd_control (decoder=0x7fa4016800, cmd_type=MPP_DEC_SET_DISABLE_ERROR, param=0x0)
    at /home/cat/mpp/mpp/codec/dec/avs/avsd_api.c:183
183	        p_dec->disable_error = *((RK_U32 *)param);
(gdb)

zp88899, 2025-03-29:
(gdb) bt
#0  0x0000007fb46687c0 in avsd_control (decoder=0x7fa4016800, cmd_type=MPP_DEC_SET_DISABLE_ERROR, param=0x0)
    at /home/cat/mpp/mpp/codec/dec/avs/avsd_api.c:183
#1  0x0000007fb461f60c in mpp_parser_control(Parser, MpiCmd, void*)
    (prs=0x7fa4016780, cmd=MPP_DEC_SET_DISABLE_ERROR, para=0x0) at /home/cat/mpp/mpp/codec/mpp_parser.cpp:222
#2  0x0000007fb461bf28 in mpp_dec_proc_cfg(MppDecImpl*, MpiCmd, void*)
    (dec=0x7fa400aca0, cmd=MPP_DEC_SET_DISABLE_ERROR, param=0x0) at /home/cat/mpp/mpp/codec/mpp_dec.cpp:105
#3  0x0000007fb462e1e8 in mpp_dec_parser_thread(void*) (data=0x55aa5d7b80)
    at /home/cat/mpp/mpp/codec/mpp_dec_normal.cpp:734
#4  0x0000007fb5e5c648 in start_thread () at /lib/aarch64-linux-gnu/libpthread.so.0
#5  0x0000007fb5db301c in  () at /lib/aarch64-linux-gnu/libc.so.6
(gdb)

zp88899, 2025-03-29:
使用http直播流，不太稳定，运行一段时间，好像有问题：
$ dmesg|tail
[ 1262.049509] rk_vcodec: timing: run end        : invalid
[ 1262.049525] rk_vcodec: timing: irq            : invalid
[ 1262.049541] rk_vcodec: timing: timeout cancel : invalid
[ 1262.049557] rk_vcodec: timing: isr            : invalid
[ 1262.049573] rk_vcodec: timing: finish         : invalid
[ 1262.049609] mpp_vdpu1 ff7c1000.avsd_plus: resetting...
[ 1262.050734] rk_iommu ff7c0800.iommu: Enable stall request timed out, retry_count = 0, status: 0x000189
[ 1262.051827] rk_iommu ff7c0800.iommu: Disable paging request timed out, retry_count = 0, status: 0x000189
[ 1262.052979] rk_iommu ff7c0800.iommu: Enable stall request timed out, retry_count = 0, status: 0x000189
[ 1262.053015] mpp_vdpu1 ff7c1000.avsd_plus: reset done

zp88899, 2025-03-29:
这个开始正常，现在也变得不正常了
$  ~/mpp/build/linux/aarch64/test/mpi_dec_test -i ~/cctv6+.raw -t 16777221 -n 3
mpp[3758]: mpi_dec_utils: input file /home/cat/cctv6+.raw size 8797501
mpp[3758]: mpi_dec_utils: cmd parse result:
mpp[3758]: mpi_dec_utils: input  file name: /home/cat/cctv6+.raw
mpp[3758]: mpi_dec_utils: output file name: 
mpp[3758]: mpi_dec_utils: width      :    0
mpp[3758]: mpi_dec_utils: height     :    0
mpp[3758]: mpi_dec_utils: type       : 16777221
mpp[3758]: mpi_dec_utils: max frames :    3
mpp[3758]: mpi_dec_test: mpi_dec_test start
mpp[3758]: mpp_info: mpp version: 2f2e5f5c author: Yandong Lin   2025-03-10 fix[kmpp]: Fix channel dup issue
mpp[3758]: mpi_dec_test: 0x5598091120 mpi_dec_test decoder test start w 0 h 0 type 16777221
mpp[3758]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[3758]: mpi_dec_test: 0x5598091120 decode_get_frame get info changed found
mpp[3758]: mpi_dec_test: 0x5598091120 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 3760128
mpp[3758]: mpp_buf_slot: mismatch size_total 4177920 - 3760128
mpp[3758]: mpi_dec_test: 0x5598091120 decode get frame 0 err 3 discard 0
mpp[3758]: mpi_dec_test: 0x5598091120 decode get frame 1 err 3 discard 0
mpp[3758]: mpi_dec_test: 0x5598091120 decode get frame 2 err 3 discard 0
mpp[3758]: mpi_dec_test: decode 3 frames time 1628 ms delay 1093 ms fps 1.84
mpp[3758]: mpi_dec_test: test success max memory 10.76 MB
$ dmesg|tail
[ 1497.014953] rk_vcodec: timing: run end        : invalid
[ 1497.014969] rk_vcodec: timing: irq            : invalid
[ 1497.014986] rk_vcodec: timing: timeout cancel : invalid
[ 1497.015002] rk_vcodec: timing: isr            : invalid
[ 1497.015018] rk_vcodec: timing: finish         : invalid
[ 1497.015055] mpp_vdpu1 ff7c1000.avsd_plus: resetting...
[ 1497.016184] rk_iommu ff7c0800.iommu: Enable stall request timed out, retry_count = 0, status: 0x000189
[ 1497.017280] rk_iommu ff7c0800.iommu: Disable paging request timed out, retry_count = 0, status: 0x000189
[ 1497.018434] rk_iommu ff7c0800.iommu: Enable stall request timed out, retry_count = 0, status: 0x000189
[ 1497.018480] mpp_vdpu1 ff7c1000.avsd_plus: reset done

HermanChen, 2025-04-01:
disable_error 只是关报错，并不解决问题

zp88899, 2025-04-01:
嗯，谢谢，头晕，先不搞了

closed 2025-04-03 by zp88899