rockchip-linux/mpp #813: RV1126 MPI和RKmedia 解码 mjpeg失败

twinkle311, 2025-03-31:
采集单帧保存文件看是正常的，

[image]

[image]

rkmedia报错输入不是jpeg格式， MPI没有报错，但是dumpsys 可以看到都是 format error，
```cpp
void TJVideoDecoder::init()
{
	RK_MPI_SYS_Init();
	/*VDEC_MOD_PARAM_S vdecModParam{};
	vdecModParam.enVdecMBSource = MB_SOURCE_PRIVATE;
	vdecModParam.u32MiniBufMode = {};
	vdecModParam.u32ParallelMode = {};
	vdecModParam.stVideoModParam = {};
	RK_S32 ret0 = RK_MPI_VDEC_SetModParam(&vdecModParam);
	if (ret0 != 0)
	{
		cout << "RK_MPI_VDEC_SetModParam failed" << endl;
	}*/
	VDEC_CHN_ATTR_S vdecChnAttr{};
	vdecChnAttr.enType = RK_VIDEO_ID_MJPEG;
	vdecChnAttr.enMode = VIDEO_MODE_FRAME;
	vdecChnAttr.u32PicWidth = VIDEO_WIDTH;
	vdecChnAttr.u32PicHeight = VIDEO_HEIGHT;
	vdecChnAttr.u32PicVirWidth = VIDEO_WIDTH;
	vdecChnAttr.u32PicVirHeight = VIDEO_HEIGHT;
	vdecChnAttr.u32StreamBufSize = VIDEO_WIDTH * VIDEO_HEIGHT *  2;
	vdecChnAttr.u32StreamBufCnt = V4L2_MAX_BUFFER_COUNT;
	vdecChnAttr.u32FrameBufSize = 0; // 0 means use auto size
	vdecChnAttr.u32FrameBufCnt = V4L2_MAX_BUFFER_COUNT;
	RK_S32 ret = RK_MPI_VDEC_CreateChn(0, &vdecChnAttr);
	if (ret != 0)
	{
		cout << "RK_MPI_VDEC_CreateChn failed" << endl;
	}
	VDEC_CHN_PARAM_S vdecChnParam{};
	vdecChnParam.enType = RK_VIDEO_ID_MJPEG;
	vdecChnParam.u32DisplayFrameNum = 0;
	vdecChnParam.stVdecPictureParam = {RK_FMT_YUV420SP, 0};
	ret = RK_MPI_VDEC_SetChnParam(0,&vdecChnParam);
	if (ret != 0)
	{
		cout << "RK_MPI_VDEC_SetChnParam failed" << endl;
	}
	start();
}

void TJVideoDecoder::start()
{
	RK_MPI_VDEC_StartRecvStream(0);
}
void TJVideoDecoder::deinit()
{
	RK_MPI_VDEC_StopRecvStream(0);
	RK_MPI_VDEC_DestroyChn(0);
}
```

twinkle311, 2025-03-31:
SDK版本 RV1126 1109 linux SDK V3.0.4 20240531

HermanChen, 2025-04-01:
sdk 版本更新下，rkmedia 是比较早期的中间件，后面升级成 rockit 了

twinkle311, 2025-04-01:
是送编码器的长度值没有赋值，已解决