rockchip-linux/mpp #820: Failure diagnostic messages are showing in 1.0.9

dbermond, 2025-04-05:
When running FFmpeg compiled with mpp 1.0.9, failure diagnostic messages are shown right before the normal FFmpeg output:

```
mpp[7679]: kmpp_obj: kmpp_objdef_get invalid param def 0x7a5c6c444f30 name 0x7a5c6c3bf4fd objs (nil)
mpp[7679]: kmpp_frame: kmpp_frame_init KMPP_OBJ_NAME init failed
```

This does not happen when building FFmpeg with mpp 1.0.8 or earlier.

Operating System: Arch Linux

HermanChen, 2025-04-07:
This log is removed on latest version. It is just a warning.

brauliobo, 2025-06-08:
got the same here

dbermond, 2025-06-29:
Still present in version 1.0.10, but now with different messages:

```
mpp[46065]: mpp_soc: open /proc/device-tree/compatible error
mpp[46065]: mpp_platform: can not found match soc name:
```