rockchip-linux/mpp #821: RV1126 使用rkmedia库vi能正常取图， 使用rockit库无法取图

twinkle311, 2025-04-07:
rkmedia 
```cpp
int TJVideoDevice::init()
{
	uartInit();
	setMipiOutput();
	sleep(2);
	RK_MPI_SYS_Init();
	int ret = 0;
	VI_CHN_ATTR_S vi_chn_attr = {};
	vi_chn_attr.pcVideoNode = m_pDeviceName.c_str();
	vi_chn_attr.u32BufCnt = 3;
	vi_chn_attr.u32Width = 1280;
	vi_chn_attr.u32Height = 1024;
	vi_chn_attr.enPixFmt = IMAGE_TYPE_YUYV422;
	vi_chn_attr.enWorkMode = VI_WORK_MODE_NORMAL;
	vi_chn_attr.enBufType = VI_CHN_BUF_TYPE_MMAP;
	ret = RK_MPI_VI_SetChnAttr(m_camId, 0, &vi_chn_attr);
	ret |= RK_MPI_VI_EnableChn(m_camId, 0);
	if (ret)
	{
		cout << "Create VI[0] failed! ret=" << ret << endl;
		return -1;
	}
	return 0;
}
```
rockit
```cpp
int TJVideoDevice::init()
{
	uartInit();
	setMipiOutput();
	sleep(3);
	RK_MPI_SYS_Init();
	// return 0;
	VI_DEV_ATTR_S devAttr{};
	devAttr.enIntfMode = VI_MODE_MIPI;
	devAttr.enWorkMode = VI_WORK_MODE_1Multiplex;
	devAttr.enDataSeq = VI_DATA_SEQ_VUVU;
	devAttr.enInputDataType = VI_DATA_TYPE_RGB;
	devAttr.stMaxSize = {4096, 4096};
	devAttr.enDataRate = DATA_RATE_X1;
	RK_S32 ret = RK_FAILURE;
	/*RK_S32 ret = RK_MPI_VI_SetDevAttr(0, &devAttr);
	if (ret != 0)
	{
		cerr << "RK_MPI_VI_SetDevAttr failed" << endl;
	}*/
	ret = RK_MPI_VI_GetDevAttr(1, &devAttr);
	if (ret == RK_ERR_VI_NOT_CONFIG)
	{
		// 0-1.config dev
		ret = RK_MPI_VI_SetDevAttr(1, &devAttr);
		if (ret != RK_SUCCESS)
		{
			RK_LOGE("RK_MPI_VI_SetDevAttr %x", ret);
			cerr << "RK_MPI_VI_SetDevAttr error" << endl;
			// goto __FAILED1;
		}
	}
	else
	{
		RK_LOGE("RK_MPI_VI_SetDevAttr already");
		cout << "RK_MPI_VI_SetDevAttr already" << endl;
	}
	// 1.get dev enable status
	ret = RK_MPI_VI_GetDevIsEnable(1);
	if (ret != RK_SUCCESS)
	{
		// 1-2.enable dev
		ret = RK_MPI_VI_EnableDev(1);
		if (ret != RK_SUCCESS)
		{
			RK_LOGE("RK_MPI_VI_EnableDev %x", ret);
			cerr << "RK_MPI_VI_EnableDev error" << endl;
		}
		// 1-3.bind dev/pipe
		VI_DEV_BIND_PIPE_S devBindPipe{};
		devBindPipe.u32Num = 1;
		devBindPipe.PipeId[0] = 1;
		ret = RK_MPI_VI_SetDevBindPipe(1, &devBindPipe);
		if (ret != RK_SUCCESS)
		{
			RK_LOGE("RK_MPI_VI_SetDevBindPipe %x", ret);
			cerr << "RK_MPI_VI_SetDevBindPipe error" << endl;
		}
	}
	else
	{
		RK_LOGE("RK_MPI_VI_EnableDev already");
		cout << "RK_MPI_VI_EnableDev already" << endl;
	}
	/*ret = RK_MPI_VI_EnableDev(0);
	if (ret != 0)
	{
		cerr << "RK_MPI_VI_EnableDev failed" << endl;
	}
	VI_DEV_BIND_PIPE_S devBindPipe{};
	devBindPipe.u32Num = 0;
	devBindPipe.PipeId[0] = 0;
	ret = RK_MPI_VI_SetDevBindPipe(0, &devBindPipe);
	if (ret != 0)
	{
		cerr << "RK_MPI_VI_SetDevBindPipe failed" << endl;
	}*/
	VI_CHN_ATTR_S chnAttr{};
	chnAttr.stSize = {VIDEO_WIDTH, VIDEO_HEIGHT};
	chnAttr.enPixelFormat = RK_FMT_YUV422_YUYV;
	chnAttr.enDynamicRange = DYNAMIC_RANGE_SDR8;
	chnAttr.enVideoFormat = VIDEO_FORMAT_LINEAR;
	chnAttr.enCompressMode = COMPRESS_MODE_NONE;
	chnAttr.bMirror = RK_FALSE;
	chnAttr.bFlip = RK_FALSE;
	chnAttr.u32Depth = 6;
	chnAttr.stFrameRate = {-1, -1};
	// chnAttr.enAllocBufType = VI_ALLOC_BUF_TYPE_EXTERNAL;
	chnAttr.enAllocBufType = VI_ALLOC_BUF_TYPE_INTERNAL;
	chnAttr.stIspOpt.u32BufCount = 6;
	chnAttr.stIspOpt.u32BufSize = VIDEO_WIDTH * VIDEO_HEIGHT * 2;
	chnAttr.stIspOpt.enCaptureType = VI_V4L2_CAPTURE_TYPE_VIDEO_CAPTURE_MPLANE;
	// chnAttr.stIspOpt.enMemoryType = VI_V4L2_MEMORY_TYPE_MMAP;
	chnAttr.stIspOpt.enMemoryType = VI_V4L2_MEMORY_TYPE_DMABUF;
	snprintf(chnAttr.stIspOpt.aEntityName, sizeof(chnAttr.stIspOpt.aEntityName), m_pDeviceName.c_str());
	chnAttr.stIspOpt.bNoUseLibV4L2 = RK_FALSE;
	chnAttr.stIspOpt.stMaxSize = {4096, 4096};
	ret = RK_MPI_VI_SetChnAttr(1, 1, &chnAttr);
	if (ret != RK_SUCCESS)
	{
		cerr << "RK_MPI_VI_SetChnAttr failed" << endl;
	}
	/*ret = RK_MPI_VI_DisableChn(1,1);
	if (ret != RK_SUCCESS)
	{
		cerr << "RK_MPI_VI_DisableChn failed" << endl;
	}*/
	ret = RK_MPI_VI_EnableChn(1, 1);
	if (ret != RK_SUCCESS)
	{
		cerr << "RK_MPI_VI_EnableChn failed" << endl;
	}
	return 0;
}
```
使用rockit库取图， dumpsys vi 可以看到 vbfail字段一直增加

twinkle311, 2025-04-07:
使用 v4l2 取图正常

HermanChen, 2025-04-07:
mpp 之外的问题提到 redmine 上吧……

twinkle311, 2025-04-07:
> mpp 之外的问题提到 redmine 上吧……

麻烦给个链接

closed 2025-04-07 by twinkle311