rockchip-linux/mpp #826: 运行mpi_decode_test出现out of memory

laborer996, 2025-04-10:
日志如下：
```bash
4-11 02:16:27.606   705  5343 I ActivityManager:   ntv   ?? 7183546: mpi_dec_test (pid 5340) native
04-11 02:16:36.254  5340  5340 W         : mpi_dec_test invoked oom-killer: gfp_mask=0x8140dca(GFP_HIGHUSER_MOVABLE|__GFP_COMP|__GFP_ZERO|__GFP_CMA), order=0, oom_score_adj=-950
04-11 02:16:36.254  5340  5340 W         : CPU: 4 PID: 5340 Comm: mpi_dec_test Tainted: G           O       6.1.75 #1
04-11 02:16:36.255  5340  5340 I         : [   5340]     0  5340  6869087  1853439 22454272   932811          -950 mpi_dec_test
04-11 02:16:36.255  5340  5340 I oom-kill: constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=mpi_dec_test,pid=5340,uid=0
04-11 02:16:36.255  5340  5340 E         : Out of memory: Killed process 5340 (mpi_dec_test) total-vm:27476348kB, anon-rss:7412896kB, file-rss:608kB, shmem-rss:252kB, UID:0 pgtables:21928kB oom_score_adj:-950
```

解码视频为1920*1080p@20fps， 总帧数20.
启动命令为：
```bash
./mpi_dec_test  -t 7 -i output_n20_1920x1080_20fps.h264 -w 1920 -h 1080  -n 1
```

系统为Android 14，内存为16GB大小，编译版本为arm64-v8a

HermanChen, 2025-04-11:
-n 1 是只解码一帧啊，应该不会说有问题的，是不是系统内存已经很紧张了，解码器只是最后一根稻草？
还是说这个码流有什么特别？传上来看下

oneflyingfish, 2025-04-11:
码流是通过ubuntu的ffmpeg生成的
```bash
ffmpeg -f lavfi -i testsrc=size=1920x1080:rate=20 -c:v libx264 -preset veryfast -crf 23 -pix_fmt yuv420p -frames:v 20 -f h264 output.h264
```

系统通过free -m查看还有7.GB以上内存可用，而且板卡没有跑什么程序，我只是在测试mpp可用性

oneflyingfish, 2025-04-11:
> -n 1 是只解码一帧啊，应该不会说有问题的，是不是系统内存已经很紧张了，解码器只是最后一根稻草？ 还是说这个码流有什么特别？传上来看下

生成的命令我在下面回复了，感谢您的时间。这个h264文件我用ffmpeg试着解码过，可以正常使用

oneflyingfish, 2025-04-11:
> > -n 1 是只解码一帧啊，应该不会说有问题的，是不是系统内存已经很紧张了，解码器只是最后一根稻草？ 还是说这个码流有什么特别？传上来看下
> 
> 生成的命令我在下面回复了，感谢您的时间。这个h264文件我用ffmpeg试着解码过，可以正常使用

不知道是不是我的使用方式不对，交叉编译完成后，是直接拷贝mpi_dec_test到Android板卡即可运行吗，是否需要额外的操作，比如额外拷贝libmpp.so，或者安装一些配置

laborer996, 2025-04-11:
> -n 1 是只解码一帧啊，应该不会说有问题的，是不是系统内存已经很紧张了，解码器只是最后一根稻草？ 还是说这个码流有什么特别？传上来看下

码流是通过ubuntu的ffmpeg生成的
```bash
ffmpeg -f lavfi -i testsrc=size=1920x1080:rate=20 -c:v libx264 -preset veryfast -crf 23 -pix_fmt yuv420p -frames:v 20 -f h264 output.h264
```

系统通过free -m查看还有7.GB以上内存可用，而且板卡没有跑什么程序，我只是在测试mpp可用性。另外，这个h264文件我用ffmpeg试着解码过，可以正常使用。

不知道是不是我的使用方式不对，交叉编译完成后，是直接拷贝mpi_dec_test到Android板卡即可运行吗，是否需要额外的操作，比如额外拷贝libmpp.so，或者安装一些配置

wenyue7, 2025-04-11:
是在哪个平台上测试？生成的码流也可以发一下

laborer996, 2025-04-11:
> 是在哪个平台上测试？生成的码流也可以发一下

板卡是RK3576，测试板卡系统为安卓14，编译平台为Ubuntu x86_64交叉编译，NDK版本为r25c。工具链编译正常的C++代码是可以运行的，另外osal目录下的mpp*文件都能运行

目前板卡不在手边，前面描述的视频我暂时没有，晚上补充上传一下。

但是我测试另一个h.264遇到了同样的问题，它的下载链接为https://github.com/CYEdgeai/mpp/releases/download/v0.0.1/test.h264

laborer996, 2025-04-11:
> > 是在哪个平台上测试？生成的码流也可以发一下
> 
> 板卡是RK3576，测试板卡系统为安卓14，编译平台为Ubuntu x86_64交叉编译，NDK版本为r25c。工具链编译正常的C++代码是可以运行的，另外osal目录下的mpp*文件都能运行
> 
> 目前板卡不在手边，前面描述的视频我暂时没有，晚上补充上传一下。
> 
> 但是我测试另一个h.264遇到了同样的问题，它的下载链接为https://github.com/CYEdgeai/mpp/releases/download/v0.0.1/test.h264

这个test.h264是我在网上下载的，起初我以为是网络资源有问题，所以我重新通过ffmpeg生成了一个，依旧遇到这个错误

wenyue7, 2025-04-11:
您那边编译kernel的config文件发一下

laborer996, 2025-04-11:
> 您那边编译kernel的config文件发一下

您是说操作系统内核吗，稍等，我得问一下提供商

laborer996, 2025-04-11:
> 是的，怀疑可能显示相关的buffer配置有问题，可以检查下drm相关内容是否有开启

kernel config如下:
```bash
# CONFIG_GKI_HIDDEN_DRM_CONFIGS is not set
CONFIG_ROCKCHIP_SERDES_DRM_PANEL=y
CONFIG_DRM=y
CONFIG_DRM_EDID=y
CONFIG_DRM_IGNORE_IOTCL_PERMIT=y
CONFIG_DRM_MIPI_DSI=y
# CONFIG_DRM_DEBUG_MM is not set
CONFIG_DRM_KMS_HELPER=y
# CONFIG_DRM_DEBUG_DP_MST_TOPOLOGY_REFS is not set
# CONFIG_DRM_DEBUG_MODESET_LOCK is not set
CONFIG_DRM_LOAD_EDID_FIRMWARE=y
CONFIG_DRM_DP=y
CONFIG_DRM_DISPLAY_HELPER=y
CONFIG_DRM_DISPLAY_DP_HELPER=y
CONFIG_DRM_DISPLAY_HDCP_HELPER=y
CONFIG_DRM_DISPLAY_HDMI_HELPER=y
CONFIG_DRM_DP_AUX_CHARDEV=y
# CONFIG_DRM_DP_CEC is not set
CONFIG_DRM_GEM_DMA_HELPER=y
# CONFIG_DRM_I2C_CH7006 is not set
# CONFIG_DRM_I2C_SIL164 is not set
# CONFIG_DRM_I2C_NXP_TDA998X is not set
# CONFIG_DRM_I2C_NXP_TDA9950 is not set
# CONFIG_DRM_HDLCD is not set
# CONFIG_DRM_MALI_DISPLAY is not set
# CONFIG_DRM_KOMEDA is not set
# CONFIG_DRM_RADEON is not set
# CONFIG_DRM_AMDGPU is not set
# CONFIG_DRM_NOUVEAU is not set
# CONFIG_DRM_VGEM is not set
# CONFIG_DRM_VKMS is not set
CONFIG_DRM_ROCKCHIP=y
# CONFIG_ROCKCHIP_DRM_DEBUG is not set
# CONFIG_ROCKCHIP_DRM_DIRECT_SHOW is not set
CONFIG_ROCKCHIP_DRM_TVE=y
# CONFIG_DRM_ROCKCHIP_VVOP is not set
CONFIG_DRM_ROCKCHIP_RK618=y
# CONFIG_DRM_VMWGFX is not set
# CONFIG_DRM_UDL is not set
# CONFIG_DRM_AST is not set
# CONFIG_DRM_MGAG200 is not set
# CONFIG_DRM_RCAR_DW_HDMI is not set
# CONFIG_DRM_RCAR_USE_LVDS is not set
# CONFIG_DRM_RCAR_USE_MIPI_DSI is not set
# CONFIG_DRM_QXL is not set
# CONFIG_DRM_VIRTIO_GPU is not set
CONFIG_DRM_PANEL=y
# CONFIG_DRM_PANEL_ARM_VERSATILE is not set
# CONFIG_DRM_PANEL_ASUS_Z00T_TM5P5_NT35596 is not set
# CONFIG_DRM_PANEL_BOE_BF060Y8M_AJ0 is not set
# CONFIG_DRM_PANEL_BOE_HIMAX8279D is not set
# CONFIG_DRM_PANEL_BOE_TV101WUM_NL6 is not set
# CONFIG_DRM_PANEL_DSI_CM is not set
# CONFIG_DRM_PANEL_LVDS is not set
CONFIG_DRM_PANEL_SIMPLE=y
# CONFIG_DRM_PANEL_EDP is not set
# CONFIG_DRM_PANEL_EBBG_FT8719 is not set
# CONFIG_DRM_PANEL_ELIDA_KD35T133 is not set
# CONFIG_DRM_PANEL_FEIXIN_K101_IM2BA02 is not set
# CONFIG_DRM_PANEL_FEIYANG_FY07024DI26A30D is not set
# CONFIG_DRM_PANEL_ILITEK_ILI9881C is not set
# CONFIG_DRM_PANEL_INNOLUX_P079ZCA is not set
# CONFIG_DRM_PANEL_JDI_LT070ME05000 is not set
# CONFIG_DRM_PANEL_JDI_R63452 is not set
# CONFIG_DRM_PANEL_KHADAS_TS050 is not set
# CONFIG_DRM_PANEL_KINGDISPLAY_KD097D04 is not set
[... 84 lines cut ...]
CONFIG_DRM_DW_HDMI_I2S_AUDIO=y
# CONFIG_DRM_DW_HDMI_GP_AUDIO is not set
CONFIG_DRM_DW_HDMI_CEC=y
CONFIG_DRM_DW_MIPI_DSI=y
# CONFIG_DRM_ETNAVIV is not set
# CONFIG_DRM_HISI_HIBMC is not set
# CONFIG_DRM_HISI_KIRIN is not set
# CONFIG_DRM_LOGICVC is not set
# CONFIG_DRM_ARCPGU is not set
# CONFIG_DRM_BOCHS is not set
# CONFIG_DRM_CIRRUS_QEMU is not set
# CONFIG_DRM_GM12U320 is not set
# CONFIG_DRM_SIMPLEDRM is not set
# CONFIG_DRM_PL111 is not set
# CONFIG_DRM_LIMA is not set
# CONFIG_DRM_PANFROST is not set
# CONFIG_DRM_TIDSS is not set
# CONFIG_DRM_GUD is not set
# CONFIG_DRM_SSD130X is not set
# CONFIG_DRM_LEGACY is not set
CONFIG_DRM_PANEL_ORIENTATION_QUIRKS=y
CONFIG_DRM_NOMODESET=y
CONFIG_ROCKCHIP_RKNPU_DRM_GEM=y

```

wenyue7, 2025-04-14:
设备上电之后，直接执行mpi_dec_test 就会出现，还是运行一段时间别的?

可以执行一下命令，将log上传一下
cat /proc/meminfo
cat d/dma_buf/bufinfo
dumpsys meminfo | grep DMA