rockchip-linux/mpp #835: mpp h264 编码，在静态画面下会卡住，出现矩阵问题

cppcoffee, 2025-04-15:
mpp h264 编码，在静态画面下（没有开启任何 app 的 winows 桌面）

1. 移动鼠标不会显示鼠标移动位置
2. 静止划片的时间越长，矩阵就越多
3. 如果打开全屏的一个 app，画面会刷新，矩阵也会消失。如果是静止的画面（没有播放视频），会再次出现

[image]

放的时间长一些会出现更多的矩阵

[image]

用的是 Orange Pi 5 Plus，它的 CPU 是 RK3588, 以下是 mpp h264 编码的配置:

```c
int
setup_mpp_enc_cfg(context_t *p)
{
    int rc;
    int width = 1920;
    int height = 1080;
    MppEncCfg cfg = p->enc_cfg;
    /* setup default parameter */
    RK_S32 fps_in_flex = 0;
    RK_S32 fps_out_flex = 0;
    RK_S32 fps_in_den = 1;
    RK_S32 fps_in_num = 30;
    RK_S32 fps_out_den = 1;
    RK_S32 fps_out_num = 30;
    RK_S32 bps = width * height / 8 * (fps_out_num / fps_out_den);

    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_i", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_p", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_i", aq_step_i_ipc);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_p", aq_step_p_ipc);

    mpp_enc_cfg_set_s32(cfg, "rc:max_reenc_times", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:anti_flicker_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_p", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atl_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_p", 0);

    mpp_enc_cfg_set_s32(cfg, "tune:scene_mode", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_en", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:rc_container", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:vmaf_opt", 0);
    mpp_enc_cfg_set_s32(cfg, "hw:qbias_en", 1);
    mpp_enc_cfg_set_s32(cfg, "hw:qbias_i", 0);
    mpp_enc_cfg_set_s32(cfg, "hw:qbias_p", 0);
    mpp_enc_cfg_set_s32(cfg, "hw:skip_bias_en", 0);
    mpp_enc_cfg_set_s32(cfg, "hw:skip_bias", 4);
    mpp_enc_cfg_set_s32(cfg, "hw:skip_sad", 8);

    mpp_enc_cfg_set_s32(cfg, "prep:width", width);
    mpp_enc_cfg_set_s32(cfg, "prep:height", height);
    mpp_enc_cfg_set_s32(cfg, "prep:hor_stride", width);
    mpp_enc_cfg_set_s32(cfg, "prep:ver_stride", height);
    mpp_enc_cfg_set_s32(cfg, "prep:format", MPP_FMT_YUV444SP);
    mpp_enc_cfg_set_s32(cfg, "prep:range", MPP_FRAME_RANGE_JPEG);

    mpp_enc_cfg_set_s32(cfg, "rc:mode", MPP_ENC_RC_MODE_VBR);
    mpp_enc_cfg_set_u32(cfg, "rc:max_reenc_times", 0);
    mpp_enc_cfg_set_u32(cfg, "rc:super_mode", 0);

    /* fix input / output frame rate */
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_flex", fps_in_flex);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_num", fps_in_num);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_in_denom", fps_in_den);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_flex", fps_out_flex);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_num", fps_out_num);
    mpp_enc_cfg_set_s32(cfg, "rc:fps_out_denom", fps_out_den);

    /* drop frame or not when bitrate overflow */
    mpp_enc_cfg_set_u32(cfg, "rc:drop_mode", MPP_ENC_RC_DROP_FRM_DISABLED);
    mpp_enc_cfg_set_u32(cfg, "rc:drop_thd", 20);        /* 20% of max bps */
    mpp_enc_cfg_set_u32(cfg, "rc:drop_gap", 1);         /* Do not continuous drop frame */

    /* setup bitrate for different rc_mode */
    mpp_enc_cfg_set_s32(cfg, "rc:bps_target", bps);

    /* VBR mode has wide bound */
    mpp_enc_cfg_set_s32(cfg, "rc:bps_max", bps * 17 / 16);
    mpp_enc_cfg_set_s32(cfg, "rc:bps_min", bps * 1 / 16);

    /* setup qp for different codec and rc_mode */
    mpp_enc_cfg_set_s32(cfg, "rc:qp_init", -1);
    mpp_enc_cfg_set_s32(cfg, "rc:qp_max", 51);
    mpp_enc_cfg_set_s32(cfg, "rc:qp_min", 10);
    mpp_enc_cfg_set_s32(cfg, "rc:qp_max_i", 51);
    mpp_enc_cfg_set_s32(cfg, "rc:qp_min_i", 10);
    mpp_enc_cfg_set_s32(cfg, "rc:qp_ip", 2);
    mpp_enc_cfg_set_s32(cfg, "rc:fqp_min_i", 10);
    mpp_enc_cfg_set_s32(cfg, "rc:fqp_max_i", 45);
    mpp_enc_cfg_set_s32(cfg, "rc:fqp_min_p", 10);
    mpp_enc_cfg_set_s32(cfg, "rc:fqp_max_p", 45);

    /* setup codec  */
    mpp_enc_cfg_set_s32(cfg, "codec:type", MPP_VIDEO_CodingAVC);

    /*
     * H.264 profile_idc parameter
     * 66  - Baseline profile
     * 77  - Main profile
     * 100 - High profile
     */
    mpp_enc_cfg_set_s32(cfg, "h264:profile", 100);
    /*
     * H.264 level_idc parameter
     * 10 / 11 / 12 / 13    - qcif@15fps / cif@7.5fps / cif@15fps / cif@30fps
     * 20 / 21 / 22         - cif@30fps / half-D1@@25fps / D1@12.5fps
     * 30 / 31 / 32         - D1@25fps / 720p@30fps / 720p@60fps
     * 40 / 41 / 42         - 1080p@30fps / 1080p@30fps / 1080p@60fps
     * 50 / 51 / 52         - 4K@30fps
     */
    mpp_enc_cfg_set_s32(cfg, "h264:level", 40);
    mpp_enc_cfg_set_s32(cfg, "h264:cabac_en", 1);
    mpp_enc_cfg_set_s32(cfg, "h264:cabac_idc", 0);
    mpp_enc_cfg_set_s32(cfg, "h264:trans8x8", 1);

    mpp_enc_cfg_set_s32(cfg, "prep:mirroring", 0);
    mpp_enc_cfg_set_s32(cfg, "prep:rotation", 0);
    mpp_enc_cfg_set_s32(cfg, "prep:flip", 0);

    // config gop_len and ref cfg
    mpp_enc_cfg_set_s32(cfg, "rc:gop", fps_out_num * 2);

    rc = p->mpi->control(p->ctx, MPP_ENC_SET_CFG, cfg);
    if (rc) {
        LOG_ERROR("failed to mpi control enc set cfg, rc=%d", rc);
        return -1;
    }

    RcApiBrief rc_api_brief;
    rc_api_brief.type = MPP_VIDEO_CodingAVC;
    rc_api_brief.name = "default";

    rc = p->mpi->control(p->ctx, MPP_ENC_SET_RC_API_CURRENT, &rc_api_brief);
    if (rc) {
        LOG_ERROR("failed to mpi control enc set rc api, rc=%d", rc);
        return -1;
    }

    /* optional */
    {
        RK_U32 sei_mode = MPP_ENC_SEI_MODE_DISABLE;
        rc = p->mpi->control(p->ctx, MPP_ENC_SET_SEI_CFG, &sei_mode);
        if (rc) {
            LOG_ERROR("failed to mpi control enc set sei cfg, rc=%d", rc);
            return -1;
        }
    }

    MppEncHeaderMode header_mode = MPP_ENC_HEADER_MODE_EACH_IDR;
    rc = p->mpi->control(p->ctx, MPP_ENC_SET_HEADER_MODE, &header_mode);
    if (rc) {
        LOG_ERROR("failed to mpi control enc set header mode, rc=%d", rc);
        return -1;
    }

    return 0;
}
```

HermanChen, 2025-04-16:
传输时是否有丢包？

cppcoffee, 2025-04-16:
> 传输时是否有丢包？

3q, 检查下来确实丢包导致的

closed 2025-04-16 by cppcoffee