rockchip-linux/mpp #836: RK3562 MJPEG编码Y400时报错

ippfcox, 2025-04-16:
使用mpp库版本为本仓库最新版本编译得到
现象：
创建mjpeg编码器，编码YUV的格式也为NV12，编码器format为MPP_FMT_YUV420SP时编码正常，输出mjpeg也正常
仅修改编码器的format为MPP_FMT_YUV400（基于像素格式进行的frame size之类的调整是自动的，除此以外，其它参数都不动），出现报错：
[24688.966198] rkisp-vir0: MIPI drop frame
[24689.066063] mpp_rkvenc2 ff360000.rkvenc: found error status 00000100
[24689.066247] mpp_rkvenc2 ff360000.rkvenc: resetting...
[24689.066291] rkvenc_reset:1967: soft reset timeout, use cru reset
[24689.066362] mpp_rkvenc2 ff360000.rkvenc: reset done

目的是进行灰度图的编码，是否支持？或者还有哪些参数需要配置？

ippfcox, 2025-04-16:
补充一些，打开了mpp的调试输出：
（由于开了三路编码，一路NV12编码，两路Y400编码，所以打印可能有重复的）
mpp[17377]: mpp_info: mpp version: ff3ae5c author: Xiaobo Liu     2025-04-11 fix[mpi]: Fix typo
mpp[17377]: mpp_enc: MPP_ENC_SET_RC_CFG bps 47185920 [2949120 : 50135040] fps [30:30] gop 60
mpp[17377]: jpege_api_v2: setup quant 7 change mode vbr fixqp
mpp[17377]: mpp_info: mpp version: ff3ae5c author: Xiaobo Liu     2025-04-11 fix[mpi]: Fix typo
mpp[17377]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[17377]: jpege_api_v2: setup quant 7 change mode vbr fixqp
mpp[17377]: mpp_info: mpp version: ff3ae5c author: Xiaobo Liu     2025-04-11 fix[mpi]: Fix typo
mpp[17377]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7200000 [450000 : 7650000] fps [30:30] gop 60
mpp[17377]: jpege_api_v2: setup quant 7 change mode vbr fixqp
mpp[17377]: rc_model_v2: invalid fix 0 qp found set default qp 26
mpp[17377]: vepu540c_common: unknown color space: 10
mpp[17377]: rc_model_v2: invalid fix 0 qp found set default qp 26
mpp[17377]: rc_model_v2: invalid fix 0 qp found set default qp 26
mpp[17377]: vepu540c_common: unknown color space: 10
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check hw_status: 0x00000100
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check RKV_ENC_INT_TIMEOUT_ERROR
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check hw_status: 0x00000100
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check RKV_ENC_INT_TIMEOUT_ERROR
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check hw_status: 0x00000001
mpp[17377]: hal_jpege_v540c: hal_jpege_vepu540c_status_check RKV_ENC_INT_ONE_FRAME_FINISH
mpp[17377]: vepu540c_common: unknown color space: 10

HermanChen, 2025-04-16:
3562 的 jpeg 编码器不支持 400，我们看下怎么在开始就报错退出下……

ippfcox, 2025-04-17:
> 3562 的 jpeg 编码器不支持 400，我们看下怎么在开始就报错退出下……

好的，感谢，它是硬件上就不支持是吧？另外3562的MJPEG编码能力您这边方便大概说一下吗？

closed 2025-05-15 by ippfcox