rockchip-linux/mpp #837: 解码代码有内存泄漏，抄的examples

wzw88486969, 2025-04-20:
`
int decode_simple(MpiDecLoopData *data, AVPacket *av_packet )
{
    RK_U32 pkt_done = 0;
    RK_U32 pkt_eos  = 0;
    RK_U32 err_info = 0;
    MPP_RET ret = MPP_OK;
    MppCtx ctx  = data->ctx;
    MppApi *mpi = data->mpi;
    MppPacket packet = data->packet;

    mpp_packet_set_data(packet, av_packet->data);
    mpp_packet_set_size(packet, av_packet->size);
    mpp_packet_set_pos(packet, av_packet->data);
    mpp_packet_set_length(packet, av_packet->size);

    do {
        RK_S32 times = 30;
        // send the packet first if packet is not done
        if (!pkt_done) {
            ret = mpi->decode_put_packet(ctx, packet);
            if (MPP_OK == ret)
                pkt_done = 1;
        }

        // then get all available frame and release
        do {
            RK_S32 get_frm = 0;
            RK_U32 frm_eos = 0;
            MppFrame  frame  = NULL;

            try_again:
            ret = mpi->decode_get_frame(ctx, &frame);

            if (MPP_ERR_TIMEOUT == ret) {
                if (times > 0) {
                    times--;
                    msleep(1);
                    printf("decode_get_frame MPP_ERR_TIMEOUT try_again\n");

                    goto try_again;
                }
                printf("decode_get_frame failed too much time\n");
            }
            if (MPP_OK != ret) {
                printf("decode_get_frame failed ret %d\n", ret);
                break;
            }

            if (frame) 
            {
                frm_eos = mpp_frame_get_eos(frame);
                mpp_frame_deinit(&frame);

                frame = NULL;
                get_frm = 1;
            }

            
            if (get_frm)
                continue;

            break;
        } while (1);

        if (data->frame_num > 0 && data->frame_count >= data->frame_num) {
            data->eos = 1;
            printf("reach max frame number %d\n", data->frame_count);
            break;
        }

        if (pkt_done)
            break;

        msleep(1);
    } while (1);

    return ret;
}`

通过注代码，发现这段代码有泄漏，每次264KB，请各位指点，谢谢

wzw88486969, 2025-04-25:
请关闭这个帖子吧，是其他地方泄漏

xiao-sheng-av, 2025-08-09:
> 请关闭这个帖子吧，是其他地方泄漏

你好，我也是使用ffmpeg读取usb摄像头的数据然后使用mpp解码，每隔一段时间都会泄漏264kb，请问是哪里的问题呢

closed 2025-08-09 by HermanChen

xiao-sheng-av, 2025-08-09:
> > 请关闭这个帖子吧，是其他地方泄漏
> 
> 你好，我也是使用ffmpeg读取usb摄像头的数据然后使用mpp解码，每隔一段时间都会泄漏264kb，请问是哪里的问题呢

已经解决了