rockchip-linux/mpp #838: YUYV 的格式，画面不正常

cppcoffee, 2025-04-22:
使用 h264 编码和 yuyv 的 HDMI IN 格式，输出的画面不正常：

格式配置相关的 C 的代码：

```c
    struct v4l2_format fmt;

    ...

    p->dev_fd = open(opt->video_dev, O_RDWR | O_CLOEXEC);
    if (p->dev_fd == -1) {
        LOG_ERROR("failed to open video device %s: %s", opt->video_dev, xstrerror());
        goto failed;
    }

    LOG_INFO("opened video device %s", opt->video_dev);

    dump_video_supported_formats(p);

    memset(&fmt, 0, sizeof(struct v4l2_format));

    fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE_MPLANE;
    fmt.fmt.pix_mp.width = opt->width;
    fmt.fmt.pix_mp.height = opt->height;
    fmt.fmt.pix_mp.pixelformat = V4L2_PIX_FMT_YUYV;  // 使用 V4L2 YUYV
    fmt.fmt.pix_mp.field = V4L2_FIELD_ANY;
    fmt.fmt.pix_mp.num_planes = 1;

    if (ioctl(p->dev_fd, VIDIOC_S_FMT, &fmt) < 0) {
        LOG_ERROR("failed to v4l2 set format: %s", xstrerror());
        goto failed;
    }

...

        mpp_frame_set_width(frame, opt->width);
        mpp_frame_set_height(frame, opt->height);
        mpp_frame_set_hor_stride(frame, opt->hor_stride);
        mpp_frame_set_ver_stride(frame, opt->ver_stride);
        mpp_frame_set_fmt(frame, MPP_FMT_YUV422_YUYV);      // 使用 MPP YUYV
        mpp_frame_set_eos(frame, 0);

...
```

使用 YUYV 格式后，画面不正常是 MPP BUG 吗？

cppcoffee, 2025-04-22:
每一个都尝试了一下，只有 `MPP_FMT_YUV422_UYVY` 格式比较接近:

[image]

其他的 MPP_FMT_XXX 色彩对 YUYV 格式都差异很大

HermanChen, 2025-04-22:
看起来是宽高 stride 的配置有问题，stride 是配置成宽度的 2 倍么？

HermanChen, 2025-04-24:
什么芯片平台？

xiaohuihuige, 2025-06-27:
> > > > 什么芯片平台？
> > > 
> > > 
> > > RK3566 芯片
> > 
> > 
> > 是你的步长有问题
> 
> 步长指的是什么参数呢？

hor_stride
ver_stride

cppcoffee, 2025-06-27:
步长计算公式是参考：

https://github.com/rockchip-linux/mpp/blob/develop/test/mpi_enc_test.c#L274-L277

步长计算公式使用:

```c
#define MPP_ALIGN(x, a)         (((x)+(a)-1)&~((a)-1))

    p->hor_stride   = MPP_ALIGN(cmd->width, 16);
    p->ver_stride   = MPP_ALIGN(cmd->height, 16);
```

计算结果:

```
 width: 1920
 height: 1080
 hor_stride: 3840
 ver_stride: 1080
```

是不是

closed 2025-06-27 by cppcoffee