rockchip-linux/mpp #842: 用MPP_VIDEO_CodingMJPEG的方式，将NV12的数据编码成bgr888保存成jpg保存出错。

HDL199411, 2025-04-27:
使用的是1.0.9的库。在rk3588的安卓平台运行。
代码如下：
`Encoder::Encoder() {
    LOGI("[Encoder] init 1");
    fp = fopen("output0427.jpg", "wb");
    // 初始化编码器参数
    // 初始化MPP
    ret = mpp_create(&ctx_, &mpi_);
    if (ret) {
        LOGI("mpp_create failed");
        return;
    }
    // 配置编码器参数
    MppCodingType codec_type = MPP_VIDEO_CodingMJPEG;
    ret = mpp_init(ctx_, MPP_CTX_ENC, codec_type);
    if (ret) {
        LOGI("mpp_init failed");
        return;
    }
    LOGI("[Encoder] init 3");
    // 计算对齐的宽度和高度
    width_ = 1280;
    height_ = 720;
    hor_stride_ = MPP_ALIGN(width_, MPP_ALIGN_HOR);
    ver_stride_ = MPP_ALIGN(height_, MPP_ALIGN_VER);

    ret = mpp_enc_cfg_init(&cfg);
    if (ret) {
        LOGI("Failed to init encoder config");
        return;
    }
    mpp_enc_cfg_set_s32(cfg, "prep:width", width_);
    mpp_enc_cfg_set_s32(cfg, "prep:height", height_);
    mpp_enc_cfg_set_s32(cfg, "prep:hor_stride", hor_stride_);
    mpp_enc_cfg_set_s32(cfg, "prep:ver_stride", ver_stride_);
    mpp_enc_cfg_set_s32(cfg, "jpeg:q_factor", 80);
    mpp_enc_cfg_set_s32(cfg, "jpeg:qf_max", 99);
    mpp_enc_cfg_set_s32(cfg, "jpeg:qf_min", 1);
    LOGI("[Encoder] init 4");
    ret = mpi_->control(ctx_, MPP_ENC_SET_CFG, cfg); //MPP_ENC_SET_REF_CFG
    if (ret) {
        LOGI("Failed to set encoder config");
        return;
    }

}

Encoder::~Encoder() {
    // 释放资源
    if (ctx_) {
        mpp_destroy(ctx_);
        ctx_ = nullptr;
        mpi_ = nullptr;
    }
}

int Encoder::encode(uint8_t* input_data, size_t size) {
    // 准备MPP输入buffer
    MppFrame frame = nullptr;
    mpp_frame_init(&frame);
    mpp_frame_set_width(frame, width_);
    mpp_frame_set_height(frame, height_);
    mpp_frame_set_hor_stride(frame, hor_stride_);
    mpp_frame_set_ver_stride(frame, ver_stride_);
    mpp_frame_set_fmt(frame, MPP_FMT_BGR888);
    // 设置输入数据
    MppBuffer input_buf;
    // size_t input_size = hor_stride_ * ver_stride_ * 3;  // BGR格式，每像素3字节
    size_t input_size = size;
    ret = mpp_buffer_get(nullptr, &input_buf, input_size);
    if (ret) {
        LOGE("Failed to get input buffer");
        mpp_frame_deinit(&frame);
        return -1;
    }
    // 复制图像数据到MPP buffer，需要考虑stride
    void* buf_ptr = mpp_buffer_get_ptr(input_buf);
    uint8_t* src_ptr = input_data;
    uint8_t* dst_ptr = (uint8_t*)buf_ptr;
   
    memcpy(dst_ptr, src_ptr, input_size);

    mpp_frame_set_buffer(frame, input_buf);
    // 编码
    MppPacket packet = nullptr;
    ret = mpi_->encode_put_frame(ctx_, frame);
    if (ret) {
        LOGE("Failed to put frame ret:%d", ret);
        mpp_buffer_put(input_buf);
        mpp_frame_deinit(&frame);
        return -1;
    } else {
        LOGI("Put frame success");
    }
    ret = mpi_->encode_get_packet(ctx_, &packet);
    if (ret || !packet) {
        LOGE("Failed to get packet");
        mpp_buffer_put(input_buf);
        mpp_frame_deinit(&frame);
        return -1;
    }
    // 获取 压缩后的数据
    void* compressed_data = mpp_packet_get_pos(packet);
    size_t compressed_size = mpp_packet_get_length(packet);
    // 创建压缩图像
    fwrite(compressed_data, 1, compressed_size, fp);
    fclose(fp);
    // 清理资源
    mpp_packet_deinit(&packet);
    mpp_buffer_put(input_buf);
    mpp_frame_deinit(&frame);
    LOGI("[Encoder] encode 10");
}`

HermanChen, 2025-04-27:
3588 的 jpeg 编码器只支持 RGBX8888 不支持 RGB888

HDL199411, 2025-04-28:
> 3588 的 jpeg 编码器只支持 RGBX8888 不支持 RGB888

我的输入格式是NV12，目标格式是BGR888，不是rgb888

HDL199411, 2025-04-28:
> 3588 的 jpeg 解码输出不再支持格式转换了，需要用 RGA 转一下

我修改了下代码，可以实现从nv12的YUV数据编码成mjpeg的图片了，但是代码中  mpp_enc_cfg_set_s32(cfg, "prep:format", MPP_FMT_YUV420SP);和mpp_frame_set_fmt(frame, MPP_FMT_RGBA8888); 这两个设置不设置好像都没什么效果。麻烦解答一下
```
int main() {
    struct timeval time_pose_start1, time_pose_stop1;

    MPP_RET ret = MPP_OK;
    MppCtx ctx = NULL;
    MppApi *mpi = NULL;
    MppEncCfg cfg = NULL;
    
    // 1. 初始化MPP上下文
    ret = mpp_create(&ctx, &mpi);
    if (ret != MPP_OK) {
        printf("mpp_create failed\n");
        return -1;
    }

    // 2. 设置编码类型为MJPEG
    ret = mpp_init(ctx, MPP_CTX_ENC, MPP_VIDEO_CodingMJPEG);
    if (ret != MPP_OK) {
        printf("mpp_init failed\n");
        return -1;
    }

    // 3. 配置编码参数
    ret = mpp_enc_cfg_init(&cfg);
    mpi->control(ctx, MPP_ENC_GET_CFG, cfg);

    int hor_stride = MPP_ALIGN(WIDTH, ALIGNMENT);
    int ver_stride = MPP_ALIGN(HEIGHT, ALIGNMENT);
    printf("Hor stride: %d, Ver stride: %d\n", 
        hor_stride, ver_stride);
    int hor_stride1 = MPP_ALIGN(WIDTH, 64);
    int ver_stride1 = MPP_ALIGN(HEIGHT, 64);
    printf("Hor stride: %d, Ver stride: %d\n", 
        hor_stride1, ver_stride1);
    // 基础参数配置
    mpp_enc_cfg_set_s32(cfg, "prep:width", WIDTH);
    mpp_enc_cfg_set_s32(cfg, "prep:height", HEIGHT);
    mpp_enc_cfg_set_s32(cfg, "prep:hor_stride", 1920);
    mpp_enc_cfg_set_s32(cfg, "prep:ver_stride", 1080);
    mpp_enc_cfg_set_s32(cfg, "prep:format", MPP_FMT_YUV420SP);
    
    // MJPEG专用参数
    mpp_enc_cfg_set_s32(cfg, "jpeg:q_factor", 80);
    mpp_enc_cfg_set_s32(cfg, "jpeg:qf_max", 99);
    mpp_enc_cfg_set_s32(cfg, "jpeg:qf_min", 1);
    mpp_enc_cfg_set_s32(cfg, "rc:mode", MPP_ENC_RC_MODE_FIXQP);
    
    // 应用配置
    ret = mpi->control(ctx, MPP_ENC_SET_CFG, cfg);
    if (ret != MPP_OK) {
        printf("set cfg failed\n");
        return -1;
    }

    // 4. 准备输入输出
    FILE *fp_input = fopen("test.yuv", "rb");
    FILE *fp_output = fopen("output0427_3.jpg", "wb");
    
    if (!fp_input || !fp_output) {
        printf("open file failed\n");
        return -1;
    }

    gettimeofday(&time_pose_start1, NULL);
    // 计算单帧大小
    size_t frame_size = WIDTH * HEIGHT * 3 / 2; // YUV420SP大小
    size_t align_size =  hor_stride * ver_stride * 3 / 2;

    // 创建输入缓冲区
    MppBuffer frm_buf = NULL;
    ret = mpp_buffer_get(NULL, &frm_buf, align_size);
    if (ret != MPP_OK) {
        printf("get frame buffer failed\n");
        return -1;
    }

    // 5. 读取单帧YUV数据
    void *buf_ptr = mpp_buffer_get_ptr(frm_buf);
    size_t read_size = fread(buf_ptr, 1, frame_size, fp_input);
    
    if (read_size != frame_size) {
        printf("read yuv data failed (need %zu, got %zu)\n", 
               frame_size, read_size);
               return -1;
    }

    // 6. 创建输入帧
    MppFrame frame = NULL;
    mpp_frame_init(&frame);
    
    mpp_frame_set_buffer(frame, frm_buf);
    mpp_frame_set_width(frame, WIDTH);
    mpp_frame_set_height(frame, HEIGHT);
    mpp_frame_set_hor_stride(frame, MPP_ALIGN(WIDTH, ALIGNMENT));
    mpp_frame_set_ver_stride(frame, MPP_ALIGN(HEIGHT, ALIGNMENT));
    mpp_frame_set_fmt(frame, MPP_FMT_RGBA8888); //NOTE: MPP_FMT_RGBA8888
    mpp_frame_set_eos(frame, 1); // 设置结束标志

    // 7. 编码单帧
    ret = mpi->encode_put_frame(ctx, frame);
    mpp_frame_deinit(&frame); // 立即释放帧
    
    if (ret != MPP_OK) {
        printf("put frame failed\n");
        return -1;
    }

    MppPacket packet = NULL;
    ret = mpi->encode_get_packet(ctx, &packet);
    
    if (ret == MPP_OK && packet) {
        // 获取编码数据
        void *data = mpp_packet_get_data(packet);
        size_t size = mpp_packet_get_length(packet);

        // // 添加JPEG文件头尾
        // const unsigned char jpeg_header[] = {0xFF, 0xD8}; // SOI
        // const unsigned char jpeg_footer[] = {0xFF, 0xD9}; // EOI
        
        // fwrite(jpeg_header, 1, sizeof(jpeg_header), fp_output);
        fwrite(data, 1, size, fp_output);
        // fwrite(jpeg_footer, 1, sizeof(jpeg_footer), fp_output);
        
        printf("JPEG encoded successfully! Size: %zu bytes\n", size);
        
        // 释放数据包资源
        MppBuffer pkt_buf = mpp_packet_get_buffer(packet);
        if (pkt_buf) mpp_buffer_put(pkt_buf);
        mpp_packet_deinit(&packet);
    } else {
        printf("get packet failed (ret=%d)\n", ret);
    }
    gettimeofday(&time_pose_stop1, NULL);
    double time_ms = getDiffMs(time_pose_start1, time_pose_stop1);
    printf("cost time: %.4f\n", time_ms);
    // 9. 释放资源
    if (ctx) mpp_destroy(ctx);
    if (cfg) mpp_enc_cfg_deinit(cfg);
    if (frm_buf) mpp_buffer_put(frm_buf);
    if (fp_input) fclose(fp_input);
    if (fp_output) fclose(fp_output);
    
    return ret == MPP_OK ? 0 : -1;
}
```

GOAWAY88, 2025-04-29:
@HermanChen

HermanChen, 2025-04-29:
1. prep:format 这个是指示输入数据格式的，但会以实际输入的 MppFrame 格式为主
2. 四核是并行的，单核能力只有 1080p@30fps，单核时间就长了
3. 编码器不带缩放功能，也需要用 RGA 来做缩放

GOAWAY88, 2025-04-30:
> 1. prep:format 这个是指示输入数据格式的，但会以实际输入的 MppFrame 格式为主
> 2. 四核是并行的，单核能力只有 1080p@30fps，单核时间就长了
> 3. 编码器不带缩放功能，也需要用 RGA 来做缩放

感谢回答。还有一些问题：
1. 3588 的 jpeg 编码器只支持 RGBX8888 不支持 RGB888，是指编码器输入数据格式不支持RGB888，还是指编码器输出数据格式不支持RGB888？
2. 以实际输入的 `MppFrame `格式为主，是指通过` mpp_frame_set_fmt(frame, xxxxx);`显示设置的`Mppframe`数据格式吗？还是编码时，内部自己做数据类型检查，不需要`mpp_frame_set_fmt`显示设置？
3. `MppFrame `可以设置外部内存空间吗？目前我是通过`mpp_buffer_get `申请`MppBuffer `后，将外部数据再拷贝到`MppBuffer `中，有点麻烦。我能否直接将外部内存的指针传给`MppFrame`，直接送去编码，省去一次数据拷贝的过程？
4. 我只编码一帧，保存为一张图片的话，单核时间也会比四核时间长吗？

HermanChen, 2025-04-30:
1. 是的，是指输入数据，输出格式都是 yuv420
2. 是的，指的 MppFrame 的 fmt 和 prep:format 是一致的
3. 可以的，编码器就是这样的用的
4. 会长，单核是四倍时间

GOAWAY88, 2025-05-06:
> 1. 是的，是指输入数据，输出格式都是 yuv420
> 2. 是的，指的 MppFrame 的 fmt 和 prep:format 是一致的
> 3. 可以的，编码器就是这样的用的
> 4. 会长，单核是四倍时间

第三点：编码器设置外部内存，怎么做呢？有例子吗？

HermanChen, 2025-05-06:
编码器的输入只支持挂在 MppFrame 下的 MppBuffer，对于一般 malloc 的 buffer 是不支持的

GOAWAY88, 2025-05-06:
> 编码器的输入只支持挂在 MppFrame 下的 MppBuffer，对于一般 malloc 的 buffer 是不支持的

没有明白挂在 MppFrame 下的 MppBuffer的意思。现在我的原始数据存在dma_buf_alloc申请下的dma内存中，我怎么将dma_buf_alloc申请到的地址src_buf_，导入到MppBuffer中呢？

GOAWAY88, 2025-05-07:
@HermanChen

HermanChen, 2025-05-07:
MppBuffer 有个 import 接口，用 ext_dma 的方式 import 进来

ippfcox, 2025-05-15:
导入外部内存的方式可以参考camera_source.c中关于mpp_buffer_import的使用，我是用来导入v4l2申请的buffer，工作正常

尝试导入过malloc出来的内存，失败，查了源码感觉是不支持（group为空了）

HermanChen, 2025-05-16:
malloc 的不支持

byg-git, 2025-05-23:
请问一下，rga转换格式,1080p和4K耗时大概多少。比如目前是30帧，再经过一次rga转换，会降到28帧以下吗 @HermanChen