rockchip-linux/mpp #843: Full range video with bt709 primaries are not handled correctly during encoding.

gnattu, 2025-05-06:
#### How to reproduce:

 - Download this sample file: https://github.com/user-attachments/assets/a73b2282-ea74-4532-9056-f091d0d1f9dd
- Use [ffmpeg-rockchip](https://github.com/nyanmisaka/ffmpeg-rockchip) to transcode it:

```
./ffmpeg -init_hw_device rkmpp=rk -hwaccel rkmpp -hwaccel_output_format drm_prime -noautorotate -afbc rga -i grayscale_full_range_8bit.mp4 -codec:v:0 hevc_rkmpp -tag:v:0 hvc1 test.mp4
```

Notice that the result video has much darker gradients (top original, bottom encoded):

[image]

qvoid, 2025-05-06:
@gnattu Sorry, I failed to download the video sample. Could you please share it over One Drive, or Google Drive. Or Baidu Drive is also OK. And also uploading the encoding relsult bitstream is appreciated.

gnattu, 2025-05-06:
You can generate one with ffmpeg using this image (please don't tell me you cannot see this picture):

[image]

```
ffmpeg \
  -loop 1 -framerate 30 -t 10 \
  -i grayscale.png \
  -pix_fmt yuv420p \
  -c:v libx265 \
  -color_primaries bt709 \
  -color_trc bt709 \
  -colorspace bt709 \
  -color_range pc \
  -movflags +faststart \
  -tag:v:0 hvc1 \
  grayscale_full_range_8bit.mp4
```

qvoid, 2025-05-06:
The above command will generate a H.265 **limit range** content encode as **full range BT.709 YUV**. Scene you are using `-pix_fmt yuv420p` to make ffmpeg to generate **limit range** YUV from a full range PNG file.
Please check with `-pix_fmt yuvj420p` command, which is for full range YUV.

gnattu, 2025-05-06:
No. In new version of ffmpeg the `yuvj420p ` range is already deprecated and the result video is in correct full range. And if you use ffprobe to probe it it will even tell you it isi in yuvj420p:

```
  Stream #0:0[0x1](und): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt709/unknown/unknown, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 31 kb/s, 30 fps, 30 tbr, 15360 tbn (default)
```

qvoid, 2025-05-06:
After compile ffmpeg 7.1.1 and try with it. It is confirm that the YUV encoded inside H.265 bistream with above command is full range.

Checked with MPP encoder driver and  [mpp caller at ffmpeg-rockchip](https://github.com/nyanmisaka/ffmpeg-rockchip/blob/master/libavcodec/rkmppenc.c). Looks like color space and range is well handled.
It looks like an encoder bitrate setting problem, not about color range and space. You can try to encode with fix qp mode and   setting qp to 10. 

I don't have environment for testing [ffmpeg-rockchip](https://github.com/nyanmisaka/ffmpeg-rockchip) currently. Maybe @nyanmisaka  can help to do some test and debugging. To see if it is RGA problem or MPP problem.

nyanmisaka, 2025-05-06:
@qvoid I can also reproduce the problem. The above ffmpeg-rockchip command only involves the MPP encoder, no RGA CSC conversion occurs.

The MPP encoder encodes directly from the AFBC data output by the MPP decoder, and ffmpeg correctly set the color range and other properties.

```c
    mpp_enc_cfg_set_s32(cfg, "prep:colorspace", avctx->colorspace);
    mpp_enc_cfg_set_s32(cfg, "prep:colorprim", avctx->color_primaries);
    mpp_enc_cfg_set_s32(cfg, "prep:colortrc", avctx->color_trc);
    mpp_enc_cfg_set_s32(cfg, "prep:colorrange", avctx->color_range);
...
    mpp_frame_set_colorspace(mpp_frame, avctx->colorspace);
    mpp_frame_set_color_primaries(mpp_frame, avctx->color_primaries);
    mpp_frame_set_color_trc(mpp_frame, avctx->color_trc);
    mpp_frame_set_color_range(mpp_frame, avctx->color_range);
```

gnattu, 2025-05-06:
> It looks like an encoder bitrate setting problem, not about color range and space. You can try to encode with fix qp mode and setting qp to 10.

No it is not. If you analyze the output frames closer you will notice that the input signals <16 are all 0 in the frames encoded with rkmpp, and the input signal =17 will be encoded as 1 and starts from here, indicating that the encoder is not correctly working handling full range mode inputs as it clamps the input in the limited range and then stretches it to full range in the output.

nyanmisaka, 2025-05-06:
I found that the video output by the MPP encoder were not correctly marked as pc/full range even though I had set it to do so. Remux the file with `-bsf:v hevc_metadata=video_full_range_flag=1 -c:v copy` does the trick.

```
  Stream #0:0[0x1](und): Video: hevc (Main) (hvc1 / 0x31637668), yuv420p(tv, progressive), 1920x1080, 36 kb/s, SAR 1:1 DAR 16:9, 30 fps, 30 tbr, 15360 tbn (default)
      Metadata:
        handler_name    : VideoHandler
        vendor_id       : [0][0][0][0]
        encoder         : Lavc61.19.101 hevc_rkmpp
```

nyanmisaka, 2025-05-06:
@gnattu Found the issue. ffmpeg-rockchip does not set the color range before get header sync, I thought another function did that. I think the issue can be closed.

https://github.com/user-attachments/assets/2c1e167d-7203-4e63-b9e0-fb56ecfade53