rockchip-linux/mpp #845: 3568 mpp编码支持NV21吗

nazjat-ar, 2025-05-08:
使用 mpi_enc_test, 指定输入格式为NV21 ，编出来的码流 红蓝反色。本地调试看 rbuv_swap的值已经是1 了，但是貌似没生效。
`./mpi_enc_test -i enc_in.bin -w 640 -h 480 -o mpip.h264 -f 5`

nazjat-ar, 2025-05-08:
执行日志如下：
mpi_enc_utils: cmd parse result:
mpi_enc_utils: input  file name: enc_in.bin
mpi_enc_utils: output file name: mpi.h264
mpi_enc_utils: width      : 640
mpi_enc_utils: height     : 480
mpi_enc_utils: format     : 5
mpi_enc_utils: type       : 7
mpi_enc_test: ./mpi_enc_test start
mpp_info: mpp version: aaa4c8e author: Chandler Chen 2024-10-29 fix[hal_vdpu383]: fix fbc hor_stride mismatch issue
mpi_enc_test: 0xf7664740 encoder test start w 640 h 480 type 7
mpp_serivce: check_mpp_service_cap vcodec_support 1005220a
mpp_serivce: check_mpp_service_cap query query_cmd   support 0003
mpp_serivce: check_mpp_service_cap query init_cmd    support 0103
mpp_serivce: check_mpp_service_cap query send_cmd    support 0205
mpp_serivce: check_mpp_service_cap query poll_cmd    support 0302
mpp_serivce: check_mpp_service_cap query control_cmd support 0404
mpp_serivce: check_mpp_service_cap client vdpu2      hw_id 00000000
mpp_serivce: check_mpp_service_cap client vdpu2_pp   hw_id 00000000
mpp_serivce: check_mpp_service_cap client rkvdec     hw_id 032a3f03
mpp_serivce: check_mpp_service_cap client rkjpegd    hw_id db1f0005
mpp_serivce: check_mpp_service_cap client rkvenc     hw_id 50600311
mpp_serivce: check_mpp_service_cap client vepu2      hw_id 00000000
mpp_serivce: check_mpp_service_cap client iep        hw_id 00000000
h264e_api_v2: h264e_init enter
h264e_api_v2: h264e_init leave
mpp_enc: MPP_ENC_SET_RC_CFG bps 1152000 [72000 : 1224000] fps [30:30] gop 60
h264e_api_v2: h264e_proc_cfg enter ctx 0xf765dae0 cmd 320001 param 0xf76297c0
h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [640:480] stride [640:480]
h264e_api_v2: h264e_proc_cfg leave ret 0
mpp_enc: mode vbr bps [72000:1152000:1224000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
rc: using rc impl default
mpp_enc: mode vbr bps [72000:1152000:1224000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
h264e_api_v2: h264e_gen_hdr enter
h264e_api_v2: h264e_gen_hdr leave
mpp: 0xf769a650 input frame pts 0
h264e_api_v2: h264e_start enter
h264e_api_v2: h264e_start leave
h264e_api_v2: h264e_proc_dpb enter
h264e_api_v2: h264e_proc_dpb leave
h264e_api_v2: h264e_proc_hal enter
h264e_api_v2: h264e_proc_hal leave
mpp: 0xf769a650 output packet pts 0
mpi_enc_test: chn 0 encoded frame 0    size 19632   qp 19 bps_rt 1328660
mpp: 0xf769a650 input frame pts 0
h264e_api_v2: h264e_start enter
h264e_api_v2: h264e_start leave
h264e_api_v2: h264e_proc_dpb enter
h264e_api_v2: h264e_proc_dpb leave
h264e_api_v2: h264e_proc_hal enter
h264e_api_v2: h264e_proc_hal leave
mpp: 0xf769a650 output packet pts 0
mpi_enc_test: chn 0 encoded frame 1    size 4784    qp 19 bps_rt 1261480
mpp: 0xf769a650 input frame pts 0
h264e_api_v2: h264e_start enter
h264e_api_v2: h264e_start leave
h264e_api_v2: h264e_proc_dpb enter
h264e_api_v2: h264e_proc_dpb leave
h264e_api_v2: h264e_proc_hal enter
h264e_api_v2: h264e_proc_hal leave
mpp: 0xf769a650 output packet pts 0
mpi_enc_test: chn 0 encoded frame 2    size 4217    qp 17 bps_rt 1260661
mpp: 0xf769a650 input frame pts 0
h264e_api_v2: h264e_start enter
h264e_api_v2: h264e_start leave
h264e_api_v2: h264e_proc_dpb enter
h264e_api_v2: h264e_proc_dpb leave
h264e_api_v2: h264e_proc_hal enter
h264e_api_v2: h264e_proc_hal leave
mpp: 0xf769a650 output packet pts 0
mpi_enc_test: chn 0 encoded frame 3    size 2993    qp 17 bps_rt 1258294
mpp: 0xf769a650 input frame pts 0
h264e_api_v2: h264e_start enter
h264e_api_v2: h264e_start leave
h264e_api_v2: h264e_proc_dpb enter
h264e_api_v2: h264e_proc_dpb leave
h264e_api_v2: h264e_proc_hal enter
h264e_api_v2: h264e_proc_hal leave
mpp: 0xf769a650 output packet pts 0
......

nazjat-ar, 2025-05-08:
[image]

Liaoyyyy, 2025-05-09:
你好，rk3568不支持nv12的uv_swap.

closed 2025-05-21 by nazjat-ar