rockchip-linux/mpp #850: rv1103 有支持计划吗？

Ncerzzk, 2025-05-18:
试了一下之前在rk3568上的代码，mpp初始化的时候报错：
```
mpp[904]: mpp_rt: can NOT found any allocator
mpp[904]: mpp_log: allocator_std_open Warning: std allocator should be used on simulation mode only
mpp[904]: mpp_log: allocator_std_alloc Warning: std allocator should be used on simulation mode only
mpp[904]: mpp_buffer: mpp_buffer_create failed to create buffer with size 1253376
failed to get buffer for output packet ret -1
mpp[904]: mpp_log: allocator_std_alloc Warning: std allocator should be used on simulation mode only
mpp[904]: mpp_buffer: mpp_buffer_create failed to create buffer with size 12512
```
是rv1103还不支持吗？

HermanChen, 2025-05-19:
rv1103 用的是特殊的内核 kmpp，通用的 mpp 不支持

Ncerzzk, 2025-05-19:
好的，感谢

closed 2025-05-19 by Ncerzzk