rockchip-linux/mpp #851: 视频断层

DianKeWei, 2025-05-19:
https://github.com/user-attachments/assets/683f820c-e5a3-4755-bb77-211e77c3b3a9 刚开始的osd还没调好，中途最下面出现断层闪烁是什么原因

ippfcox, 2025-05-20:
这不就是撕裂吗，yuv buffer没用完就被下一帧使用了，你这是输出环节还是哪个环节

DianKeWei, 2025-05-21:
这是编码器直接输出呢，可是我在每个的编码那边使用了mpp_buffer_sync_end ，难道不起作用。

> 这不就是撕裂吗，yuv buffer没用完就被下一帧使用了，你这是输出环节还是哪个环节

ippfcox, 2025-05-22:
> 这是编码器直接输出呢，可是我在每个的编码那边使用了mpp_buffer_sync_end ，难道不起作用。
> 
> > 这不就是撕裂吗，yuv buffer没用完就被下一帧使用了，你这是输出环节还是哪个环节

编码出来的ES就是这样？有没有可能是编码器编码当前的buffer还没用完，同一块内存又已经被前级采集重复使用了？

DianKeWei, 2025-05-22:
这样子说的话是不是，意味者v4l2 usb摄像头采集到数据，传给解码器，解码完成，释放采集帧，生成frame，传送给，编码器打包成成packet，最后释放packed，其中编码器用到采集帧的buffer么？ @ippfcox

ippfcox, 2025-05-23:
> 这样子说的话是不是，意味者v4l2 usb摄像头采集到数据，传给解码器，解码完成，释放采集帧，生成frame，传送给，编码器打包成成packet，最后释放packed，其中编码器用到采集帧的buffer么？ [@ippfcox](https://github.com/ippfcox)

哦你不是采集直接采YUV，这样实际就是解码器内部的YUVbuffer与编码器交互，有没有可能是编码还没有用完这个buffer，解码那边已经重新复用了？确认一下是否是在编码器用完了这个frame之后才调用的mpp_frame_deinit看看？我觉得编码器异常的可能性不大，我用过的3588和3562都没有这种问题，我猜测还是buffer管理的问题

ippfcox, 2025-05-23:
> 这样子说的话是不是，意味者v4l2 usb摄像头采集到数据，传给解码器，解码完成，释放采集帧，生成frame，传送给，编码器打包成成packet，最后释放packed，其中编码器用到采集帧的buffer么？ [@ippfcox](https://github.com/ippfcox)

还可能是解码器相关的buffer管理问题，但是这块比较复杂，我可能也说不清楚，我之前用的是半内部的分配方式，没遇到过这种问题

ippfcox, 2025-05-26:
> [@ippfcox](https://github.com/ippfcox) 问题定位到了，已经把所有的值编解码出来拿出来比较，发现是mjepg解码器有点问题，camera采取内部分配buffer，解码器使用外部分配的方式，相关操作解码， 到这种问题下，有时候偶尔性会出现撕裂， 3566板子 ，收集到的mjpeg mppbuffer释放在 ` dec->data_clear(&data, &dec->user_data)`
> 
> ` dec->data_entry ? dec->data_entry(&data, &dec->user_data) : (void)0; if (mpi->poll(dec->ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK)) { mpp_err_f("mpp input poll failed\n"); continue; } if (mpi->dequeue(dec->ctx, MPP_PORT_INPUT, &dec->task)) { mpp_err_f("mpp input dequeue failed\n"); continue; } mpp_buffer_sync_end(mpp_packet_get_buffer(data.packet)); mpp_task_meta_set_packet(dec->task, KEY_INPUT_PACKET, data.packet); mpp_task_meta_set_frame(dec->task, KEY_OUTPUT_FRAME, data.frame);
> 
> ```
>     if (mpi->enqueue(dec->ctx, MPP_PORT_INPUT, dec->task))
>     {
>         mpp_err_f("mpp input enqueue failed\n");
>         continue;
>     }
> 
>     if (mpi->poll(dec->ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK))
>     {
>         mpp_err_f("mpp output poll failed\n");
>         continue;
>     }
>     if (mpi->dequeue(dec->ctx, MPP_PORT_OUTPUT, &dec->task))
>     {
>         mpp_err_f("mpp output enqueue failed\n");
>         continue;
>     }
>     dec_frame = NULL;
>     mpp_task_meta_get_frame(dec->task, KEY_OUTPUT_FRAME, &dec_frame);
>     if (mpi->enqueue(dec->ctx, MPP_PORT_OUTPUT, dec->task))
>     {
>         mpp_err_f("mpp output enqueue failed\n");
>         continue;
>     }
>     /**! 释放缓冲区 */
>     mpp_frame_set_hor_stride(dec_frame, dec->hor_stride);
>     mpp_frame_set_ver_stride(dec_frame, dec->ver_stride);
>     mpp_buffer_sync_end(data.frm_buf);
>     /**! 传送编码任务 需要加入释放packet函数 mpp_packet_deinit(&data.packet) */
>     dec->data_handler ? dec->data_handler(&data, &dec->user_data) : (void)0;
>     dec->data_clear ? dec->data_clear(&data, &dec->user_data) : (void)0;`
> ```

🤔好的，这样看起来我这边也需要自查一下，感谢提供思路