rockchip-linux/mpp #855: 在应用层的native层中使用mpp报错：dma_buf_map_attachment fd 420 failed

thiagooo0, 2025-05-22:
前100帧左右可以编出来，然后就会报下述错误 ，并且生成的都是sei帧。再过几十帧app就会崩溃。有点怀疑某些内存没释放，但是不太知道是哪里的内存。求教

240: dma_buf_map_attachment fd 420 failed(-22)
 mpp_task_attach_fd:1774: can't import dma-buf 420
 mpp_translate_reg_address:1831: reg[  0]: 0x000001a4 fd 420 failed
 mpp_task_dump_mem_region:2017: --- dump task 0 mem region ---
 no memory region mapped
 mpp_process_task_default:628: alloc_task failed.
 2040: session 000000004748aa7d pending list is empty!
 mpp_msgs_wait:1627: session 41 wait result ret -5

-----------------------------------
相关代码如下：
enqueue：
void MppEncoder::putPacket(char *buf, size_t size) {
    auto pollRet = mMppApi->poll(mMppCtx, MPP_PORT_INPUT, MPP_POLL_BLOCK);
    MppTask inputTask = nullptr;
    auto dequeueRet = mMppApi->dequeue(mMppCtx, MPP_PORT_INPUT, &inputTask);
    if (inputTask != nullptr) {
        MppFrame frame = nullptr;
        auto frameInitRet = mpp_frame_init(&frame);

        MppBuffer buffer = nullptr;
        mpp_buffer_get(group, &buffer, size);

        MppMeta meta = nullptr;
        MppPacket packet = nullptr;
        mpp_buffer_write(buffer, 0, buf, size);

        mpp_frame_set_buffer(frame, buffer);
        mpp_frame_set_width(frame, width);
        mpp_frame_set_height(frame, height);
        mpp_frame_set_hor_stride(frame, width);
        mpp_frame_set_ver_stride(frame, height);
        mpp_frame_set_pts(frame, getTimestamp());
//        mpp_frame_set_fmt(frame, MPP_FMT_YUV420SP);
        mpp_frame_set_fmt(frame, MPP_FMT_BGR888);

        meta = mpp_frame_get_meta(frame);
        mpp_packet_init_with_buffer(&packet, buffer);
        mpp_packet_set_length(packet, 0);
        mpp_meta_set_packet(meta, KEY_OUTPUT_PACKET, packet);

        auto setFrameRet = mpp_task_meta_set_frame(inputTask, KEY_INPUT_FRAME, frame);
        auto enqueueRet = mMppApi->enqueue(mMppCtx, MPP_PORT_INPUT, inputTask);
    }
}

-----------------
dequeue：
auto pollRet = mMppApi->poll(mMppCtx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);
            MppTask outputTask = nullptr;
            auto dequeueRet = mMppApi->dequeue(mMppCtx, MPP_PORT_OUTPUT, &outputTask);
            MppPacket packet = nullptr;
            auto packetRet = mpp_task_meta_get_packet(outputTask, KEY_OUTPUT_PACKET, &packet);
            auto length = mpp_packet_get_length(packet);
            auto data = static_cast<char *>(mpp_packet_get_data(packet));
            fwrite(data, 1, length, file);
            mpp_packet_deinit(&packet);
            auto enqueueRet = mMppApi->enqueue(mMppCtx, MPP_PORT_OUTPUT, outputTask);

HermanChen, 2025-09-29:
看 fd 到 420 了，估计是 buffer 有泄漏，用 mpi_enc_test 跑一跑，比较大概率是帧 buffer 在编码完之后没有 deinit

GOAWAY88, 2025-10-15:
请问使用mpp_buffer_group_get_internal获取缓存后，我在程序结束时调用mpp_buffer_group_put，来释放缓存，那么还要调用mpp_buffer_group_clear不？ @HermanChen