rockchip-linux/mpp #863: 硬解码最大缓冲数据包数量是否支持配置

William-Li01, 2025-06-12:
1.1.4 最大缓冲数据包数量
MPP实例默认可以接收4个输入码流包在处理队列中，如果码流送得太快，就会报出错误码要求用户等待后再送。

在用户手册中有写到MPP硬解码最大缓冲数据量为4，那么会导致输入4帧数据才会输出解码内容，目前我希望输入一帧图片到解码器，立即输出解码结果，所以需要配置这个缓冲数据量，如何配置为1，是否支持配置？

HermanChen, 2025-06-12:
这要求码流的解码序和显示序是相同的，可以开 base:fast_out 选项

William-Li01, 2025-06-12:
我这边测试MPP解码程序输入前三帧数据，解码器不输出，当输入第四帧数据，才解码出第一帧图像。

Breeze
***@***.***

William-Li01, 2025-06-12:
我这添加了ret = mpp_dec_cfg_set_u32(cfg, "base:fast_out", fast_out);配置后，依然是缓存着三帧图片，不能立即输出。

Breeze
***@***.***

William-Li01, 2025-06-13:
我这个解码的是H264数据，只有第一包不是SPS/PPS头信息，其他每一包都是。

Breeze
***@***.***

 
看下前三包大小，是不是 vps / sps / pps 这些头信息

William-Li01, 2025-06-13:
你好，我这边测试了最新版的mpi_dec_test程序，用来测试解码延迟，输入的H264文件每一帧都为关键帧，测试发现最后有三帧缓存，那么我希望解码一帧立即输出，队列不进行缓存，参考你的建议，我在例程中添加了base:fast_out配置，编译运行后，效果和没加配置一样，依然缓存了三帧数据，没效果，所以还请教下如何解决。

Breeze
***@***.***