rockchip-linux/mpp #865: RK3568 mpi_dec_test 解码 16M 的 H264 (320*240-15FPS) 的文件, 其CPU占用率高达: 100%

GarenXie1, 2025-06-16:
RK Mpp 团队您好:

RK3568 执行解码命令

- mpi_dec_test 解码 H264 (320*240-15FPS) 的文件

> mpi_dec_test -t 7 -i /data/testNPU_2.h264

[image]

htop 命令查询 mpi_dec_test CPU 占用率

[image]

自行尝试办法:

- 打印 VPU 的 DEBUG_TIMING log, 得知是: VPU 硬件解码

> echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug

[image]
疑问:  VPU 硬解码 320*240 的 H264, 为啥会存在 100% CPU 占用率呢?  @HermanChen

GarenXie1, 2025-06-17:
增加信息.

- htop 找到最大cpu 占用的 线程名: **mpp_dec_parser**

[image]

HermanChen, 2025-06-17:
mpi_dec_test 是全速解码的 demo，占满 cpu 也是可能的，是否可以提供下码流我们分析下
比较大的可能是码流分辨率太小，硬件解码器速度太快，cpu 停不下来休息，直到占满 cpu，如果输入码流的地方休眠一下控制送帧速度，cpu 就不会占那么高了