rockchip-linux/mpp #869: MJPEG解码异常

youyou0819, 2025-07-03:
进程一：从USB摄像头取MJPEG对其解码，再进行编码成H264的rtsp流;
进程二：从进程一的H264码流对其解码成NV12数据，部分情况下会编码成JPEG图片;

mpp已经更新到最新
进程一会遇到:
mpp[7968]: hal_jpegd_rkv: decode result: failed, irq 0x0000000d
[2024-08-27 19:38:44][W] mpp_frame_get_errinfo:256 mpp_frame_get_discard:0
mpp[7968]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[2024-08-27 19:38:50][W] mpp_frame_get_errinfo:1 mpp_frame_get_discard:0
mpp[7968]: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[2024-08-27 19:38:50][W] mpp_frame_get_errinfo:1 mpp_frame_get_discard:0

最终持续打印：
mpp[7968]: hal_jpegd_rkv: decode result: failed, irq 0x0000514c
[2024-08-27 19:44:20][W] mpp_frame_get_errinfo:256 mpp_frame_get_discard:0

请问下是什么问题导致的;

youyou0819, 2025-07-03:
[image]
内核的日志信息

HermanChen, 2025-07-03:
什么芯片，多大分辨率