rockchip-linux/mpp #873: RK3588 buildroot 运行 mpi_enc_test 报段错误

satchelwu, 2025-07-10:
./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg

root@RK3588-Tronlong:~/xllcamera_rk/bin# strace ./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg
execve("./mpi_enc_test", ["./mpi_enc_test", "-w", "1280", "-h", "720", "-n", "1", "-o", "test.jpg"], 0x7fcff9aee0 /* 46 vars */) = 0
brk(NULL)                               = 0x557d980000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f98dc0000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/build/lib/opencv-mobile/install/lib/librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/build/lib/opencv-mobile/install/lib/", 0x7ff69f94d0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/lib/ffmpeg/lib/librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/lib/ffmpeg/lib/", 0x7ff69f94d0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/build/thirdparty/libtiff/install/lib/librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/build/thirdparty/libtiff/install/lib/", 0x7ff69f94d0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/bin/librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/wuqiang/company_projects/xllcamera_rk/bin/", 0x7ff69f94d0, 0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0700, st_size=2599952, ...}, AT_EMPTY_PATH) = 0
getcwd("/root/xllcamera_rk/bin", 128)   = 23
mmap(NULL, 2376544, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f98b7b000
mmap(0x7f98da5000, 102400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22a000) = 0x7f98da5000
mmap(0x7f98dbe000, 4960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f98dbe000
close(3)                                = 0
openat(AT_FDCWD, "libm.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=35505, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 35505, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f98b72000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=472872, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 471064, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f98afe000
mmap(0x7f98b70000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x72000) = 0x7f98b70000
close(3)                                = 0
openat(AT_FDCWD, "libstdc++.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1790472, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 1801632, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f98946000
mmap(0x7f98aec000, 61440, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a6000) = 0x7f98aec000
mmap(0x7f98afb000, 11680, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f98afb000
close(3)                                = 0
openat(AT_FDCWD, "libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\314\264\2\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=1388392, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 1433336, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f987e8000
mmap(0x7f98935000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x14d000) = 0x7f98935000
mmap(0x7f9893a000, 48888, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f9893a000
close(3)                                = 0
openat(AT_FDCWD, "libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=75968, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 78552, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f987d4000
mmap(0x7f987e6000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x11000) = 0x7f987e6000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f987d2000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f987d0000
set_tid_address(0x7f987d00f0)           = 1198
set_robust_list(0x7f987d0100, 24)       = 0
rseq(0x7f987d0740, 0x20, 0, 0xd428bc00) = 0
mprotect(0x7f98935000, 12288, PROT_READ) = 0
mprotect(0x7f987e6000, 4096, PROT_READ) = 0
mprotect(0x7f98b70000, 4096, PROT_READ) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f987ce000
mprotect(0x7f98aec000, 45056, PROT_READ) = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f987cb000
mprotect(0x7f98da5000, 28672, PROT_READ) = 0
mprotect(0x5560f42000, 4096, PROT_READ) = 0
mprotect(0x7f98de6000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7f98b72000, 35505)             = 0
getrandom("\xb4\xc6\x10\xb5\xca\x66\x83\x14", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x557d980000
brk(0x557d9a1000)                       = 0x557d9a1000
openat(AT_FDCWD, "/dev/kmpp_objs", O_RDWR) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/localtime", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=561, ...}, AT_EMPTY_PATH) = 0
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=561, ...}, AT_EMPTY_PATH) = 0
read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 561
lseek(3, -342, SEEK_CUR)                = 219
read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 342
close(3)                                = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 3
connect(3, {sa_family=AF_UNIX, sun_path="/dev/log"}, 110) = 0
sendto(3, "<11>Jul 10 20:26:48 mpi_enc_test"..., 120, MSG_NOSIGNAL, NULL, 0) = 120
sendto(3, "<11>Jul 10 20:26:48 mpi_enc_test"..., 88, MSG_NOSIGNAL, NULL, 0) = 88
openat(AT_FDCWD, "/dev/dma_heap/system-uncached", O_RDONLY|O_CLOEXEC) = 4
openat(AT_FDCWD, "/dev/dma_heap/system-uncached-dma32", O_RDONLY|O_CLOEXEC) = 5
openat(AT_FDCWD, "/dev/dma_heap/system", O_RDONLY|O_CLOEXEC) = 6
openat(AT_FDCWD, "/dev/dma_heap/system-dma32", O_RDONLY|O_CLOEXEC) = 7
openat(AT_FDCWD, "/dev/dma_heap/cma-uncached", O_RDONLY|O_CLOEXEC) = 8
openat(AT_FDCWD, "/dev/dma_heap/cma-uncached-dma32", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/dev/dma_heap/cma", O_RDONLY|O_CLOEXEC) = 9
openat(AT_FDCWD, "/dev/dma_heap/cma-dma32", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
fcntl(8, F_DUPFD_CLOEXEC, 3)            = 10
fcntl(10, F_DUPFD_CLOEXEC, 3)           = 11
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x40} ---
+++ killed by SIGSEGV +++
Segmentation fault

HermanChen, 2025-07-11:
用 root 运行，可能是权限问题？

satchelwu, 2025-07-11:
> 用 root 运行，可能是权限问题？

确认是root运行的

root@RK3588-Tronlong:~/xllcamera_rk/bin# LD_PRELOAD=librockchip_mpp.so.1 ./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg
Segmentation fault

satchelwu, 2025-07-11:
会不会需要用很老的版本才行呢？