rockchip-linux/mpp #883: rockit好像不支持GDR编码的h264流？解码器一直报错。mpp_dec_test也会报错

zhanglm666, 2025-08-01:
[image]

[image]

HermanChen, 2025-08-04:
加个 control disable_error

zhanglm666, 2025-08-05:
> 加个 control disable_error

用rockit进行解码呢？该如何支持GDR

zhanglm666, 2025-08-05:
> 加个 control disable_error

[image]
我在初始化和解码前都配置了disable_error，也同样报错。视频流是GDR编码的，请问是解码器不支持吗？

HermanChen, 2025-08-05:
报错是正常的，出来的帧能显示么

zhanglm666, 2025-08-05:
> 报错是正常的，出来的帧能显示么

还有，rockit是否支持GDR解码，MPP是否支持，还请明确说明，不支持我就找别的方案了，不再继续浪费时间。

HermanChen, 2025-08-06:
rockit 内部是调用 mpp 来处理的，mpp 有支持，正常 gdr 码流在开 disable_error 后是可以解的
有问题的话提供下片源

closed 2025-08-09 by HermanChen