rockchip-linux/mpp #884: 使用mpp解码rtsp h265流报错

GOAWAY88, 2025-08-06:
C2RKChips               and...hardware.media.c2@1.1-service  E  open /sys/bus/nvmem/devices/rockchip-efuse0/nvmem error
2025-08-06 14:23:29.488   391-9674  C2RKChips               and...hardware.media.c2@1.1-service  E  open /sys/bus/nvmem/devices/rockchip-efuse0/nvmem error
2025-08-06 14:23:29.489  7487-8004  BufferPoolAccessor2.0   com.honghe.integratedmachine         D  bufferpool2 0xb4000079eb6c1b88 : 5(31334400 size) total buffers - 4(25067520 size) used buffers - 5796/5801 (recycle/alloc) - 17/11599 (fetch/transfer)
2025-08-08 22:26:39.266  9681-9681  mpp_dma_import_fd       pid-9681                             E  240: dma_buf_map_attachment fd 482 failed(-22)
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_attach_fd:1780: can't import dma-buf 482
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_translate_reg_address:1837: reg[130]: 0x000001e2 fd 482 failed
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2023: --- dump task 0 mem region ---
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[128]: 0x00000000fa030000, size a9000
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[129]: 0x00000000fa030000, size a9000
2025-08-08 22:26:39.266  9681-9681  rk_vcodec               pid-9681                             E  mpp_process_task_default:628: alloc_task failed.
2025-08-08 22:26:39.266  9682-9682  rk_vcodec               pid-9682                             E  mpp_wait_result_default:919: session 9674:564 pending list is empty!
2025-08-08 22:26:39.266  9682-9682  rk_vcodec               pid-9682                             E  mpp_msgs_wait:1633: session 564 wait result ret -5
2025-08-08 22:26:39.267  9681-9681  mpp_dma_import_fd       pid-9681                             E  240: dma_buf_map_attachment fd 482 failed(-22)
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_attach_fd:1780: can't import dma-buf 482
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_translate_reg_address:1837: reg[132]: 0x000001e2 fd 482 failed
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2023: --- dump task 0 mem region ---
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[128]: 0x00000000fe310000, size a9000
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[129]: 0x00000000fe310000, size a9000
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[130]: 0x000000001a0a0000, size c74000
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_task_dump_mem_region:2029: reg[131]: 0x000000000c040000, size ff000
2025-08-08 22:26:39.267  9681-9681  rk_vcodec               pid-9681                             E  mpp_process_task_default:628: alloc_task failed.
2025-08-08 22:26:39.267  9682-9682  rk_vcodec               pid-9682                             E  mpp_wait_result_default:919: session 9674:564 pending list is empty!
2025-08-08 22:26:39.267  9682-9682  rk_vcodec               pid-9682                             E  mpp_msgs_wait:1633: session 564 wait result ret -5

GOAWAY88, 2025-08-06:
if (is_stream_active) {
                uint8_t* pkt_data = stPacket.pPacketBuffer;
                int pkt_size = stPacket.dwPacketSize;
                RK_U32 pkt_done = 0;
                RK_U32 err_info = 0;
                RK_U32 pkt_eos  = 0;
                ret = mpp_packet_init(&(pThis->packet), NULL, 0);
                LOGI("[DecodeStream] decode_mpp 2");
                mpp_packet_set_data(pThis->packet, pkt_data);
                mpp_packet_set_size(pThis->packet, pkt_size);
                mpp_packet_set_pos(pThis->packet, pkt_data);
                mpp_packet_set_length(pThis->packet, pkt_size);

                if (pkt_eos)
                    mpp_packet_set_eos(pThis->packet);

                if (!pkt_done) {
                    ret = pThis->mpi->decode_put_packet(pThis->ctx, pThis->packet);
                    if (MPP_OK == ret)
                        pkt_done = 1;
                }
                RK_S32 times = 5;
                while (true) {
                    ret = pThis->mpi->decode_get_frame(pThis->ctx, &(pThis->frame));
                    if (MPP_ERR_TIMEOUT == ret) {
                        if (times > 0) {
                            times--;
                            usleep(2000);
                            continue;
                        }
                        LOGD("decode_get_frame failed too much time ");
                        break;
                    }
                    if (MPP_OK != ret) {
                        LOGD("decode_get_frame failed ret %d \n", static_cast<int>(ret));
                        break;
                    }
                    LOGI("[DecodeStream] decode_mpp 3 frame:%p", static_cast<void*>(pThis->frame));
                    if (pThis->frame) {
                        RK_U32 hor_stride = mpp_frame_get_hor_stride(pThis->frame);
                        RK_U32 ver_stride = mpp_frame_get_ver_stride(pThis->frame);
                        RK_U32 hor_width = mpp_frame_get_width(pThis->frame);
                        RK_U32 ver_height = mpp_frame_get_height(pThis->frame);
                        RK_S64 pts = mpp_frame_get_pts(pThis->frame);
                        RK_S64 dts = mpp_frame_get_dts(pThis->frame);

                        if (mpp_frame_get_info_change(pThis->frame)) {
                            if (NULL == pThis->frm_grp) {
                                ret = mpp_buffer_group_get_internal(&(pThis->frm_grp), MPP_BUFFER_TYPE_ION);
                                if (ret) break;

                                ret = pThis->mpi->control(pThis->ctx, MPP_DEC_SET_EXT_BUF_GROUP, pThis->frm_grp);
                                if (ret) break;
                            } else {
                                ret = mpp_buffer_group_clear(pThis->frm_grp);
                                if (ret) break;
                            }

                            ret = pThis->mpi->control(pThis->ctx, MPP_DEC_SET_INFO_CHANGE_READY, NULL);
                            if (ret) break;
                        } else {
                            err_info = mpp_frame_get_errinfo(pThis->frame) | mpp_frame_get_discard(pThis->frame);
                            if (!err_info) {
                                 //解码当前帧并返回
                                dump_mpp_frame_to_file(frame, "decode");
                            }
                        }
                        bool frm_eos = mpp_frame_get_eos(pThis->frame);
                        if (frm_eos) {
                            LOGD("found last frame ");
                        }
                        mpp_frame_deinit(&(pThis->frame));
                        break;
                    }

                    if (pkt_eos && pkt_done) {
                        usleep(1000);
                        continue;
                    }
                    break;
                } // while
                mpp_packet_deinit(&(pThis->packet));
            }

HermanChen, 2025-08-09:
rtsp 是会丢包的，把数据存下为用 ffmpeg 验证下是不是丢包错包了，码流有错的话，解码器报错是正常的

HermanChen, 2025-08-25:
编码端发个 I 帧来，解码就自然恢复了

closed 2025-10-15 by GOAWAY88