rockchip-linux/mpp #885: mpp h264 硬件编码后，输出的视频会有横纹

cppcoffee, 2025-08-06:
mpp h264 硬件编码后，输出的视频会有横纹，大概3秒后消失，然后又出现横纹，如此往复

[image]

cppcoffee, 2025-08-07:
压缩包里面是 mpp 对 HDMI-IN 的视频流做的 h264 压缩后，出现黑白横纹的问题的 h264 数据

[issue.h264.zip](https://github.com/user-attachments/files/21662202/issue.h264.zip)

```shell
root@noble-he2221:~# v4l2-ctl --all
Driver Info:
        Driver name      : rkcif
        Card type        : rkcif
        Bus info         : platform:rkcif_mipi_lvds
        Driver version   : 6.1.115
        Capabilities     : 0x84201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04201000
                Video Capture Multiplanar
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : rkcif
        Model            : rkcif_mipi_lvds
        Serial           :
        Bus info         : platform:rkcif_mipi_lvds
        Media version    : 6.1.115
        Hardware revision: 0x00000000 (0)
        Driver version   : 6.1.115
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : stream_cif_mipi_id0
        Function         : V4L2 I/O
        Pad 0x01000004   : 0: Sink
          Link 0x0200002f: from remote pad 0x1000022 of entity 'rockchip-mipi-csi2' (Unknown V4L2 Sub-Device): Data, Enabled
          Link 0x0200003d: from remote pad 0x1000023 of entity 'rockchip-mipi-csi2' (Unknown V4L2 Sub-Device): Data
          Link 0x0200004b: from remote pad 0x1000024 of entity 'rockchip-mipi-csi2' (Unknown V4L2 Sub-Device): Data
          Link 0x02000059: from remote pad 0x1000025 of entity 'rockchip-mipi-csi2' (Unknown V4L2 Sub-Device): Data
Priority: 2
Format Video Capture Multiplanar:
        Width/Height      : 2560/1440
        Pixel Format      : 'NV12' (Y/UV 4:2:0)
        Field             : None
        Number of planes  : 1
        Flags             :
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Plane 0           :
           Bytes per Line : 2560
           Size Image     : 5529600
Selection Video Capture: crop, Left 0, Top 0, Width 2560, Height 1440, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 2560, Height 1440, Flags:
Selection Video Output: crop, Left 0, Top 0, Width 2560, Height 1440, Flags:
Selection Video Output: crop_bounds, Left 0, Top 0, Width 2560, Height 1440, Flags:
```

cppcoffee, 2025-08-07:
其中编码参数实现源码如下所示：

```c
#include "enc_cfg.h"
#include <mpp_rc_api.h>

#include "logger.h"
#include "options.h"

static double calculate_bitrate(option_t *opt, RK_S32 fps_out_num, RK_S32 fps_out_den);

static RK_S32 aq_thd[16] = {
    0,  0,  0,  0,
    3,  3,  5,  5,
    8,  8,  8,  15,
    15, 20, 25, 25
};

static RK_S32 aq_step_i_ipc[16] = {
    -8, -7, -6, -5,
    -4, -3, -2, -1,
    0,  1,  2,  3,
    5,  7,  7,  8,
};

static RK_S32 aq_step_p_ipc[16] = {
    -8, -7, -6, -5,
    -4, -2, -1, -1,
    0,  2,  3,  4,
    6,  8,  9,  10,
};

/*
 * opt参数
 *  opt->fps: 60
 *  opt->width: 2560
 *  opt->height: 1440
 *  opt->hor_stride: 2560
 *  opt->ver_stride: 1440
 */
int
setup_mpp_enc_cfg(context_t *p, option_t *opt)
{
    int rc;
    MppEncCfg cfg = p->enc_cfg;
    /* setup default parameter */
    RK_S32 fps_in_flex = 0;
    RK_S32 fps_out_flex = 0;
    RK_S32 fps_in_den = 1;
    RK_S32 fps_in_num = opt->fps;
    RK_S32 fps_out_den = 1;
    RK_S32 fps_out_num = opt->fps;
    RK_S32 bps = calculate_bitrate(opt, fps_out_num, fps_out_den);

    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_i", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_p", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_i", aq_step_i_ipc);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_p", aq_step_p_ipc);

    mpp_enc_cfg_set_s32(cfg, "rc:max_reenc_times", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:anti_flicker_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_p", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atl_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_p", 0);

    mpp_enc_cfg_set_s32(cfg, "tune:scene_mode", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_en", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_str", 0);
[... 142 lines cut ...]

    switch (opt->video_quality) {
    case VIDEO_QUALITY_FLUENT:
        bps_factor = 0.05;
        break;

    case VIDEO_QUALITY_STANDARD:
        bps_factor = 0.1;
        break;

    case VIDEO_QUALITY_HIGH:
        bps_factor = 0.5;
        break;

    case VIDEO_QUALITY_SUPER:
    default:
        bps_factor = 1.0;
        break;
    }

    base_bps = opt->width * opt->height / 8 * (fps_out_num / fps_out_den);

    return base_bps * bps_factor;
}
```

cppcoffee, 2025-08-07:
编码参数配置（拷贝来自 https://github.com/rockchip-linux/mpp/blob/develop/test/mpi_enc_test.c）如下所示：

```c
#include "enc_cfg.h"
#include <mpp_rc_api.h>

#include "options.h"

static double calculate_bitrate(option_t *opt, RK_S32 fps_out_num, RK_S32 fps_out_den);

static RK_S32 aq_thd[16] = {
    0,  0,  0,  0,
    3,  3,  5,  5,
    8,  8,  8,  15,
    15, 20, 25, 25
};

static RK_S32 aq_step_i_ipc[16] = {
    -8, -7, -6, -5,
    -4, -3, -2, -1,
    0,  1,  2,  3,
    5,  7,  7,  8,
};

static RK_S32 aq_step_p_ipc[16] = {
    -8, -7, -6, -5,
    -4, -2, -1, -1,
    0,  2,  3,  4,
    6,  8,  9,  10,
};

/*
 * opt参数
 *  opt->fps: 60
 *  opt->width: 2560
 *  opt->height: 1440
 *  opt->hor_stride: 2560
 *  opt->ver_stride: 1440
 */
int
setup_mpp_enc_cfg(context_t *p, option_t *opt)
{
    int rc;
    MppEncCfg cfg = p->enc_cfg;
    /* setup default parameter */
    RK_S32 fps_in_flex = 0;
    RK_S32 fps_out_flex = 0;
    RK_S32 fps_in_den = 1;
    RK_S32 fps_in_num = opt->fps;
    RK_S32 fps_out_den = 1;
    RK_S32 fps_out_num = opt->fps;
    RK_S32 bps = calculate_bitrate(opt, fps_out_num, fps_out_den);

    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_i", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_thrd_p", aq_thd);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_i", aq_step_i_ipc);
    mpp_enc_cfg_set_st(cfg, "hw:aq_step_p", aq_step_p_ipc);

    mpp_enc_cfg_set_s32(cfg, "rc:max_reenc_times", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:anti_flicker_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atr_str_p", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:atl_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_i", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:sao_str_p", 0);

    mpp_enc_cfg_set_s32(cfg, "tune:scene_mode", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_en", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:deblur_str", 0);
    mpp_enc_cfg_set_s32(cfg, "tune:rc_container", 0);
[... 105 lines cut ...]
    }

    MppEncHeaderMode header_mode = MPP_ENC_HEADER_MODE_EACH_IDR;
    rc = p->mpi->control(p->ctx, MPP_ENC_SET_HEADER_MODE, &header_mode);
    if (rc) {
        LOG_ERROR("failed to mpi control enc set header mode, rc=%d", rc);
        return -1;
    }

    return 0;
}

static double
calculate_bitrate(option_t *opt, RK_S32 fps_out_num, RK_S32 fps_out_den)
{
    double base_bps;
    double bps_factor;

    bps_factor = 1.0;

    base_bps = opt->width * opt->height / 8 * (fps_out_num / fps_out_den);

    return base_bps * bps_factor;
}
```

cppcoffee, 2025-08-08:
soc 使用的是 rk3566，看到支持 1080P h264 的编码，如果超过了1080P的编码，是不是会发生内核崩溃？

这里捕捉到了 使用 2940 x 1912 分辨率，mpp h264 编码，内核出现崩溃的崩溃日志:

[kernel_coredump.txt](https://github.com/user-attachments/files/21681976/kernel_coredump.txt)

想请问一下，rk3566 h264编码的 1080P 指的是 1920 x 1080吗？

HermanChen, 2025-08-09:
3566 能力是 1080p 的，能编码大于 1080p，但是固件硬件限制只能每行切slice，效果会变差，也就是上面看到的情况
所以不要用 3566 编码超过 1080p 的分辨率