rockchip-linux/mpp #891: rk3576硬解，调用内存出现问题

fys18648950197, 2025-08-21:
(+) Video --vid=1 (*) (h264 1920x1080 60.000fps)
 (+) Audio --aid=1 (*) (mp3 2ch 48000Hz)
     Audio --aid=2 (*) (ac3 6ch 48000Hz)
File tags:
 Artist: Blender Foundation 2008, Janus Bager Kristensen 2013
 Comment: Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
 Composer: Sacha Goedegebure
 Genre: Animation
 Title: Big Buck Bunny, Sunflower version
arm_release_ver: g13p0-01eac0, rk_so_ver: 11
Cannot load libcuda.so.1
[ffmpeg/video] h264_rkmpp: Failed to open system-dma32 heap
[ffmpeg/video] h264_rkmpp: Failed to initialize RKMPP decoder.
Could not open codec.
[ffmpeg/video] h264_rkmpp: Failed to open system-dma32 heap
[ffmpeg/video] h264_rkmpp: Failed to initialize RKMPP decoder.
Could not open codec.
Decoder init failed for h264_rkmpp
[ffmpeg/video] h264_rkmpp: Failed to open system-dma32 heap
[ffmpeg/video] h264_rkmpp: Failed to initialize RKMPP decoder.
Could not open codec.

HermanChen, 2025-08-22:
要把内核的 dmaheap 支持给打开，MPP 是用 dmaheap 来做为内存分配器，确认下有打开么？

fys18648950197, 2025-08-22:
在 /dev/dma_heap下我能查到system ，应该是已经开启了吧

fys18648950197, 2025-08-25:
我的系统是ubuntu22.04，我主要是想解决本地播放掉帧的问题，rk不支持ffmpeg，我用gst发现还是掉帧，佬你有什么思路吗，之前在验证mpp的编解码的时候就发现了，硬解始终调用不起来

HermanChen, 2025-08-26:
@JeffyCN 这里有个 gst 相关的问题，感觉应该是使用上的问题