rockchip-linux/mpp #898: MPP JPEG解码卡主

tubzby, 2025-09-02:
我有一个USB摄像头，支持YUYV422 + MJPEG 格式，需要做一个rtsp服务，目前YUYV422 格式已经通了，因为分辨率比较低，需要支持MJPEG输入。

现在碰到的问题是：
1. 打开摄像头，获取MJPEG 输出，OK
2. 调用MJPEG 解码，卡主。 我看了一些帖子，MJPEG解码需要参照dec_advance 来做，但是还是在poll 等待解码器输出时卡主。

测试代码如下：

```c
MPP_RET decode_one_jpeg(const char * path) {
    MPP_RET ret;
    MppApi *mpi;
    MppCtx ctx;
    MppTask task = NULL;
    MppDecCfg cfg = NULL;
    RK_U32 need_split = 1;
    MppFrameFormat jpgOutputFmt = MPP_FMT_YUV420SP;
    MppPacket packet;
    MppFrame frame;
    FILE *fp = NULL;
    char *jpg_buf = NULL;
    int n;
    MppBufferGroup buf_grp;
    MppBuffer frm_buf;

    // read buffer
    jpg_buf = malloc(1024*1024);
    if (NULL == jpg_buf) {
        printf("failed to allocate\n");
        ret = MPP_ERR_NULL_PTR;
        goto EXIT;
    }

    fp = fopen(path, "r");
    if (NULL == fp) {
        printf("open input error\n");
        ret = MPP_ERR_OPEN_FILE;
        goto EXIT;
    }

    n = fread(jpg_buf, 1, 1024*1024, fp);
    if (n <= 0) {
        printf("fread error\n");
        ret = MPP_ERR_OPEN_FILE;
        goto EXIT;
    }
    mpp_packet_init(&packet, jpg_buf, n);

    // allocate buffer for frame
    ret = mpp_buffer_group_get_internal(&buf_grp, MPP_BUFFER_TYPE_DRM | MPP_BUFFER_FLAGS_CACHABLE);
    if (ret) {
        printf("failed to get mpp buffer group ret %d\n", ret);
        goto EXIT;
    }

    ret = mpp_buffer_get(buf_grp, &frm_buf, 640 * 480 * 4);
    if (ret) {
        printf("failed to get buffer for input frame ret %d\n", ret);
        goto EXIT;
    }

    mpp_frame_init(&frame);
    mpp_frame_set_buffer(frame, frm_buf);

    // init decoder
    ret = mpp_create(&ctx, &mpi);
    if (ret) {
        printf("mpp_create error: %d\n", ret);
        goto EXIT;
    }

    ret = mpp_init(ctx, MPP_CTX_DEC, MPP_VIDEO_CodingMJPEG);
    if (ret) {
        printf("mpp_init error: %d\n", ret);
        goto EXIT;
    }

    mpp_dec_cfg_init(&cfg);

    ret = mpi->control(ctx, MPP_DEC_GET_CFG, cfg);
    if (ret) {
        printf("control error: %d\n", ret);
        goto EXIT;
    }
    ret = mpp_dec_cfg_set_u32(cfg, "base:split_parse", need_split);
    if (ret) {
        printf("mpp_dec_cfg_set_u32 ret: %d", ret);
        goto EXIT;
    }

    ret = mpi->control(ctx, MPP_DEC_SET_CFG, cfg);
    if (ret) {
        printf("failed to set decoder cfg ret: %d", ret);
        goto EXIT;
    }

    ret = mpi->control(ctx, MPP_DEC_SET_OUTPUT_FORMAT, &jpgOutputFmt);
    if (ret) {
        printf("failed to set output fromat ret: %d", ret);
        goto EXIT;
    }

    printf("start poll input\n");
    ret = mpi->poll(ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK);
    if (ret) {
        printf("input poll error: %d", ret);
        goto EXIT;
    }
    printf("finish poll input\n");

    ret = mpi->dequeue(ctx, MPP_PORT_INPUT, &task);
    if (ret) {
        printf("dequeue input error: %d", ret);
        goto EXIT;
    }
    printf("finish dequeue input\n");

    mpp_assert(task);

    mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, packet);
    mpp_task_meta_set_packet(task, KEY_OUTPUT_FRAME, frame);

    ret = mpi->enqueue(ctx, MPP_PORT_INPUT, task);
    if (ret) {
        printf("enqueue input error: %d", ret);
        goto EXIT;
    }
    printf("finish enqueue input\n");
    

    // 此处卡主
    ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);
    if (ret) {
        printf("output poll error: %d", ret);
        goto EXIT;
    }
    printf("finish poll output\n");

    ret = mpi->dequeue(ctx, MPP_PORT_OUTPUT, &task);
    if (ret) {
        printf("dequeue output error: %d", ret);
        goto EXIT;
    }

EXIT:
    if (fp) {
        fclose(fp);
    }
    if (jpg_buf) {
        free(jpg_buf);
    }
    if (frm_buf) {
        mpp_buffer_put(frm_buf);
    }
    if (buf_grp) {
        mpp_buffer_group_put(buf_grp);
    }
    mpp_packet_deinit(&packet);
    return ret;
}

```

执行时输出，然后卡主

```console
start poll input
finish poll input
finish dequeue input
finish enqueue input
```

tubzby, 2025-09-02:
去掉了也还是卡住

HermanChen, 2025-09-02:
用最新代码里的 dec_advanced 来试试

tubzby, 2025-09-02:
mpi_dec_test 跑是正常的。
目前看起来差别就是MppPacket MppFrame 的 MppBuffer 类型，有问题吗？

```console
rpdzkj@debian11:~/rockit_rtsp$ mpi_dec_test -i /tmp/img.jpg -o /tmp/output.yuv -w 640 -h 480 -t 8 -f 0
mpp[23346]: mpi_dec_utils: input file /tmp/img.jpg size 36864
mpp[23346]: mpi_dec_utils: cmd parse result:
mpp[23346]: mpi_dec_utils: input  file name: /tmp/img.jpg
mpp[23346]: mpi_dec_utils: output file name: /tmp/output.yuv
mpp[23346]: mpi_dec_utils: width      :  640
mpp[23346]: mpi_dec_utils: height     :  480
mpp[23346]: mpi_dec_utils: type       :    8
mpp[23346]: mpi_dec_utils: max frames :    0
mpp[23346]: mpi_dec_test: mpi_dec_test start
mpp[23346]: mpp_info: mpp version: ed377c99 author: Herman Chen   2023-12-14 fix[hal_enc]: Add encoder internal buffer sync
mpp[23346]: mpi_dec_test: 0x55810305c0 mpi_dec_test decoder test start w 640 h 480 type 8
mpp[23346]: mpi_dec_test: 0x55810305c0 decoded frame 0
mpp[23346]: mpi_dec_test: decode 1 frames time 6 ms delay   1 ms fps 148.02
mpp[23346]: mpi_dec_test: test success max memory 0.00 MB
```

HermanChen, 2025-09-02:
修改下看看

tubzby, 2025-09-02:
@HermanChen 改l了类型为 `ION` 也不行

tubzby, 2025-09-02:
现在运行中提示:

```console
mpp[37481]: mpp_packet: check_is_mpp_packet pointer 0x7ff2599e80 failed on check
mpp[37481]: mpp_info: mpp version: ed377c99 author: Herman Chen   2023-12-14 fix[hal_enc]: Add encoder internal buffer sync

```

tubzby, 2025-09-02:
好的，我试试

tubzby, 2025-09-02:
可以了，多谢 @HermanChen 

```c
rpdzkj@debian11:~/rockit_rtsp$ ./build/app/jpeg_decoder
mpp[56131]: kmpp_obj: kmpp_objdef_get invalid param def 0x556ffc6058 name 0x556ffa1240 objs (nil)
mpp[56131]: kmpp_frame: kmpp_frame_init KMPP_OBJ_NAME init failed
mpp[56131]: camera_source: input devices:Input 1
mpp[56131]: camera_source: fmt name: [Motion-JPEG]
mpp[56131]: camera_source: fmt pixelformat: 'MJPG', description = 'Motion-JPEG'
mpp[56131]: camera_source: fmt name: [YUYV 4:2:2]
mpp[56131]: camera_source: fmt pixelformat: 'YUYV', description = 'YUYV 4:2:2'
mpp[56131]: camera_source: get width 640 height 480
mpp[56131]: camera_source: get dma buf(0)-fd: 13
mpp[56131]: camera_source: get dma buf(1)-fd: 14
mpp[56131]: camera_source: get dma buf(2)-fd: 15
mpp[56131]: camera_source: get dma buf(3)-fd: 16
captured /tmp/img.jpg
mpp[56131]: mpp_info: mpp version: 6cd2495d author: Herman Chen   2025-04-03 docs: Update 1.0.9 CHANGELOG.md
start poll input
finish poll input
finish dequeue input
finish enqueue input
mpp[56131]: mpp_dec: mpp_dec_advanced_thread line(1085): Error! Get no buffer from input packet
mpp[56131]: mpp_buffer: check buffer found NULL pointer from mpp_dec_advanced_thread
finish poll output
mpp[56131]: mpp_buffer: mpp_group 0x5577b160e8 tag unknown caller decode_one_jpeg mode internal type drm deinit with 1228800 bytes not released
mpp[56131]: mpp_buffer:
dumping buffer group 0x5577b160e8 id 2 from mpp_buffer_group_deinit
mpp[56131]: mpp_buffer: mode internal
mpp[56131]: mpp_buffer: type drm
mpp[56131]: mpp_buffer: limit size 0 count 0
mpp[56131]: mpp_buffer: used buffer count 1
mpp[56131]: mpp_buffer: buffer 0x5577b163a8 fd   19 size    1228800 ref_count   1 discard 0 caller decode_one_jpeg
mpp[56131]: mpp_buffer: unused buffer count 0
mpp[56131]: mpp_mem_pool: mpp_mem_pool_put_f invalid mem pool ptr 0x5577b15f08 node 0x5577b15ee0 check (nil)
mpp[56131]: mpp_task_impl: mpp_task_queue_deinit output queue idx 0 task 0x5577b19a20 status 3 meta size 1
mpp[56131]: mpp_meta: dumping meta 1 node count 1
mpp[56131]: mpp_meta: key ofrm - s
test passed
mpp[56131]: mpp_buffer: ~MppBufferService cleaning misc group
mpp[56131]: mpp_buffer: ~MppBufferService cleaning leaked buffer
jpeg_decoder[56131]: mpp_mem_pool: put_pool pool mpp_frame found 1 used buffer size 224
```